cnlohr/ch32fun #307: Funny bug in startup

VanyeRijan, 2024-04-27:
Hi, 
I found a funny bug in file: ch32v003fun/attic/ch32v003evt/startup_ch32v003.c

I am surprised that nobody noticed before. :-)

The first line in  SystemInit48HSI() is:

RCC->CTLR  = RCC_HSION | RCC_PLLON; 				// Use HSI, but enable PLL.

This has the assumption that CTLR is zero after booting. But this is wrong. That
register has an only readable part for correction of RC oszillator. By writing a
0 in this field the clockfrequency on my part is 5% off.

Vanye

cnlohr, 2024-04-27:
What would you like it to be?  That line does not read back CTLR, but instead sets it to a specific value.  Are you saying we should also be setting HSITRIM?  Or do you suggest it should be read back? HSICAL is unwritten.

On second note - it's interesting that there is no symbol for HSICAL.

cnlohr, 2024-04-27:
OH wait!

That code is in the attic because it's part of the EVT, that's the official material from WCH.

you've discovered another reason why we don't use it.

VanyeRijan, 2024-04-27:
The reference manual V1.7 at page 17/18 show the CTRL register.

There is HSICAL wich ist marked as RO!
But the code line in my first message write a zero to HSICAL! 

I guess in HSICAL is the calibration value that is written by the
bootloader at startup. If you write a zero in this it destroy the
messured RC calibration value from factory.
The RC frequency of my device had an error of 5% after 
destroying the value!

I changed it to:
RCC->CTLR  |= (RCC_HSION | RCC_PLLON);

Now  my HSI is spot on! (measured with my RTB)

But a funny remark the HSI has a horrible, realy horrible
jitter that is also changing in depency of what happens inside
of the MCU. That is bad design, no internal LDO for the HSI on the die.
But unfortunatly this is something we can not change.

Let me show you a picture:
[image]

I wounder what is the impact of this for the ADC. :-D

Vanye

cnlohr, 2024-04-27:
Please read my second message. the code there is incorrect.  You are using the wrong code.

TommyMurphyTM1234, 2024-06-14:
Could this issue be closed to avoid any (further) confusion given that it's effectively invalid?

cnlohr, 2024-06-14:
Yes. Done.

closed 2024-06-14 by cnlohr