cnlohr/ch32fun #310: "stdio.h" file not found

panikinator, 2024-04-30:
The LSP(clangd) integration of my neovim seems to be complaining about "stdio.h" file not found on arch linux x64. Can't seem to figure out how to make my LSP aware of the (custom?) stdio.h header location. I'm a beginner btw and thanks in advance.

[image]

panikinator, 2024-04-30:
I forgot to mention that the make commands work just fine with no warnings etc whatsoever
it's just the neovim/LSP that's complaining.

eeucalyptus, 2024-04-30:
I'm not exactly sure about neovim but makefile project integration into editors is often a bit ugly. As far as I understand lsp, it relies on clangd, which in turn requires a compile_commands.json database. For cmake and other complex build systems, these are just spit out with the other build artifacts, but for makefiles you might need to generate it separately. There's also other options like clangd's "compile_flags.txt", but I tend to just not use clangd for that kind of stuff...

prosper00, 2024-04-30:
most likely, you need to configure your LSP on how to find the headers. if you have bear installed, you can probably just run 'bear -- make' - it'll 'watch' the compile and pull out the various include dirs for you. I think clang will automatically use the 'compile commands' json file that bear spits out. At least for the YCM vim extension, thats how it works

panikinator, 2024-04-30:
So it turns out I was lazy and trying to find alternative to the compile from source(from AUR) packages taking time so decided meanwhile to install pre built binary and somewhere down the road I installed ```riscv64-linux-gnu-gcc```. removing it solved the header file issues

thanks for the offered help
and sorry for bothering by fuck ups of my own

prosper00, 2024-04-30:
> So it turns out I was lazy and trying to find alternative to the compile from source(from AUR) packages taking time so decided meanwhile to install pre built binary and somewhere down the road I [...]

weird... on Debian, using riscv64-linux- causes header/library issues for me, seems like it wants to include some stuff from the linux kernel headers riscv64-unknown- is needed to find the right newlib.