cnlohr/ch32fun #314: USART example is empty

chrizw, 2024-05-10:
I'm tring to develop a USART receiver program, hoping to use RX interrupt, and I found that the example of uartdemo is not correct. seems like a blink program. is that demo correct? or if there is other demo for USART?

```
// Really basic self-contained demo for the ch32v003
// Doesn't rely on any of the weird HAL stuff from CH
// Final executable is ~1/4th the size.

// See funconfig.h to see how this uses the uart.

#include "ch32v003fun.h"
#include <stdio.h>

uint32_t count;

int main()
{
	SystemInit();

	// Enable GPIOD.
	RCC->APB2PCENR |= RCC_APB2Periph_GPIOD;

	// GPIO D0 Push-Pull, 10MHz Output
	GPIOD->CFGLR &= ~(0xf<<(4*0));
	GPIOD->CFGLR |= (GPIO_Speed_10MHz | GPIO_CNF_OUT_PP)<<(4*0);

	while(1)
	{
		GPIOD->BSHR = 1;	 // Turn on GPIOD0
		Delay_Ms( 50 );
		GPIOD->BSHR = 1<<16; // Turn off GPIOD0
		Delay_Ms( 50 );
		printf( "Count: %lu\n", count++ );
	}
}
```

prosper00, 2024-05-10:
this example uses the UART to handle the printf(). You'd need to dig into the ch32v003fun.c/.h files to see how the actual implementation of the UART is done.

See https://github.com/cnlohr/ch32v003fun/blob/acf5917158ca01d6616640e8b97ac5a75ff60266/ch32v003fun/ch32v003fun.c#L1303-L1353

chrizw, 2024-05-11:
yes, thx.
I have add '#define FUNCONF_USE_UARTPRINTF 1' in funconfig.h and rebuild it. it works.

closed 2024-05-11 by chrizw