cnlohr/ch32fun #315: Support for new CH32V006 / CH32V005 boards?

brian-smith-github, 2024-05-14:
Hi again.
Do you know if support will be added for the upcoming new versions of the chip - e.g.  CH32V006 and CH32V005. Both have more ram/flash, and a new version of the RISC-V core that now includes hardware-multiply (zmmul extension).
Thanks.

cnlohr, 2024-05-15:
We intend to add support as soon as we can get them in-hand.  Making use of the zmmul instruction will require at least GCC 12, which we don't yet have a recommended Windows GCC build yet, but for Linux we hope to support the platforms shortly after we get chips!

TommyMurphyTM1234, 2024-06-04:
> We intend to add support as soon as we can get them in-hand. Making use of the zmmul instruction will require at least GCC 12, which we don't yet have a recommended Windows GCC build yet

Hi @cnlohr - could you use the Embecosm or xPack binary builds perhaps?

* https://www.embecosm.com/resources/tool-chain-downloads/
* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/

Note that the xPack toolchain uses a different prefix to the "usual" one - `riscv-none-elf-*` rather than `riscvXX-unknown-elf-*` in case that's an issue (e.g. for your `Makefile`s):

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#riscv-none-elf-gcc

If those binary builds are not adequate and you need a specific build then, in my experience, using [crosstool-NG](https://crosstool-ng.github.io/) to do a [Canadian Cross build](https://en.wikipedia.org/wiki/Cross_compiler#Canadian_Cross) (build host = Linux, run host = Windows, target ISA = RISC-V) of the toolchain for Windows works well - e.g.:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1442

If I can be of any assistance let me know.

brian-smith-github, 2025-03-12:
Just to mention that CH32V002 and CH32V006 chips are now available in bulk lots over on Aliexpress (no dev boards yet though that I can see).

jrudolph, 2025-03-12:
Thanks for the notice, first batch seems almost gone already...

cnlohr, 2025-03-12:
To clarify, we now have support for Xpack compilers, and there is even an windows installer powershell script in the misc/ folder.

brian-smith-github, 2025-03-20:
I'll close this as dwillmore is already working on it (https://github.com/cnlohr/ch32fun/issues/524) and CH32V006 dev boards are available.

closed 2025-03-20 by brian-smith-github