cnlohr/ch32fun #318: How to use GPIO_digitalWrite_branching ?

romain145, 2024-05-31:
I think I need to use GPIO_digitalWrite_branching() because I need to write a variable to a pin.
I tried multiple ways to call GPIO_digitalWrite_branching
```
GPIO_digitalWrite_branching(GPIOv_from_PORT_PIN(GPIO_port_D, 0), 1);
GPIO_digitalWrite_branching(GPIOv_from_PORT_PIN(GPIO_port_D, 0), high);
```

but they all return errors:
```
In file included from examples\GPIO\GPIO.c:17:
examples\GPIO\GPIO.c: In function 'main':
extralibs/ch32v003_GPIO_branchless.h:331:72: error: lvalue required as left operand of assignment
 #define GPIO_digitalWrite_lo(GPIOv)     GPIOv_to_GPIObase(GPIOv)->BSHR = (1 << (16 + GPIOv_to_PIN(GPIOv)))
                                                                        ^
extralibs/ch32v003_GPIO_branchless.h:341:96: note: in expansion of macro 'GPIO_digitalWrite_lo'
 #define GPIO_digitalWrite_branching(GPIOv, lowhigh)   (lowhigh ? GPIO_digitalWrite_hi(GPIOv) : GPIO_digitalWrite_lo(GPIOv))
                                                                                                ^~~~~~~~~~~~~~~~~~~~
examples\GPIO\GPIO.c:103:3: note: in expansion of macro 'GPIO_digitalWrite_branching'
   GPIO_digitalWrite_branching(GPIOv_from_PORT_PIN(GPIO_port_D, 0), 1);
```

What am I doing wrong?

hexaedron, 2024-05-31:
Use funDigitalWrite()/funDigitalRead() and funPinMode() instead.

romain145, 2024-05-31:
Exactly what I was looking for.
Thanks!

closed 2024-05-31 by romain145