cnlohr/ch32fun #325: Error while flashing: (Processor not halted.) DMSTATUS: 004c0c82

struc, 2024-06-21:
I am getting this error while trying to flash CH32V003F4P6

Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82                                                                                                                         
Fault on DefaultReadWord Part 1                                                                                                                                                                                  
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82                                                                                                                         
Fault on DefaultWriteWord Part 3                                                                                                                                                                                 
Error: Flash operation error                                                                                                                                                                                     
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82                                                                                                                         
Fault on DefaultWriteWord Part 3                                                                                                                                                                                 
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82                                                                                                                         
Fault on DefaultWriteWord Part 3
....

I am using Ardulink for flashing and a bare chip (with tssop-20 adapter board) with 100nf MLCC decoupling capacitor between VDD and VSS.

I do not get any errors when I first power off the chip, though the MCU does not function at all and the blink program only works for 1-3 seconds then stops, then if I power off the chip for 5 minutes it will work again for few seconds.

I have tried to find any information about the (DMSTATUS: 004c0c82) part but could not find any.
I have already tried 10 chips, and got the same result with 9, while only one works fine.

so my question is what does DMSTATUS means? and if there are anything I need to change in my setup.

cnlohr, 2024-06-21:
(1) I don't know anything about Ardulink?

(2) What happens if you "unbrick" your processor after a fresh poweron, with minichlink -u?

cnlohr, 2024-06-21:
08000402 = The processor is currently running and not in halt mode.

I would like to know more about your programmer.

cnlohr, 2024-06-21:
Just ccing @maxgerhardt in case they have any ideas, since they were who originally wrote the programmer.  It looks like they are not getting into the processor-halt state when programming.

cnlohr, 2024-06-21:
@struc can you post your full log?

maxgerhardt, 2024-06-21:
Ardulink (https://gitlab.com/BlueSyncLine/arduino-ch32v003-swio) was originally written by @BlueSyncLine, he knows more than me. :D

struc, 2024-06-21:
@cnlohr I am using Arduino Uno R3 with arduino-ch32v003-swio and pin 11 as SWIO and 2.2k resistor(also tried 1k and 4k7)

I have also tried to play a little with baud rates from 9600 up to 222200

The funny thing is that this only happens after awhile (5 seconds up to 1 minute).

I guess I should try another flashing method, but I do not have at hand at the moment.

Running `minichlink -c /dev/ttyACM0 -u` gave me this output:

```
Opening serial port /dev/ttyACM0 at 115200 baud.
Ardulink: synced.
Found Ardulink Programmer
Interface Setup
Entering Unbrick Mode
Ardulink: target power 0
Ardulink: target power 1
Connection starting
DMStatus After Halt: /0/004c0c82
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadWord Part 1
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadWord Part 1
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadWord Part 1
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadWord Part 1
Error: Option Byte Unlock Failed
Whole-chip erase
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 3
Error: Flash operation error

```

And this is the output of `make` for the blink program:

```
make -C ../../ch32v003fun/../minichlink all
make[1]: Entering directory '/home/mohamed/Downloads/ch32v00-esp32/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/mohamed/Downloads/ch32v00-esp32/ch32v003fun/minichlink'
../../ch32v003fun/../minichlink/minichlink -c /dev/ttyACM0 -w blink.bin flash -b
Opening serial port /dev/ttyACM0 at 115200 baud.
Ardulink: synced.
Found Ardulink Programmer
Interface Setup
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadWord Part 1
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 3
Error: Flash operation error
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 3
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 3
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 1
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
[... 390 lines cut ...]
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 2
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteWord Part 3
Fault writing memory (DMABSTRACTS = 08000402) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadWord Part 1
Image written.

```

cnlohr, 2024-06-24:
When you say that resistor, you mean it's pull-up to 3.3V, right?  If so anything 2.2k-10k should be fine.  Still this is dubious, and would probably take input from the original author, I don't understand how their logic works for halting the processor.

One thing you can try is to manually put the processor in a halt state, when calling minichlink, make the first command before the write a `-a` to halt the chip.

cnlohr, 2025-11-25:
I feel we should just close this out of staleness, there's nothing more that could be done for the current code.