cnlohr/ch32fun #328: Which USB driver are you using for the CH-link?

Rockwell1799, 2024-06-29:
For some reason, I can't get minichlink to work, it's complaining about "bad USB device".I tried winusb and libusb using Zadig,but no luck! I'm using windows 11.

Rockwell1799, 2024-06-30:
I managed to install WinUsb on interface 0 and my CH-Link is now recognized, but when sending the command : minichlink.exe -u to unbrick the device (CH32V003) I get the following message:
Error sending WCH command (on recv): 81 0d 01 0f 09
Can someone help me please?

Rockwell1799, 2024-07-02:
Never mind ! I found a solution by myself!

cnlohr, 2024-07-03:
What was the solution?  

Sorry I was out for a few days?

cnlohr, 2024-07-03:
Was your programmer in ARM mode? Did you need to do the hold-mode-button-and-reboot it?

Rockwell1799, 2024-07-04:
Because my CH32V003 was bricked, I was getting that error, so, I had only to plug the CH-LINKE without the microcontroller connected, execute the command then plug the MCU back.

cnlohr, 2024-07-04:
Ah! Thanks! I am surprised that didn't get fixed by -u in general, but there do seem to be some corner cases.

closed 2024-07-04 by cnlohr