cnlohr/ch32fun #334: Printf over SWIO?

Rockwell1799, 2024-07-04:
Hello
I have problem with Printf using SWIO.Using Printf over Uart is working fine.
I have this line in my funconfig.h:
#define FUNCONF_USE_DEBUGPRINTF 1
The problem, I have nothing coming in my Terminal.
I think I'm missing something! Is there any connection I have to do before? If someone can help, it would be appreciated!
Thank you in advance

eeucalyptus, 2024-07-04:
Did you run the `SystemInit()` before printing? Also have you tried out the "debugprintfdemo" example?

Rockwell1799, 2024-07-04:
Yes, I run SystemInit();
this is my code:
`#include "ch32v003fun.h"

volatile uint32_t count;
int main()
{
    SystemInit();

    // Enable GPIOs
    funGpioInitAll();

    funPinMode( PC0, GPIO_Speed_10MHz | GPIO_CNF_OUT_PP );

    while(1)
    {

        funDigitalWrite( PC0, FUN_HIGH );

        Delay_Ms( 100 );

        funDigitalWrite( PC0, FUN_LOW );

        Delay_Ms( 100 );
        printf("\n %d",count++);
    }
}
`
I haven't checked  "debugprintfdemo" example yet, but I'm going to take a look at it now! 
Thank you

cnlohr, 2024-07-04:
1. What programmer are you using?
2. Let us know when you get a chance to try the debugprintf example.  You can run it with `make clean all monitor` and that will make sure to build from scratch.

Rockwell1799, 2024-07-05:
Hello
1-I have this programmer:
[https://imgur.com/a/SHm2SwM](url)
2-I tried the debugprintf example, but no joy!

maxgerhardt, 2024-07-07:
>I have nothing coming in my Terminal

Just to sanity check you, you open the SWIO (or, SDI) printf monitor by executing `minichlink -T`, yeah?

Rockwell1799, 2024-07-07:
> minichlink -T

Sorry, my bad, I was using an external terminal viewer, but with minchlink it's working fine.

maxgerhardt, 2024-07-07:
And for completeness' sake: If you have a WCH-LinkE, you **can** actually make the SDI printfs show up in the regular COM port, with a special feature from the [wlink](https://github.com/ch32-rs/wlink/releases/tag/nightly) program: Just use `wlink sdi-print enable`. Not sure if it persists across replugging the WCH-LinkE though.

Rockwell1799, 2024-07-07:
Thank you for the heads-up, it's very much appreciated

closed 2024-07-07 by Rockwell1799

wanggaoteng, 2024-07-19:
> And for completeness' sake: If you have a WCH-LinkE, you **can** actually make the SDI printfs show up in the regular COM port and thus your terminal viewer, with a special feature from the [wlink](https://github.com/ch32-rs/wlink/releases/tag/nightly) program: Just use `wlink sdi-print enable`. Not sure if it persists across replugging the WCH-LinkE though.
> 
> And yeah the `-T` or "SDI printf monitor" is quite hidden in the README section (https://github.com/cnlohr/ch32v003fun?tab=readme-ov-file#a-fast-printf-debug-over-the-programming-interface)

@maxgerhardt 
Hi, can you give me some instructions about how to get the "SDI printf monitor" ? I have click the "Enable SDI Printf" button on the left panel of vscode/platformio, and then execute the "minichlink -T" in the terminal, but get:
[image]
 Does my operation is wrong?

cnlohr, 2024-07-19:
@wanggaoteng Please follow the directions here https://github.com/cnlohr/ch32v003fun/wiki/Installation#windows and install the drivers for the wch link e.

cnlohr, 2024-07-19:
I added an extra warning with that link if that event happens to future users. https://github.com/cnlohr/ch32v003fun/commit/50ce2c924b5d3926ecc7a1578c3f6a4cd00e37c3

wanggaoteng, 2024-07-19:
> @wanggaoteng Please follow the directions here https://github.com/cnlohr/ch32v003fun/wiki/Installation#windows and install the drivers for the wch link e.

Thank you @cnlohr

wanggaoteng, 2024-07-19:
> @wanggaoteng Please follow the directions here https://github.com/cnlohr/ch32v003fun/wiki/Installation#windows and install the drivers for the wch link e.

Hi, @cnlohr 
I use zadig to install WinUSB to the WCH-Link interface 0, then, sdi printf works, messages printed in the serial monitor.
But...errors appeared when clicking the upload button of vscode/platformio:
[image]
I'm very frustrated by this error, it makes me cannot upload anything to the chip, can you give me some instructions ?

maxgerhardt, 2024-07-19:
Set```upload_protocol = minichlink```In the platformio.ini.After you changed the drivers to WinUSB/libusb, WCH's openocd version can't speak to the programmer anymore.

image.png (view on web)

wanggaoteng, 2024-07-19:
> upload_protocol = minichlink

Thank you @maxgerhardt 
It works.
But...can you give me some instructions about how to change the driver (now it is winusb) to the old one (namely, the driver before winusb)? 
According to the figure below, there are 3 drivers when clicking the downward triangle in the red circle, such as libusb-win32, libusbk, USB Serial (CDC). I try to change winusb to one of them, hope to get the old one, but the errors appeared when clicking the upload button everytime.
[image]
[image]

cnlohr, 2024-07-19:
@maxgerhardt is there a reasonable way to "guess" which interface to use? 

@wanggaoteng You will need to remove the driver and install the one from WCH.

maxgerhardt, 2024-07-19:
To get back to MounRiver drivers, you can open your Windows Device Manager, use View -> Device by Container, then search for your WCHLink device, then use "Update Drivers" with the target path of `C:\MounRiver\MounRiver_Studio\LinkDrv\WCHLink` (assuming you have a regular MounRiver installation)

[image]

Mhm the interface driver is not that simple.. Since each method has different capabilities as of now
* WCH's OpenOCD version can only interact with the WCH driver for the WCHLinkE and can flash and debug and decode SDI prints **in the style of their SDK**, not ch32v003fun, when nudged to do so via `wlink`
* `minichlink` can only interact with the WinUSB driver for the WCHLinkE  and can flash and decode SDI prints in the style of ch32v003fun. (integration with the debugging capabilities is still not done, have to revisit after https://github.com/cnlohr/ch32v003fun/issues/159 is now done)
* interestingly, [wlink](https://github.com/ch32-rs/wlink/releases/tag/nightly) can interact with the WCHLink in both driver flavors. I don't yet know how they do it.

wanggaoteng, 2024-07-20:
> Set ``` upload_protocol = minichlink ``` In the platformio.ini. After you changed the drivers to WinUSB/libusb, WCH's openocd version can't speak to the programmer anymore.

Thank you @maxgerhardt 
It works.
But...is there anyway to roll back to the driver which used before  WinUSB/libusb ?

wanggaoteng, 2024-07-20:
> To get back to MounRiver drivers, you can open your Windows Device Manager, use View -> Device by Container, then search for your WCHLink device, then use "Update Drivers" with the target path of `C:\MounRiver\MounRiver_Studio\LinkDrv\WCHLink` (assuming you have a regular MounRiver installation)
> 
> ![grafik](https://private-user-images.githubusercontent.com/26485477/350520868-4ee88209-9078-4761-828a-c0b0ad98efd7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE0NTE3NjQsIm5iZiI6MTcyMTQ1MTQ2NCwicGF0aCI6Ii8yNjQ4NTQ3Ny8zNTA1MjA4NjgtNGVlODgyMDktOTA3OC00NzYxLTgyOGEtYzBiMGFkOThlZmQ3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzIwVDA0NTc0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE0MjdhYmIwZWRjOWYyYzNlNDI0YTJlYTVkOWUwMzk5ZmZkY2VjM2M4NzE5YWUwN2Q5N2M4NWQwMDVlMmUyYzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xlcmKU2As44TKcZWvpogaq097C7YzmY4zMdErtUJcKk)
> 
> Mhm the interface driver is not that simple.. Since each method has different capabilities as of now
> 
> * WCH's OpenOCD version can only interact with the WCH driver for the WCHLinkE and can flash and debug and decode SDI prints **in the style of their SDK**, not ch32v003fun, when nudged to do so via `wlink`
> * `minichlink` can only interact with the WinUSB driver for the WCHLinkE  and can flash and decode SDI prints in the style of ch32v003fun. (integration with the debugging capabilities is still not done, have to revisit after [minichlink doesn't support gdb "load" #159](https://github.com/cnlohr/ch32v003fun/issues/159) is now done)
> * interestingly, [wlink](https://github.com/ch32-rs/wlink/releases/tag/nightly) can interact with the WCHLink in both driver flavors. I don't yet know how they do it.

Thank you a lot @maxgerhardt