cnlohr/ch32fun #339: share the detailed steps to use ch32v003fun under vscode/platformio

wanggaoteng, 2024-07-10:
After many attempts, I want to share the steps to use ch32v003fun under vscode/platformio
1. Install git first if you have not install.
2. Open vscode/Platformio, click on Platforms, and in the Installed tab, click on "Advanced Installation". In the dialog box input https://github.com/Community-PIO-CH32V/platform-ch32v.git, Click Install.
3. Create a new ch32v003 project and delete the "system_ch32v00x.c" file in the "C:\Users\xxx\.platformio\packages\framework-wch-noneos-sdk\System\ch32v00x" directory. Otherwise, there will be errors during compilation because of SystemInit() multiple definition.
4. Copy the ch32v003fun.c and ch32v003fun.h files from the "ch32v003fun" directory to the "src" directory which created by platformio.
5. Copy the corresponding example files, take funconfig. h and blink. c under blink folder for illustrate, to the src directory of the platformio project.
6. Compile, everything works very well, and the usage of flash and ram of ch32v003 is really very small.

Thanks to @cnlohr

AlexanderMandera, 2024-07-10:
And opening this repository as a PlatformIO project did not work? There is a platformio.ini in there.

maxgerhardt, 2024-07-10:
(Platform-CH32V maintainer here) This project should work out of the box without preinstalling anything. The `platformio.ini` has `platform = platform = https://github.com/Community-PIO-CH32V/platform-ch32v.git` set, so if you open VSCode and have the PlatformIO extension installed, it should automatically download everything needed. You can then select an environment in the [project environment switcher](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks), such as `blink`, and compile and upload it.

I think what you're trying to do, @wanggaoteng, is to create a new project that has ch32v003fun in it. But you mistakenly select `framework = noneos-sdk`, which compiles in files from `framework-wch-noneos-sdk`. It is **wrong** to modify any of the files inside there. ch32v003fun is a standalone framework, there must be **no** `framework = ...` line in the `platformio.ini` using it. Simply fork the ch32v003fun project and create a new environment in the `platformio.ini` [as documented](https://github.com/cnlohr/ch32v003fun/blob/44959b2ca16e5eb87a041afd89ac2ecccff35e15/platformio.ini#L48-L53), in which you can place all your new code. The description aboves creates some Frankenstein monster of a project, don't do that.

maxgerhardt, 2024-07-10:
I'm gonna take the lead on this issue here and make an addition to `platform-ch32v` so that `framework = ch32v003fun` becomes a selectable framework option in the "Create new project" dialog. It's gonna simplify things and prevents people creating projects in a very wrong way. I'll then PR in some better documentation for "How to create a project based on ch32v003fun", with the "fork this repository and add a new environment" and "create a new `framework = ch32v00fun` project".

cnlohr, 2024-07-10:
@maxgerhardt I am excited for this.  Thank you!

Rockwell1799, 2024-07-11:
I successfully built a project that includes all the Ch32v003fun examples using Vscode with the Platformio extension,but I'm struggling to build only an example at a time .Is there a way to force Vscode to build only  an example from the list of examples already in that folder?

wanggaoteng, 2024-07-11:
> I'm gonna take the lead on this issue here and make an addition to `platform-ch32v` so that `framework = ch32v003fun` becomes a selectable framework option in the "Create new project" dialog. It's gonna simplify things and prevents people creating projects in a very wrong way. I'll then PR in some better documentation for "How to create a project based on ch32v003fun", with both the "fork this repository and add a new environment" and "create a new `framework = ch32v003fun` project" way.

Hi, @maxgerhardt   
I'm very appreciate. Thank you very much.
Best regards.

closed 2024-07-13 by wanggaoteng

maxgerhardt, 2024-07-14:
Please see the projects
* https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/blinky-ch32v003fun
* https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/ch32v003fun-oled
* https://github.com/maxgerhardt/pio-ch32v003fun-fastprimes

it uses the new framework = ch32v003fun option in platform-ch32v.

This is how user's ch32v003fun projects should look like when using PlatformIO. A 4 line platformio.ini. Standard folder layout. Only your code in src/. Github CI activated, giving you the green checkmark. No bullshit.

I've added documentation for this in #349.

mstegen, 2024-07-14:
Thanks for this!
While the build-in uart of the WCH-link works, it would be nice to use the 
`#define FUNCONF_USE_DEBUGPRINTF 1`
option and use the 1wire interface not only for programming, but for debug output as well.
So i can use the v003's uart for other purposes.
Any pointers on how to set that up?

maxgerhardt, 2024-07-14:
This already exists. If you have the latest platform-ch32v version, regardless of `upload_protocol = wch-link/minichlink` etc. you should get a [project task](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) that says "Enable SDI Printf" and it uses "[wlink](https://github.com/ch32-rs/wlink/) sdi-print enable". After that, the regular COM port of the WCH-LinkE should spit out the decoded prints. There is also a "Minichlink Monitor SDI Printf" task to invoke `minichlink -T`, if `upload_protocol = minichlink` is set

That's part of the documentation that I have to get better for https://pio-ch32v.readthedocs.io/. (But it's a new addition anyway)

[image]

wanggaoteng, 2024-07-15:
Thank you for your great work @maxgerhardt 
[image]

mstegen, 2024-07-15:
Testing `upload_protocol= minichlink`
Under Linux the "Minichlink Monitor SDI Printf" works, as long as glibc_2.34 is installed.
On ubuntu 20.04 you will be stuck on an older version, and will only work if you install it manually (i used https://packages.ubuntu.com/jammy/amd64/libc6/download) and then 
```
sudo apt update
sudo apt install libc6
```
On Windows i used Zadig to replace the driver on the WCHlink (interface 0) to WinUSB.
Then the monitoring Printf output also worked fine.

On the other hand i could not get 'Enable SDI Printf' to do anything.
I can see it sends some data (for a second or so) over the SWIO pin on my oscilloscope when i enable the option, but after that second it becomes silent. (the minichlink option showed a constant data stream over the SWIO line)

[image]

I tried wlink and wch-link as upload protocol, both behave the same.

maxgerhardt, 2024-07-15:
* The minichlink binaries come from this repo's CI that uses `ubuntu-latest` (aka ubuntu22); you're right, this is not good for people using the older version. Maybe I'll build for Ubuntu 20 and the binaries will still work for Ubuntu22.
* I'm not 100% sure what `wlink` sends to the WCH-LinkE, but I can imagine that it's very specific to the firmware version running on the WCH-LinkE. They do like to change the USB interface / command packets around as I've seen from @cnlohr on other occasions. Will retest with my WCH-LinkE and the latest firmware version for it (update via MounRiver)

mstegen, 2024-07-16:
It does work if i compile the SDI_printf demo on MounRiver, and then use their WCH-LinkUtility to program the chip.
There was another update for the WCH-LinkE, i'm now on v2.12
Might be a misunderstanding on my part, but is SDI Printf supposed to work with the `ch32v003fun` framework?

maxgerhardt, 2024-07-16:
I think it might just be that the SDI printf in ch32v003fun choses a different data or protocol format than SDI printf in official WCH SDK, so they are incompatible with each other. Will recheck on my side.

mstegen, 2024-07-16:
Yes exactly, that was my conclusion too. 
Then i think the ch32v003 framework should not show the SDI enable/disable options, only minichlink.

cnlohr, 2024-07-16:
Do we know what their SDI protocol is?  As long as it allows bidirectional communication and is as fast, I wouldn't mind swapping.

mstegen, 2024-07-17:
Ref the speed, i found this https://www.wch.cn/bbs/thread-107881-1.html
Slightly faster then 115200 bps, so no where near the minichlink debug printf speed.

maxgerhardt, 2024-07-17:
WCH's format is at 

https://github.com/openwch/ch32v003/blob/9d875e937885569116e2887246ac9f1897ee39b0/EVT/EXAM/SRC/Debug/debug.c#L153-L218

while ch32v003's is at

https://github.com/cnlohr/ch32v003fun/blob/96cd3d2e6e34ddcc4654431f2ce6777442485614/ch32v003fun/ch32v003fun.c#L1395-L1471

They do so seem very different and WCH's way doesn't even enable input.

cnlohr, 2024-07-17:
[image]

cnlohr, 2024-08-01:
Any chance we can just use mine?  Theirs literally doesn't support input and is slower.

maxgerhardt, 2024-08-01:
Better docs for SWI Printf under PIO are on the way. With WCH SDK's SWI printf lacking, I would not advocate to use theirs. If anything, I would advicate for **them** to use ch32v003fun's way.