cnlohr/ch32fun #343: Non-amalgamated code?

niansa, 2024-07-13:
I see https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.c is amalgamated. This makes the file incredibly annoying to read and hard to understand.
Where do I find the original files this has been amalgamated into?

cnlohr, 2024-07-13:
This file ... is ... the code.  It's not amalgamated. Or at least, we have been working on it this way since we began. That said, I am making a PR to add some explaination.

https://github.com/cnlohr/ch32v003fun/compare/cleanup-asm-and-comments

That said, the file is less than 2k lines.

niansa, 2024-07-13:
Oh. I apologize.

I think there should be way shorter C files and with different files for different implementations (for different devices for example) of the same thing.
This helps with readability and with preventing merge conflicts.
I'm considering to make a pull request with the suggested changes, but I do not want to interfere with your or any other pull requests.

cnlohr, 2024-07-13:
I will merge my changes in soon.

We do not want more files for the core.  This decision was made to minimize complexity of the build system, allowing for manual compilation, or unusual bootloader compilation where appropriate.

Modern git's merging is no less effective on single large files than smaller more granular files, so for our purposes, one single file needed for C compilation is desirable.  You can feel free to create pr's against subsections of files.

niansa, 2024-07-13:
> allowing for manual compilation

I think in that case it'd probably be cleaner to just have one "main" file that `#include`s all the files. Or to just do *actual* amalgamation.

Don't get me wrong, the code is quite clean. For me it just makes attempting to find specific things in the code a lot of ctrl+f and scrolling. This is something I personally feel pretty dirty about having in my own code bases. And I think it makes it harder for people to find bugs and contribute as well.

> feel free to create pr's against subsections of files

Thanks.

dwillmore, 2024-07-13:
I'm not sure you're going to find much traction with a proposal to split the code up into multiple files.  The whole premise of this repo is that it's easy to use as just #including a file.  Once you drop that, the complexity of using it goes way up and defeats the purpose of the one file strategy.

So, such a proposal is basically saying "stop doing the most useful thing you're doing".  And I don't see how that ends well.

closed 2024-07-13 by niansa