cnlohr/ch32fun #348: Compile failure for CH32V303CBT6

maxgerhardt, 2024-07-13:
Not sure exactly how this should work, but I'm getting

```
In file included from src\blink.c:1:
C:\Users\Max\.platformio\packages\framework-ch32v003fun\ch32v003fun/ch32v003fun.h:12942:6: error: #error "Invalid PLL multiplier"
     #error "Invalid PLL multiplier"
      ^~~~~
```

My CH32V303CBT6 board definition has `-DCH32V30x -DCH32V303 -DCH32V30x_D8` defined (globally). My source for that is [the datasheet](https://github.com/Community-PIO-CH32V/platform-ch32v/blob/31cdd721af55ca35cc0787440fcda610471d4a9d/misc/scripts/gen_boarddefs.py#L16-L29).

Then

https://github.com/cnlohr/ch32v003fun/blob/49640fbccf231191aa83c6a2bbe9d385535b2d1e/ch32v003fun/ch32v003fun.h#L168-L182

is going to activate

```
#define FUNCONF_PLL_MULTIPLIER 18	// Default: 8 * 18 = 144 MHz
```

but 

https://github.com/cnlohr/ch32v003fun/blob/49640fbccf231191aa83c6a2bbe9d385535b2d1e/ch32v003fun/ch32v003fun.h#L12918-L12953

will run into the `#error` because the `#elif defined(CH32V20x) && FUNCONF_PLL_MULTIPLIER == 18` does not hold for it. 

Is that a bug in my activated macros for the CH32V303CBT6 or does ch32v003fun incorrectly account for it?

cnlohr, 2024-07-14:
I regret I don't have a CH32V303 here of any kind -  I would expect it should use FUNCONF_PLL_MULTIPLIER 18

I am 99% sure we should just remove the CH32V20x precondition from line 12949.

cnlohr, 2024-07-14:
Can you remove and test?  If so, can you make a PR?

cnlohr, 2024-07-14:
Blame falls on @AlexanderMandera  -- this line looks like an actual error. But the plot is so messy...
[image]
Just the way to get there is quite messier than most.

closed 2025-11-25 by cnlohr