cnlohr/ch32fun #358: Compilation failure using gcc 13.2.0 (latest release) vs gcc 9.3.0

brian-smith-github, 2024-07-21:
Hi again.
Can you help? I've switched from CH32V003 to CH32V103 board, but have noticed that although using gcc 9.3.0 (the current default in Ubuntu) works fine, I get compiler errors when using gcc 13.2.0 (the latest release):

../../ch32v003fun/ch32v003fun.c:1274:9: error: impossible constraint in 'asm'
 1274 |         asm volatile(
      |         ^

The offending lines in ch32v003fun.c seem to be:

"	la t0, InterruptVector\n\
	ori t0, t0, 3\n\
	csrw mtvec, t0\n"
	: : "InterruptVector" (InterruptVector) : "t0", "memory"

If i comment these lines out, the compile is fine (testing with examples_v10x/blink.c).
The CH32V003  examples/blink/ compiles and works fine, only the examples_v*/blink examples break.

TommyMurphyTM1234, 2024-07-25:
Can you provide the build log? 

> using gcc 13.2.0 (the latest release)

GCC 14.1 is actually the latest release:

- https://gcc.gnu.org/gcc-14/

Although, if you're using `riscv-gnu-toolchain` or some derivative then it hasn't updated to 14 yet: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1472

brian-smith-github, 2024-07-25:
I didn't build from source, I am just using the latest binary release from:
https://github.com/riscv-collab/riscv-gnu-toolchain/releases

Given that the error relates to interrupts, I am suspicious that the higher version of the compiler is revealing an underlying bug in the interrupt-handling code as I'm unable to get a simple timer-based interrupt working on the CH32V103 board right now.

TommyMurphyTM1234, 2024-07-25:
Build log?

TommyMurphyTM1234, 2024-07-25:
> Hi again. Can you help? I've switched from CH32V003 to CH32V103 board, but have noticed that although using gcc 9.3.0 (the current default in Ubuntu) works fine, I get compiler errors when using gcc 13.2.0 (the latest release):
> 
> ../../ch32v003fun/ch32v003fun.c:1274:9: error: impossible constraint in 'asm' 1274 | asm volatile( | ^
> 
> The offending line in ch32v003fun.c seem to be (line 1292) :
> 
> ```
>   : : "InterruptVector" (InterruptVector) : "t0", "memory"
> ```

Why do you point the finger at line 1292 when the error message mentions line 1274?

brian-smith-github, 2024-07-25:
line 1274 is just the opening section of a multi-line block of assembly code:

asm volatile(

Line 1292 is the bit of code that when commented out allows the compile to complete without errors, hence why I'm thinking that is where the error is (something to do with macro processing I think, but I can't get my head around how the compiler processes macros)

TommyMurphyTM1234, 2024-07-25:
Your lounge 1292 doesn't match the latest in source control: 

- https://github.com/cnlohr/ch32v003fun/blob/8dc0cb729e807489346ca04ec149d977d802bca8/ch32v003fun/ch32v003fun.c#L1291-L1294

You still haven't provided a build log.

brian-smith-github, 2024-07-25:
It's now line 1297, I was referencing an older version of the file I think... the bug is still there though. I'll try my interrupt test code on the latest ch32v003fun and see if it's any better though.

brian-smith-github, 2024-07-25:
My timer-generates-interrupt test program behaves so strangely that I've no hope of tracing the problem so I'll just close this and put my 103-board in a drawer I think... :)

closed 2024-07-25 by brian-smith-github

cnlohr, 2024-07-25:
@brian-smith-github what happens if you use one of our existing demo programs?

Charles

brian-smith-github, 2024-07-26:
It looks like there are 2 examples for CH32V003 that use a timer to generate an interrupt:
1) The simplest is systick_irq - This won't work on CH32V103 as the 'SysTick' data structure is different between boards and is missing the 'Service Request' (SR) variable for some reason, so won't compile (commenting out references to SR doesn't work either, no ISR seems to happen at all).
2) There is also adc_fixed_fs but again this doesn't compile for the 103 board because of different data structures.

Really though, don't worry about it, I've got a 'Arduino Nano Connect' board I'm going to use instead for now.

brian-smith-github, 2024-07-26:
(I was hoping to use the CH32V103 board as a bridge towards moving to the new CH32 boards, I'll just wait until they are actually released now instead).