cnlohr/ch32fun #359: Plugdev group on Archlinux?

Rockwell1799, 2024-07-23:
I'm on Archlinux and I have problem running minichlink without sudo privilege.
I already added my username to the group using the following command:

> sudo groupadd plugdev
> sudo usermod -aG plugdev MY_USERNAME

The problem when I run ./minichlink, I get the following error:
sudo groupadd plugdev

> Error: couldn't open wch link device (libusb_open() = -3)
> Error: Could not initialize any supported programmers
> Error: Could not initialize any supported programmers

Any help would be appreciated!
Thank you in advance

prosper00, 2024-07-23:
> [Tracking image][https://github.com/notifications/beacon/ANVXEPO3MFQNSYBBAHY6T5TZNZXWNA5CNFSM6AAAAABLKU64RWWGG33NNVSW45C7OR4XAZNFJFZXG5LFVJRW63LNMVXHIX3JMTHJBEOR4U.gif]

did you add the udev rules too? and then reboot or restart udev via udevadm?

Rockwell1799, 2024-07-23:
> sudo cp minichlink/99-minichlink.rules /etc/udev/rules.d/
> sudo udevadm control --reload-rules && sudo udevadm trigger

> […](#)

Sorry! It's my bad, I forgot to add the udev rules!
After executing the following command, everything is working fine

closed 2024-07-23 by cnlohr