cnlohr/ch32fun #371: Minichlink stuck in an infinite loop with WCH-LinkE and some CH32V003

mverdon, 2024-08-09:
I've just received a new batch of custom PCBs with 003s on them and I've been trying to probe each of them with my WCH-LinkE WCH-LinkE-R0-1v3 with firmware 2.12. I always encounter the same issue when running `minichlink -i` never exits but probing my old batch works just fine.

The difference I see just by looking at the chips is that the batch/date code differs on the packages. It shows `WCH V003F4 02D13` on the old ones and `WCH V003F4 12E22` on the new ones. These are supposed to be the same part number CH32V003F4U6. It could be nothing but may explain the difference in behaviour.

I've been trying to debug the issue and it looks like the problem starts when reading the register `DMHARTINFO` in `StaticUpdatePROGBUFRegs()`. The correct value `0x002120f4` is read on the old batch but somehow the new batch always returns `0xffffffff`. After that, since it cannot retrieve the correct offset for DATA0, the whole process fails then it gets stuck in `WaitForDoneOp()` where reading `DMABSTRACTCS` always gives `0xffffffff`.

Here are the logs for the old batch where it correctly probes the chip:
```
Found WCH Link
WCH Programmer is LinkE version 2.12
Full Chip Type Reply: [8] 82-0d-05-09-00-31-05-00
Detected: CH32V003
LEWriteReg32 10, 80000001
LEWriteReg32 10, 80000001
LEWriteReg32 10, 80000003
LEWriteReg32 10, 80000001
LEWriteReg32 16, 00000700
LEWriteReg32 18, 00000000
LEWriteReg32 17, 00221000
LEReadReg32 16, 08000002
Setup success
Flash Storage: 16 kB
Part UUID    : 29-0f-ab-cd-90-fc-bc-20
PFlags       : ff-ff-ff-ff
Part Type (B): 07-13-bb-91
Read protection: disabled
Interface Setup
LEWriteReg32 7e, 5aa50400
LEWriteReg32 7d, 5aa50400
LEWriteReg32 7d, 5aa50400
LEWriteReg32 10, 80000001
LEWriteReg32 10, 80000001
LEWriteReg32 18, 00000000
LEWriteReg32 20, 40044180
LEWriteReg32 21, c1040411
LEWriteReg32 22, 9002c180
LEReadReg32 12, 002120f4
LEWriteReg32 04, e00000f4
LEWriteReg32 17, 0023100a
LEWriteReg32 04, e00000f8
LEWriteReg32 17, 0023100b
LEWriteReg32 04, 40022010
LEWriteReg32 17, 0023100c
LEWriteReg32 04, 00050000
LEWriteReg32 17, 0023100d
LEWriteReg32 18, 00000001
LEWriteReg32 05, 1ffff800
LEWriteReg32 17, 00241000
LEReadReg32 16, 08000002
LEReadReg32 04, e8175aa5
USER/RDPR  : e817/5aa5
LEReadReg32 04, ff00ff00
DATA1/DATA0: ff00/ff00
LEReadReg32 04, 00ff00ff
WRPR1/WRPR0: 00ff/00ff
LEReadReg32 04, 00ff00ff
WRPR3/WRPR2: 00ff/00ff
LEWriteReg32 05, 1ffff7e0
LEWriteReg32 17, 00241000
LEReadReg32 16, 08000002
LEReadReg32 04, ffff0010
Flash Size: 16 kB
LEWriteReg32 05, 1ffff7e8
LEWriteReg32 17, 00241000
LEReadReg32 16, 08000002
LEReadReg32 04, 290fabcd
R32_ESIG_UNIID1: 290fabcd
LEReadReg32 04, 90fcbc20
R32_ESIG_UNIID2: 90fcbc20
LEReadReg32 04, ffffffff
R32_ESIG_UNIID3: ffffffff
```

And here are the logs for the new batch:
```
Found WCH Link
WCH Programmer is LinkE version 2.12
Full Chip Type Reply: [8] 82-0d-05-09-00-31-05-10
Detected: CH32V003
LEWriteReg32 10, 80000001
LEWriteReg32 10, 80000001
LEWriteReg32 10, 80000003
LEWriteReg32 10, 80000001
LEWriteReg32 16, 00000700
LEWriteReg32 18, 00000000
LEWriteReg32 17, 00221000
LEReadReg32 16, 08000002
Setup success
Flash Storage: 16 kB
Part UUID    : 09-8b-ab-cd-72-48-bc-f0
PFlags       : ff-ff-ff-ff
Part Type (B): 07-13-bb-91
Read protection: disabled
Interface Setup
LEWriteReg32 7e, 5aa50400
LEWriteReg32 7d, 5aa50400
LEWriteReg32 7d, 5aa50400
LEWriteReg32 10, 80000001
LEWriteReg32 10, 80000001
LEWriteReg32 18, 00000000
LEWriteReg32 20, 40044180
LEWriteReg32 21, c1040411
LEWriteReg32 22, 9002c180
LEReadReg32 12, ffffffff
LEWriteReg32 04, e00007ff
LEWriteReg32 17, 0023100a
LEWriteReg32 04, e0000803
LEWriteReg32 17, 0023100b
LEWriteReg32 04, 40022010
LEWriteReg32 17, 0023100c
LEWriteReg32 04, 00050000
LEWriteReg32 17, 0023100d
LEWriteReg32 18, 00000001
LEWriteReg32 05, 1ffff800
LEWriteReg32 17, 00241000
LEReadReg32 16, ffffffff
LEReadReg32 16, ffffffff
LEReadReg32 16, ffffffff
LEReadReg32 16, ffffffff
LEReadReg32 16, ffffffff
... x infinity
```

Overriding the value read from `DMHARTINFO` and `DMABSTRACTCS` with the correct ones doesn't seem to help, all reads eventually return `0xffffffff`.

I don't know if this is due to a memory configuration issue or an issue in the read/write sequence leading to that problem but this is specific to minichlink as I can probe and flash the new chips just fine with the official [WCHISPTool](https://wch-ic.com/downloads/WCHISPTool_Setup_exe.html) and even with [wlink](https://github.com/ch32-rs/wlink).

Has anyone else encountered the same issue?

mverdon, 2024-08-09:
I found the issue. Looks like the double write to `DMCFGR` is no longer required. Removing the second line commented with "Bug in silicon?" fixes the issue for both silicon revisions I have `02D13` and  `12E22`.

cnlohr, 2024-08-15:
I am not sure, @mverdon - can you create a PR, then I can play with it?

mverdon, 2024-08-23:
Thanks @cnlohr for the feedback, I'm closing this issue as the PR fixed it.

closed 2024-08-23 by mverdon