cnlohr/ch32fun #373: Allow use of debug printing while UART printing is enabled

niansa, 2024-08-10:
Currently you can only set either UARTPRINTF or DEBUGPRINTF. The case where UART is still needed to communicate with a peripheral but the developer wants to use debug printing is not accounted for at all: The entire debug printing function is not compiled at all if DEBUGPRINTF isn't set requiring the user to copy the entire function into their own codebase.

Would be great if there were separate functions for UART and DEBUG printing that aren't tightly coupled in into stdio. Stdio could then simply interface with those functions.

cnlohr, 2024-08-16:
I don't understand, if you just use the debug printing with `FUNCONF_USE_DEBUGPRINTF`, it printf's over the programming interface.  You are free to configure the UART to do whatever you wish at that point.

niansa, 2024-08-16:
My idea was that then I (the user) wouldn't be able to use the routines provided by ch32v003fun anymore and would have to copy them into their own code.
Now the obvious solution for me has been to just use an external UART library instead that has more functionality.

closed 2024-08-16 by niansa