cnlohr/ch32fun #374: Allow user to disable implementation of libc functions

niansa, 2024-08-11:
The user might want to load their own libc implementation including functions like `memset` and `printf`. Currently this is impossible because ch32v003fun already defines these functions.
There should be an option to not define these function or at least to define them with alternative names.

This would also require UART and DEBUG printing to be decoupled from stdio so the user would still be able to use those. See: #373

TommyMurphyTM1234, 2024-08-11:
I think that this could be facilitated by declaring the relevant functions as having `weak` linkage.

- https://gcc.gnu.org/onlinedocs/gcc-4.1.2/gcc/Function-Attributes.html#:~:text=on%20line%205.-,weak,-The%20weak%20attribute

E.g.:

- https://github.com/cnlohr/ch32v003fun/blob/fd615a141136fedb2c0b0d4e3a2b261c12d60ae9/ch32v003fun/ch32v003fun.c#L216

becomes:

```
__attribute__((weak)) void *memset(void *dest, int c, size_t n) { unsigned char *s = dest; for (; n; n--, s++) *s = c; return dest; }
```
I did a quick test using the Blink example:

- https://github.com/cnlohr/ch32v003fun/tree/master/examples/blink

If I change `blink.c` to:

```
#include "ch32v003fun.h"
#include <stdio.h>

void *memset(void *dest, int c, size_t n)
{
}

int main()
{
	SystemInit();

	// Enable GPIOs
	funGpioInitAll();
	
	funPinMode( PD0, GPIO_Speed_10MHz | GPIO_CNF_OUT_PP );
	funPinMode( PD4, GPIO_Speed_10MHz | GPIO_CNF_OUT_PP );
	funPinMode( PD6, GPIO_Speed_10MHz | GPIO_CNF_OUT_PP );
	funPinMode( PC0, GPIO_Speed_10MHz | GPIO_CNF_OUT_PP );

	while(1)
	{
		funDigitalWrite( PD0, FUN_HIGH );
		funDigitalWrite( PD4, FUN_HIGH );
		funDigitalWrite( PD6, FUN_HIGH );
		funDigitalWrite( PC0, FUN_HIGH );
		Delay_Ms( 250 );
		funDigitalWrite( PD0, FUN_LOW );
		funDigitalWrite( PD4, FUN_LOW );
		funDigitalWrite( PD6, FUN_LOW );
		funDigitalWrite( PC0, FUN_LOW );
		Delay_Ms( 250 );
	}
}
```
and when I try to compile it I get this:
```
make
riscv64-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32V003 -DMCU_PACKAGE= -DTARGET_MCU_LD=0 ../../ch32v003fun/ch32v003fun.ld > ../../ch32v003fun/generated_ch32v003.ld
riscv64-unknown-elf-gcc -o blink.elf ../../ch32v003fun/ch32v003fun.c blink.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -L../../ch32v003fun/../misc -lgcc -T ../../ch32v003fun/generated_ch32v003.ld -Wl,--gc-sections
/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccqnn7Y8.o (symbol from plugin): in function `memset':
(.text+0x0): multiple definition of `memset'; /tmp/ccLGGDcN.o (symbol from plugin):(.text+0x0): first defined here
/home/user/Downloads/riscv/riscv64-unknown-elf/include/string.h:33:10: error: 'memset' has already been defined
   33 | void *   memset (void *, int, size_t);
      |          ^
/home/user/Downloads/riscv/riscv64-unknown-elf/include/string.h:33:10: note: previously defined here
lto1: fatal error: errors during merging of translation units
compilation terminated.
lto-wrapper: fatal error: riscv64-unknown-elf-gcc returned 1 exit status
compilation terminated.
/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [../../ch32v003fun/ch32v003fun.mk:220: blink.elf] Error 1
```
But if I make the change at the top of my post then it works:
```
make
riscv64-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32V003 -DMCU_PACKAGE= -DTARGET_MCU_LD=0 ../../ch32v003fun/ch32v003fun.ld > ../../ch32v003fun/generated_ch32v003.ld
riscv64-unknown-elf-gcc -o blink.elf ../../ch32v003fun/ch32v003fun.c blink.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -L../../ch32v003fun/../misc -lgcc -T ../../ch32v003fun/generated_ch32v003.ld -Wl,--gc-sections
riscv64-unknown-elf-size blink.elf
   text    data     bss     dec     hex filename
    564       0       0     564     234 blink.elf
riscv64-unknown-elf-objdump -S blink.elf > blink.lst
riscv64-unknown-elf-objdump -t blink.elf > blink.map
riscv64-unknown-elf-objcopy -O binary blink.elf blink.bin
riscv64-unknown-elf-objcopy -O ihex blink.elf blink.hex
make -C ../../ch32v003fun/../minichlink all
make[1]: Entering directory '/home/user/Downloads/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/user/Downloads/ch32v003fun/minichlink'
../../ch32v003fun/../minichlink/minichlink -w blink.bin flash -b
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
make: *** [../../ch32v003fun/ch32v003fun.mk:224: cv_flash] Error 224
```
I don't know if there are any pitfalls to doing this but I don't think so.

I can prepare a strawman PR if you/@cnlohr like?

TommyMurphyTM1234, 2024-08-12:
Actually maybe there are two issues here:

1. **DIsabling** the use of standard library functions. As far as I know this should happen automatically already - if certain features are not used then the "fine-grained" linking implemented by `-fdata-sections -ffunction-sections` at compile time and `-gc-sections` at link time should eliminate them from the linked executable/ELF file.

- https://github.com/cnlohr/ch32v003fun/pull/273

2. **Overriding** the default implementations of standard library functions with user provided alternatives. As far as I know/can see the use of `weak` linkage here should allow for this (subject to further testing and verification).

niansa, 2024-08-12:
Weak linkage sounds like an awesome solution to me!

cnlohr, 2024-08-15:
I would be happy to include a `FUNCONF_DISABLE_LIBC` if you guys think it's particularly useful.  Just someone else comment back.

TommyMurphyTM1234, 2024-08-15:
> I would be happy to include a `FUNCONF_DISABLE_LIBC` if you guys think it's particularly useful. Just someone else comment back.

I'm not really sure how exactly that would work but it sounds more like an "all or nothing" approach? I.e. ch32v003fun implementations of standard library (and system call?) functions are all available by default but not at all available if `FUNCONF_DISABLE_LIBC` is defined? 

Whereas, I think, the original issue is more about having the option to **selectively** override certain standard library/system call functions with user implementations where the makes sense for the particular use case. That's why I floated the idea of making some or all ch32v003fun implementations of such standard library/system call functions weakly linked. 

Are there specific arguments against just using `weak` linkage for such functions?

cnlohr, 2024-08-16:
Ah, that's fair.  Yeah, I would accept a PR with `weak` linkages.

niansa, 2024-08-16:
I would create it this weekend 🙂

niansa, 2024-08-17:
#379

closed 2024-08-26 by niansa