cnlohr/ch32fun #378: [Report] Wiki-Arch toolchain

StewartBie, 2024-08-17:
Refer to Wiki - Installation - Arch/Manjaro
the AUR package `riscv64-unknown-elf-newlib` was last updated 	2020-03-23 20:29 (UTC)  so there are many obstacles to installing this AUR pac.
Here are some errors when installing
```
../../../../../../../newlib-3.3.0/newlib/libc/reent/closer.c:47:14: error: implicit declaration of function '_close'; did you mean 'close'? [-Wimplicit-function-declaration]
   47 |   if ((ret = _close (fd)) == -1 && errno != 0)
      |              ^~~~~~
      |              close
make[8]: *** [Makefile:439: lib_a-closer.o] Error 1
......
../../../../../../../newlib-3.3.0/newlib/libc/syscalls/sysisatty.c:12:10: error: implicit declaration of function '_isatty'; did you mean 'isatty'? [-Wimplicit-function-declaration]
   12 |   return _isatty (fd);
      |          ^~~~~~~
      |          isatty
make[8]: *** [Makefile:416: lib_a-sysisatty.o] Error 1
make[8]: Leaving directory '/home/ze/.cache/yay/riscv64-unknown-elf-newlib/src/build-newlib/riscv64-unknown-elf/rv32i/ilp32/newlib/libc/syscalls'
Making all in ssp
```
I wonder whether these package can replace.(WCH MRS toolchain)
[image]

TheComputerGuy96, 2024-08-17:
The attached image file is corrupted (according to pngcheck) :frog:

StewartBie, 2024-08-17:
my bad. I'll try to show it in the table below.
|                    Name                    | Version | Votes | Popularity? |                                                      Description                                                       | Maintainer |      Last Updated      |
| ------------------------------------------ | ------- | ----- | ----------- | ---------------------------------------------------------------------------------------------------------------------- | ---------- | ---------------------- |
| mounriver-studio-toolchain-bin             | 1.91-0  | 4     | 0.00        | This MRS Toolchain includes the tool chain for RISC-V kernel chip under Linux x64 and the debug download tool OpenOCD. | taotieren  | 2024-04-21 09:17 (UTC) |
| mounriver-studio-toolchain-openocd-bin     | 1.91-0  | 4     | 0.00        | MRS Toolchain OpenOCD supports erasure, programming, verification and debugging of the chip.                           | taotieren  | 2024-04-21 09:17 (UTC) |
| mounriver-studio-toolchain-riscv-gcc-bin   | 1.91-0  | 4     | 0.00        | MRS Toolchain Support for RISC-V assembly and GNU C compilation, link operation.                                       | taotieren  | 2024-04-21 09:17 (UTC) |
| mounriver-studio-toolchain-riscv-gcc12-bin | 1.91-0  | 4     | 0.00        | MRS Toolchain Support for RISC-V assembly and GNU C compilation, link operation.                                       | taotieren  | 2024-04-21 09:17 (UTC) |

This package dedends on the three others [mounriver-studio-toolchain-bin](https://aur.archlinux.org/packages/mounriver-studio-toolchain-bin)
Thanks for your reply.   ^.^

TommyMurphyTM1234, 2024-08-25:
I'm not sure that I fully understand the original issue but if the problem is getting a working (and maybe more up to date?) RISC-V bare metal toolchain for use on Arch Linux then would the xPack binary toolchain builds be of any use?

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/
  - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/releases/

The only dependency that the Linux builds should have is on a minimum GLIBC version (>= 2.27 right now):

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/blog/2024/08/07/riscv-none-elf-gcc-v14-2-0-1-released/#prerequisites

Bear in mind that the xPack toolchain uses a different tool prefix string than the [`riscv-gnu-toolchain`](https://github.com/riscv-collab/riscv-gnu-toolchain) one - `riscv-none-elf-*` versus `riscv64-unknown-elf-*` or `riscv32-unknown-elf-*`:

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/user/#riscv-none-elf-gcc

TommyMurphyTM1234, 2024-08-26:
FWIW I did a quick install of a VirtualBox VM running Arch Linux 2024.08.01, downloaded and extracted the latest xPack RISC-V GCC (14.2.0-1) and using it I was able to compile and link a simple "Hello world" program.

StewartBie, 2024-08-27:
Thanks for all your replies. My problems are solved.
1. I know that xpack is another standalone toolchain and it works
2. There is a better choice on the official repos `riscv64-elf-binutils riscv64-elf-gcc riscv64-elf-gdb riscv64-elf-newlib`
3. I'll keep in mind all your seggestions. ^.^

I think it's time to close this issues. Thanks a lot.

closed 2024-08-27 by StewartBie