cnlohr/ch32fun #383: ADC polling example problem!

Rockwell1799, 2024-08-27:
Hello
I tried to modify the code for ADC using polling method on a CH32V003, but I get stuck in this line:
    while(!(ADC1->STATR & ADC_EOC));
This is my code :

>#include "ch32v003fun.h"

void adc_init(void);
uint16_t adc_get(void);
volatile uint32_t count;

int main ()

{
 SystemInit();// system clock config, it can work without, but you'll get wrong frequency
 adc_init();
 RCC->APB2PCENR |=RCC_IOPCEN ;// enable GPIOC FOR LED
 GPIOC->CFGLR =0x44444441;//PC0 as output 10Mhz, the rest are kept on the reset value which is 0x4444444

while(1)
{
    GPIOC->BSHR |=(1<<0);
    Delay_Ms(1000);
    GPIOC->BSHR |=(1<<16);
    Delay_Ms(1000);

    printf( "Count: %d adc: %d\n\r", count++, adc_get() );
}

}

void adc_init(void)
{
    RCC->APB2PRSTR |= RCC_APB2Periph_ADC1;

    RCC->CFGR0 |=(0b01000<<11);//FOR 3.2V MAX ADC FREQ IS 12MZ ADCPRES[11:15] 010XX
    RCC->APB2PCENR |=RCC_IOPAEN;// enable GPIOA PA1
    GPIOA->CFGLR =0X44444404;//PA1 AS ANALOG INPUT
    

    // Set up single conversion on chl 0
    ADC1->RSQR1 = 0;
    ADC1->RSQR2 = 0;
    ADC1->RSQR3 = 0;    // 0-9 for 8 ext inputs and two internals
    ADC1->SAMPTR2 |=6; // 241 CYCLES

    ADC1->CTLR2 |= ADC_ADON | ADC_EXTSEL;// turn on ADC and set rule group to sw trig
//-------------------------
    // Reset calibration

}
uint16_t adc_get( void )
{
    // start sw conversion (auto clears)
    ADC1->CTLR2 |= ADC_SWSTART;

    // wait for conversion complete
    while(!(ADC1->STATR & ADC_EOC));

    // get result
    return ADC1->RDATAR;
}
I have no idea, why my code is not working! Can someone help me please?

Rockwell1799, 2024-08-29:
`RCC->APB2PRSTR |= RCC_APB2Periph_ADC1;` is wrong. It should be `RCC->APB2PCENR |=RCC_ADC1EN; `

Rockwell1799, 2024-08-29:
The issue is closed.

cnlohr, 2024-08-29:
`RCC_ADC1EN` and `RCC_APB2Periph_ADC1` are both 0x200. and it is using `RCC->APB2PCENR` https://github.com/cnlohr/ch32v003fun/blob/master/examples/adc_polled/adc_polled.c#L18C2-L18C16

cnlohr, 2024-08-29:
If there's an issue, I'd really like to fix the example.

Rockwell1799, 2024-08-29:
> If there's an issue, I'd really like to fix the example.

The examples/adc_polled/adc_polled.c line 18 must be corrected to:
` RCC->APB2PCENR |= RCC_APB2Periph_GPIOD | RCC_ADC1EN;`

emeb, 2024-08-29:
(original author here) I agree that RCC_ADC1EN is preferred given that's how the register is defined in the .h file. It may resolve to the same thing as RCC_APB2Periph_ADC1 for the V003 but might change for other targets. I'll test and submit a PR shortly.

Rockwell1799, 2024-08-29:
> (original author here) I agree that RCC_ADC1EN is preferred given that's how the register is defined in the .h file. It may resolve to the same thing as RCC_APB2Periph_ADC1 for the V003 but might change for other targets. I'll test and submit a PR shortly.
> 
> --edit--
> On further reflection I'm ambivalent about this change. As cnlohr noted above, the values of RCC_ADC1EN and RCC_APB2Periph_ADC1 are identical and if we were going to use RCC_ADC1EN in the context of of RCC->APB2PCENR then we should also be using RCC_IPOxEN for GPIO clock enables when we touch that register and that would require changing a *lot* of code.
> 
> I'm not convinced that the change from RCC_APB2Periph_ADC1 to RCC_ADC1EN is the root cause of the fix noted above.

I'm using a CH32v003 chip and when I use RCC_APB2Periph_ADC1, the ADC clock is not enabled for some reason.

cnlohr, 2024-08-31:
Can you post the binary (and .lst) file from both of your attempts, @Rockwell1799 ?

Rockwell1799, 2024-09-01:
> Can you post the binary (and .lst) file from both of your attempts, @Rockwell1799 ?
.lst file with RCC->APB2PCENR |= RCC_APB2Periph_ADC1;

```
ADC_POLLING.elf:     file format elf32-littleriscv
ADC_POLLING.elf
architecture: riscv:rv32, flags 0x00000112:
EXEC_P, HAS_SYMS, D_PAGED
start address 0x000000a0

Program Header:
    LOAD off    0x00001000 vaddr 0x00000000 paddr 0x00000000 align 2**12
         filesz 0x0000071c memsz 0x0000071c flags r-x
    LOAD off    0x00002000 vaddr 0x20000000 paddr 0x0000071c align 2**12
         filesz 0x00000000 memsz 0x00000004 flags rw-
    LOAD off    0x00002700 vaddr 0x20000700 paddr 0x20000700 align 2**12
         filesz 0x00000000 memsz 0x00000100 flags rw-

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .init         000000a0  00000000  00000000  00001000  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .text         0000067c  000000a0  000000a0  000010a0  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  2 .fini         00000000  0000071c  0000071c  0000171c  2**0
                  CONTENTS, ALLOC, LOAD, CODE
  3 .dalign       00000000  20000000  20000000  0000171c  2**0
                  CONTENTS
  4 .dlalign      00000000  0000071c  0000071c  0000171c  2**0
                  CONTENTS
  5 .data         00000000  20000000  20000000  0000171c  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  6 .bss          00000004  20000000  0000071c  00002000  2**2
                  ALLOC
  7 .stack        00000100  20000700  20000700  00002700  2**0
                  ALLOC
  8 .debug_info   00002602  00000000  00000000  0000171c  2**0
                  CONTENTS, READONLY, DEBUGGING
  9 .debug_abbrev 0000072b  00000000  00000000  00003d1e  2**0
                  CONTENTS, READONLY, DEBUGGING
 10 .debug_loc    00001fa2  00000000  00000000  00004449  2**0
                  CONTENTS, READONLY, DEBUGGING
 11 .debug_aranges 00000170  00000000  00000000  000063eb  2**0
                  CONTENTS, READONLY, DEBUGGING
 12 .debug_ranges 000002a0  00000000  00000000  0000655b  2**0
                  CONTENTS, READONLY, DEBUGGING
 13 .debug_line   00002592  00000000  00000000  000067fb  2**0
                  CONTENTS, READONLY, DEBUGGING
 14 .debug_str    00000a02  00000000  00000000  00008d8d  2**0
                  CONTENTS, READONLY, DEBUGGING
 15 .comment      00000033  00000000  00000000  0000978f  2**0
                  CONTENTS, READONLY
 16 .debug_frame  000003c8  00000000  00000000  000097c4  2**2
                  CONTENTS, READONLY, DEBUGGING
SYMBOL TABLE:
00000000 l    d  .init	00000000 .init
000000a0 l    d  .text	00000000 .text
0000071c l    d  .fini	00000000 .fini
20000000 l    d  .dalign	00000000 .dalign
0000071c l    d  .dlalign	00000000 .dlalign
20000000 l    d  .data	00000000 .data
20000000 l    d  .bss	00000000 .bss
20000700 l    d  .stack	00000000 .stack
00000000 l    d  .debug_info	00000000 .debug_info
00000000 l    d  .debug_abbrev	00000000 .debug_abbrev
00000000 l    d  .debug_loc	00000000 .debug_loc
00000000 l    d  .debug_aranges	00000000 .debug_aranges
00000000 l    d  .debug_ranges	00000000 .debug_ranges
00000000 l    d  .debug_line	00000000 .debug_line
00000000 l    d  .debug_str	00000000 .debug_str
[... 1600 lines cut ...]
 6e0:	6c070513          	addi	a0,a4,1728 # 2dc6c0 <_data_lma+0x2dbfa4>
 6e4:	35e1                	jal	5ac <DelaySysTick>
 6e6:	00048793          	mv	a5,s1
 6ea:	438c                	lw	a1,0(a5)
 6ec:	00158713          	addi	a4,a1,1
 6f0:	c02e                	sw	a1,0(sp)
 6f2:	c398                	sw	a4,0(a5)
 6f4:	3fad                	jal	66e <adc_get>
 6f6:	4582                	lw	a1,0(sp)
 6f8:	862a                	mv	a2,a0
 6fa:	00000537          	lui	a0,0x0
 6fe:	70850513          	addi	a0,a0,1800 # 708 <main+0x72>
 702:	33f5                	jal	4ee <printf>
 704:	bf6d                	j	6be <main+0x28>
 706:	0000                	unimp
 708:	6e756f43          	fmadd.q	ft10,fa0,ft7,fa3,unknown
 70c:	3a74                	lbu	a3,23(a2)
 70e:	2520                	lbu	s0,10(a0)
 710:	2064                	lbu	s1,6(s0)
 712:	6461                	lui	s0,0x18
 714:	25203a63          	0x25203a63
 718:	0a64                	addi	s1,sp,284
 71a:	000d                	c.nop	3

```

eeucalyptus, 2024-09-13:
The problem in your original code is that you use `RCC->APB2PRSTR |= RCC_APB2Periph_ADC1;` to reset the peripheral without previously ungating the ADC clock. In your suggested fix you used the correct register `RCC->APB2PCENR` to ungate the clock, but your suggestion is to do that INSTEAD of writing to `RCC->APB2PRSTR`. Your fix is exactly correct and it's the way that it's done in the example too. It's also not strictly required to reset the ADC before further configuration. 

NTL just for completeness, I tested the following adc_init with both ungate and reset and it works just fine:
```
void adc_init(void)
{
    RCC->APB2PCENR |= RCC_APB2Periph_ADC1;
    RCC->APB2PRSTR |= RCC_APB2Periph_ADC1;
    RCC->APB2PRSTR &= ~RCC_APB2Periph_ADC1;

    RCC->CFGR0 |=(0b01000<<11);//FOR 3.2V MAX ADC FREQ IS 12MZ ADCPRES[11:15] 010XX
    RCC->APB2PCENR |=RCC_IOPAEN;// enable GPIOA PA1
    GPIOA->CFGLR =0X44444404;//PA1 AS ANALOG INPUT

    // Set up single conversion on chl 0
    ADC1->RSQR1 = 0;
    ADC1->RSQR2 = 0;
    ADC1->RSQR3 = 0;    // 0-9 for 8 ext inputs and two internals
    ADC1->SAMPTR2 |=6; // 241 CYCLES

    ADC1->CTLR2 |= ADC_ADON | ADC_EXTSEL;// turn on ADC and set rule group to sw trig
//-------------------------
    // Reset calibration

}
```

closed 2024-09-13 by eeucalyptus