cnlohr/ch32fun #391: minichlink programmer precedence issues

lhartmann, 2024-09-09:
Current behavior: When both WCHLink and Ardulink are attached, it is not possible to use ardulink.

```
$ minichlink -c /dev/ttyACM2
Found WCH Link
WCH Programmer is LinkE version 2.11
Chip Type: 003
Setup success
```

Expected behavior: Explicit command line flags should have precedence over autodetect.

BTW, I also can not see a way to choose which WCHLink to use if there is more than one connected.

TommyMurphyTM1234, 2024-09-13:
> Current behavior: When both WCHLink and Ardulink are attached, it is not possible to use ardulink.
>
> ```
> $ minichlink -c /dev/ttyACM2
> Found WCH Link
> WCH Programmer is LinkE version 2.11
> Chip Type: [...]

Seems to be down to this:

Specify the Ardulink serial port:

- https://github.com/cnlohr/ch32v003fun/blob/486fd2a8cd266947fda01fc8124afc569393e176/minichlink/minichlink.c#L113-L118

But then WCHLink (and other options) still takes precedence because `init_hints->serial_port` is only checked last:

- https://github.com/cnlohr/ch32v003fun/blob/486fd2a8cd266947fda01fc8124afc569393e176/minichlink/minichlink.c#L50-L72
> 
> BTW, I also can not see a way to choose which WCHLink to use if there is more than one connected.

Does `-C ardulink` make any difference?

- https://github.com/cnlohr/ch32v003fun/blob/486fd2a8cd266947fda01fc8124afc569393e176/minichlink/minichlink.c#L119-L124

Although, even if it should work, this looks odd - why `TryInit_B003Fun()` instead of `TryInit_Ardulink()` for the Ardulink case?

- https://github.com/cnlohr/ch32v003fun/blob/486fd2a8cd266947fda01fc8124afc569393e176/minichlink/minichlink.c#L36-L49

> BTW, I also can not see a way to choose which WCHLink to use if there is more than one connected.

Yes - doesn't look like this caters for the possibility of multiple WCHLinkE's?

- https://github.com/cnlohr/ch32v003fun/blob/486fd2a8cd266947fda01fc8124afc569393e176/minichlink/pgm-wch-linke.c#L565-L589

eeucalyptus, 2024-09-13:
I see, the priority based handling has the advantage of being convenient, lean and intuitive at the same time. But I guess since you ran into it, it's worth allowing explicit selection too.  Would you consider submitting a PR with a proposed explicit handling?

TommyMurphyTM1234, 2024-12-07:
> Although, even if it should work, this looks odd - why `TryInit_B003Fun()` instead of `TryInit_Ardulink()` for the Ardulink case?
> 
> * https://github.com/cnlohr/ch32v003fun/blob/486fd2a8cd266947fda01fc8124afc569393e176/minichlink/minichlink.c#L36-L49
> 

Regarding this issue, see this PR:

- https://github.com/cnlohr/ch32v003fun/pull/472

cnlohr, 2024-12-09:
Adding this as a feature is on the table the args rework.

closed 2025-11-25 by cnlohr