cnlohr/ch32fun #395: Do not recognize my WCH-E link but works fine on naneos platform

hientv1999, 2024-09-21:
Hi, I used to develop firmware for CH32v003 on naneos platform and it works fine. Now I move to ch32v003fun to optimize flash usage. However, using the same WCH-E Link programmer, I don't know why ch32v003fun doesn't recognize my device. I have turned off any tab that can potentially use and block the COM port of my programmer.
I'm using debugprintfdemo example though
[image]

TommyMurphyTM1234, 2024-09-21:
Maybe you need to install the WinUSB drivers, e.g. using [Zadig](https://zadig.akeo.ie/)?

- https://github.com/cnlohr/ch32v003fun/issues/328

hientv1999, 2024-09-21:
But I can use on naneos just fine. Why on ch32v003fun I have to install driver? I’m using same computer, same programmer board, same board to be programmed. Everything is same exact the platform environment

eeucalyptus, 2024-09-21:
Hi, I'm not sure how programming works with pio, haven't worked with that too much, but it is possible that it uses the stock USB driver. As @TommyMurphyTM1234 said, use zadig to change the driver and you should be good to go. You can always revert, if you want to go back to the original driver.

TommyMurphyTM1234, 2024-09-21:
> Hi, I'm not sure how programming works with pio, haven't worked with that too much, but it is possible that it uses the stock USB driver. 

I think so, assuming that's what's referred to here: 

- https://pio-ch32v.readthedocs.io/en/latest/installation.html#install-drivers-rules

hientv1999, 2024-09-21:
Hi, I found the solution. I don't need to reinstall USB driver whatsoever. I just followed this tutorial and created the project again from scratch and it should be good. 1 thing to change from this tutorial is that I don't need to place ch32v003fun.c and ch32v003fun.h in src folder anymore. Just main.c and funconfig.h files are enough
https://github.com/cnlohr/ch32v003fun/issues/339

hientv1999, 2024-09-21:
Hi, I found the solution. I don't need to reinstall USB driver whatsoever. I just followed this tutorial and created the project again from scratch and it should be good. 1 thing to change from this tutorial is that I don't need to place ch32v003fun.c and ch32v003fun.h in src folder anymore. Just main.c and funconfig.h files are enough
https://github.com/cnlohr/ch32v003fun/issues/339

This keeps my workflow relatively similar to the original noneos framework. Only pressing arrow button to upload firmware and no need to use CLI to make file

closed 2024-09-21 by hientv1999