cnlohr/ch32fun #396: How to read UART printf ?

hientv1999, 2024-09-21:
Hi, I used to use noneos framework. I usually call USART_Printf_Init(9600) at the beginning of setup function before I start writing my debug message via printf. However, ch32v003fun framework doesn't support USART_Printf_Init(9600) function. I ran the debugprintfdemo example but don't get any signal on TX line. How can I still use TX line for UART printf? Simply doing printf obviously doesn't work on ch32v003fun for me.

BogdanTheGeek, 2024-09-21:
@hientv1999 debugprintf uses the debug port for printf, not uart. Check the uart examples and look for FUNCONF_USE_UARTPRINTF.

hientv1999, 2024-09-21:
Can you please explain what is debug port? Is it the SWIO pin? If yes, how can I read the printf result?

TommyMurphyTM1234, 2024-09-21:
> Can you please explain what is debug port? Is it the SWIO pin? If yes, how can I read the printf result? It would be great if I can free up the TX pin and use SWIO for printf

- https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/examples/debugprintfdemo/debugprintfdemo.c#L1-L2

hientv1999, 2024-09-21:
> * https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/examples/debugprintfdemo/debugprintfdemo.c#L1-L2

I already know this example code and compile but I see no output result.

TommyMurphyTM1234, 2024-09-21:
> I already know this example code and compile but I see no output result.

You asked if it was using SWIO and I was confirming that it is.

TommyMurphyTM1234, 2024-09-21:
Any use? 

- https://github.com/cnlohr/ch32v003fun/issues/334#issuecomment-2212480952

eeucalyptus, 2024-09-21:
To read the debug prints you can just use minichlink -T which can also most easily be called with `make terminal` with the ch32v003fun makefile

closed 2024-09-22 by hientv1999