cnlohr/ch32fun #397: Different OPA ADC reading on same hardware when transferring same code from noneos to ch32v003fun framework

hientv1999, 2024-09-22:
I spent half a day but still cannot solve this problem. I used to program OPA + ADC in the noneos framework. It was to read the analog signal from a microphone. However, when I moved to ch32v003fun framework, with the same PCB, the analog output is much higher than what I get in noneos framework. Both tests were conducted on the same PCB in the same room to keep ambient sound the same. I have checked if UART affects by remove UART but doesn't help.
Below is my original code in noneos
`#include <ch32v00x.h>
#include <stdio.h>
#include <stdbool.h>

#define BATT_ADC    ADC_Channel_3
#define MIC_ADC     ADC_Channel_7

static u16 mic_adc_max = 0;
static u16 mic_adc_min = 0;

void ADC1_IRQHandler() __attribute__((interrupt("WCH-Interrupt-fast")));
void ADC1_IRQHandler()
{
    if ((ADC1->STATR & (ADC_IT_JEOC >> 8)) != (uint32_t)RESET){
        u16 adc_value = ADC_GetInjectedConversionValue(ADC1, ADC_InjectedChannel_1);
        if (adc_value < mic_adc_min){
            mic_adc_min = adc_value;
        }
        if (adc_value > mic_adc_max){
            mic_adc_max = adc_value;
        }
        ADC_ClearITPendingBit(ADC1, ADC_IT_JEOC);
    }
}

void ADC_config(u8 mic, u8 batt, u8 sampleTime){
    // replace OPA_Init( &OPA_InitStructure );
    uint32_t tmp = 0;
    tmp = EXTEN->EXTEN_CTR;
    tmp &= ~(uint32_t)(3<<17);
    tmp |= (CHP0 << 18) | (CHN0 << 17);
    EXTEN->EXTEN_CTR = tmp;
    // replace OPA_Cmd( ENABLE );
    EXTEN->EXTEN_CTR |= (uint32_t)(1 << 16);
    
    RCC->APB2PCENR |= RCC_APB2Periph_ADC1;
    RCC_ADCCLKConfig(RCC_PCLK2_Div2);

    ADC_InitTypeDef ADC_InitStructure = {0};
    RCC->APB2PRSTR |= RCC_APB2Periph_ADC1;
    RCC->APB2PRSTR &= ~RCC_APB2Periph_ADC1;
    ADC_InitStructure.ADC_Mode = ADC_Mode_Independent;
    ADC_InitStructure.ADC_ScanConvMode = ENABLE;
    ADC_InitStructure.ADC_ContinuousConvMode = ENABLE ;
    ADC_InitStructure.ADC_ExternalTrigConv = ADC_ExternalTrigInjecConv_None;
    ADC_InitStructure.ADC_DataAlign = ADC_DataAlign_Right;
    ADC_InitStructure.ADC_NbrOfChannel = 1;
    ADC_Init(ADC1, &ADC_InitStructure);

    ADC_InjectedSequencerLengthConfig(ADC1, 2);
    ADC_InjectedChannelConfig(ADC1, mic, ADC_Channel_1, sampleTime);
    ADC_InjectedChannelConfig(ADC1, batt, ADC_Channel_2, sampleTime);
    ADC1->CTLR2 &= ((uint32_t)0xFFFF7FFF);

    NVIC_EnableIRQ( ADC_IRQn );

    ADC_Calibration_Vol(ADC1, ADC_CALVOL_50PERCENT);
    ADC1->CTLR1 |= ADC_IT_JEOC;     // Enable ADC interurpt
    mic_adc_max = 0;
    mic_adc_min = 1023;
}

void ADC_start(){
    mic_adc_max = 0;
    mic_adc_min = 1023;
    ADC1->CTLR2 |= ((uint32_t)0x00000001); // Enable ADC1
    ADC1->CTLR2 |= ((uint32_t)0x00000008); // Reset calibration
    while (ADC1->CTLR2 & ((uint32_t)0x00000008)); // GetResetCalibrationStatus
    ADC1->CTLR2 |= ((uint32_t)0x00000004);  // Start calibration
    while (ADC1->CTLR2 & ((uint32_t)0x00000004)); 
    ADC1->CTLR2 |= ((uint32_t)0x00208000);  // start injection 
    
}

u16 mic_read_range(){
    u16 range = mic_adc_max - mic_adc_min;
    mic_adc_max = 0;
    mic_adc_min = 1023;
    return range;
}

int main()
{
	SystemInit();
	ADC_config(MIC_ADC, BATT_ADC, ADC_SampleTime_241Cycles);
	ADC_start();
    while(1){
		printf("%i", mic_read_range());
		Delay_Ms(50);
    }
}`

And below is the same code but modified to work with ch32v003fun
`#include "ch32v003fun.h"
#include "ch32v003_GPIO_branchless.h"
#include <stdio.h>
#include <stdbool.h>

#define BATT_ADC    ADC_Channel_3
#define MIC_ADC     ADC_Channel_7

static u16 mic_adc_max = 0;
static u16 mic_adc_min = 0;

uint16_t ADC_GetInjectedConversionValue(ADC_TypeDef *ADCx, uint8_t ADC_InjectedChannel)
{
    __IO uint32_t tmp = 0;

    tmp = (uint32_t)ADCx;
    tmp += ADC_InjectedChannel + IDATAR_Offset;

    return (uint16_t)(*(__IO uint32_t *)tmp);
}

void ADC_ClearITPendingBit(ADC_TypeDef *ADCx, uint16_t ADC_IT)
{
    uint8_t itmask = 0;

    itmask = (uint8_t)(ADC_IT >> 8);
    ADCx->STATR = ~(uint32_t)itmask;
}

void ADC1_IRQHandler() __attribute__((interrupt));
void ADC1_IRQHandler()
{
    if ((ADC1->STATR & (ADC_IT_JEOC >> 8)) != (uint32_t)RESET){
        u16 adc_value = ADC_GetInjectedConversionValue(ADC1, ADC_InjectedChannel_1);
        if (adc_value < mic_adc_min){
            mic_adc_min = adc_value;
        }
        if (adc_value > mic_adc_max){
            mic_adc_max = adc_value;
        }
        ADC_ClearITPendingBit(ADC1, ADC_IT_JEOC);
    }
}

void ADC_InjectedChannelConfig(ADC_TypeDef *ADCx, uint8_t ADC_Channel, uint8_t Rank, uint8_t ADC_SampleTime)
{
    uint32_t tmpreg1 = 0, tmpreg2 = 0, tmpreg3 = 0;

    if(ADC_Channel > ADC_Channel_9)
    {
        tmpreg1 = ADCx->SAMPTR1;
        tmpreg2 = SAMPTR1_SMP_Set << (3 * (ADC_Channel - 10));
        tmpreg1 &= ~tmpreg2;
        tmpreg2 = (uint32_t)ADC_SampleTime << (3 * (ADC_Channel - 10));
        tmpreg1 |= tmpreg2;
        ADCx->SAMPTR1 = tmpreg1;
    }
    else
    {
        tmpreg1 = ADCx->SAMPTR2;
        tmpreg2 = SAMPTR2_SMP_Set << (3 * ADC_Channel);
        tmpreg1 &= ~tmpreg2;
        tmpreg2 = (uint32_t)ADC_SampleTime << (3 * ADC_Channel);
        tmpreg1 |= tmpreg2;
        ADCx->SAMPTR2 = tmpreg1;
    }

    tmpreg1 = ADCx->ISQR;
    tmpreg3 = (tmpreg1 & ISQR_JL_Set) >> 20;
    tmpreg2 = ISQR_JSQ_Set << (5 * (uint8_t)((Rank + 3) - (tmpreg3 + 1)));
    tmpreg1 &= ~tmpreg2;
    tmpreg2 = (uint32_t)ADC_Channel << (5 * (uint8_t)((Rank + 3) - (tmpreg3 + 1)));
    tmpreg1 |= tmpreg2;
    ADCx->ISQR = tmpreg1;
}

void ADC_config(u8 mic, u8 batt, u8 sampleTime){
    // OPA init
	u32 temp = 0;
    temp = EXTEN->EXTEN_CTR;
    temp &= ~(u32)(3<<17);
    temp |= (CHP0 << 18) | (CHN0 << 17);
    EXTEN->EXTEN_CTR = temp;
    // turn on the op-amp
	EXTEN->EXTEN_CTR |= EXTEN_OPA_EN;

    // Enable ADC
    RCC->APB2PCENR |= RCC_APB2Periph_ADC1;
    // ADCCLK = 24 MHz => RCC_ADCPRE = 0: divide by 2
	RCC->CFGR0 &= ~(0x1F<<11);

    // Reset the ADC to init all regs
	RCC->APB2PRSTR |= RCC_APB2Periph_ADC1;
	RCC->APB2PRSTR &= ~RCC_APB2Periph_ADC1;
    // enable scanning
    temp = ADC1->CTLR1;
    temp &= ADC_CTLR1_CLEAR_Mask;
	ADC1->CTLR1 = (temp | (u32)ADC_Mode_Independent | ADC_SCAN);
    // Enable continuous conversion
    temp = ADC1->CTLR2;
    temp &= CTLR2_CLEAR_Mask;
	ADC1->CTLR2 = (temp | (u32)ADC_DataAlign_Right | (u32)ADC_ExternalTrigInjecConv_None | ADC_CONT);
    // 1 channel
    temp = ADC1->RSQR1;
    temp &= RSQR1_CLEAR_Mask;
    temp |= (1-1) << 20;
    ADC1->RSQR1 = temp;

    // inject 2 sequences
    temp = ADC1->ISQR;
    temp &= ISQR_JL_Reset;
    ADC1->ISQR = (temp | ((2-1) << 20));
    ADC_InjectedChannelConfig(ADC1, mic, ADC_Channel_1, sampleTime);
    ADC_InjectedChannelConfig(ADC1, batt, ADC_Channel_2, sampleTime);
    ADC1->CTLR2 &= ((u32)0xFFFF7FFF);

    NVIC_EnableIRQ( ADC_IRQn );

    ADC1->CTLR1 &= ~(u32)(3<<25);
    ADC1->CTLR1 |= ADC_CALVOL_50PERCENT;
    ADC1->CTLR1 |= ADC_IT_JEOC;     // Enable ADC interurpt
    mic_adc_max = 0;
    mic_adc_min = 1023;
}

void ADC_start(){
    mic_adc_max = 0;
    mic_adc_min = 1023;
    ADC1->CTLR2 |= ((uint32_t)0x00000001); // Enable ADC1
    ADC1->CTLR2 |= ((uint32_t)0x00000008); // Reset calibration
    while (ADC1->CTLR2 & ((uint32_t)0x00000008)); // GetResetCalibrationStatus
    ADC1->CTLR2 |= ((uint32_t)0x00000004);  // Start calibration
    while (ADC1->CTLR2 & ((uint32_t)0x00000004)); 
    ADC1->CTLR2 |= ((uint32_t)0x00208000);  // start injection 
    
}

u16 mic_read_range(){
    u16 range = mic_adc_max - mic_adc_min;
    mic_adc_max = 0;
    mic_adc_min = 1023;
    return range;
}

int main()
{
	SystemInit();
	GPIO_port_enable(GPIO_port_A);
	GPIO_port_enable(GPIO_port_C);
	GPIO_port_enable(GPIO_port_D);
	ADC_config(MIC_ADC, BATT_ADC, ADC_SampleTime_241Cycles);
	ADC_start();
    while(1){
		printf("%i", mic_read_range());
		Delay_Ms(50);
    }
}`

TommyMurphyTM1234, 2024-09-22:
Have you compared the relevant register values after running the two versions of the program (at least until the initialisation has been done) to see if there are any differences?

hientv1999, 2024-09-22:
> Have you compared the relevant register values after running the two versions of the program (at least until the initialisation has been done) to see if there are any differences?

How can I check that? Is it as simple as printing the value of the registers via printf?

TommyMurphyTM1234, 2024-09-22:
> > Have you compared the relevant register values after running the two versions of the program (at least until the initialisation has been done) to see if there are any differences?
> 
> How can I check that? Is it as simple as printing the value of the registers via printf?

Use the debugger to inspect the relevant memory mapped registers?

hientv1999, 2024-09-22:
I found 2 registers having different values between 2 frameworks. On noneos, I got:
RCC->CFGR0: 0x0
RCC->APB2PCENR: 0x4231

On ch32v003fun, I got:
RCC->CFGR0: 0xa
RCC->APB2PCENR: 0x4230 

I tried copying the same values from noneos to ch32v003fun project, but it crashed the printf message. I receive unreadable characters. Why then?

TommyMurphyTM1234, 2024-09-23:
> I found 2 registers having different values between 2 frameworks

Did you try decoding the different register values to understand the differences between the two programs?

> I tried copying the same register values from noneos to ch32v003fun project, but it crashed the printf message. I receive unreadable characters. Why then?

Did you try debugging the code to understand what's going on; wrong?

hientv1999, 2024-09-23:
> Did you try decoding the different register values to understand the differences between the two programs?
>
>
> Did you try debugging the code to understand what's going on/wrong?

I haven't used the debugger yet so I have to figure out how to use it. But the only possible reason for this issue I can think of is the different in either SystemInit or the interrupt priority difference between 2 frameworks. I have even copied the same code for legacy functions for ADC from old framework to replicate the same code as much as possible and it still results in this difference ADC value reading.

TommyMurphyTM1234, 2024-09-23:
> I haven't used the debugger yet so I have to figure out how to use it. But the only possible reason for this issue I can think of is the different in either SystemInit or the interrupt priority [...]

Surely this is also potentially relevant to the difference in behaviour?

> After the setup process, just before entering while loop, I read the registers related to ADC1. I found 2 registers having different values between 2 frameworks. 
>
> On noneos, I got: 
> RCC->CFGR0: 0x0
> RCC->APB2PCENR: 0x4231
> 
> On ch32v003fun, I got:
> RCC->CFGR0: 0xa
> RCC->APB2PCENR: 0x4230

hientv1999, 2024-09-23:
Can you please show me how I can read more information about the register? Like RCC_CFGR0 and RCC_APB2PCENR? Online datasheet doesn't tell. Maybe I didn't find the correct datasheet.

TommyMurphyTM1234, 2024-09-23:
> Can you please show me how I can read more information about the register? Like RCC_CFGR0 and RCC_APB2PCENR? Online datasheet doesn't tell. Maybe I didn't find the correct datasheet.

The datasheet/reference manual explains these registers in detail.

- https://www.wch-ic.com/downloads/CH32V003RM_PDF.html

hientv1999, 2024-09-23:
I realized all RISC-V uses the same register map so I found out it from another document. But thank you for sharing me the above document I've never seen it before.

RCC->CFGR0: 0x0 on noneos meaning it uses HSI as system clock
RCC->CFGR0: 0xa on ch32v003fun meaning it uses PLL as system clock.

I added #define FUNCONF_USE_HSI 1 to funconfig.h but it doesn't change the register value at all. Anything else I missed?

RCC->APB2PCENR: 0x4231 on noneos meaning I/O auxiliary function module clock is enable
RCC->APB2PCENR: 0x4230 on ch32v003fun meaning I/0 auxiliary function module clock is disabled.

I don't know how to fix this difference

TommyMurphyTM1234, 2024-09-23:
> I realized all RISC-V uses the same register map

This is not correct. Different RISC-V implementations use different (often **very** different) memory maps and different memory mapped peripherals. When using a specific implementation, in particular a specific RISC-V based MCU, it's vital to use the documentation for that specific target.

hientv1999, 2024-09-23:
> This is not correct. Different RISC-V implementations use different (often **very** different) memory maps and different memory mapped peripherals. When using a specific implementation, in [...]

Ok I got it. But I just doubled check with CH32V003 datasheet and it the same. The above 2 comparisons are still valid. Do you have any idea how to fix?

TommyMurphyTM1234, 2024-09-23:
> RCC->CFGR0: 0x0 on noneos meaning it uses HSI as system clock RCC->CFGR0: 0xa on ch32v003fun meaning it uses PLL as system clock.
> 
> I added #define FUNCONF_USE_HSI 1 to funconfig.h but it doesn't change the register value at all. Anything else I missed?

Maybe you also need to ensure that `FUNCONF_USE_PLL` is undefined or set to 0?

- https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/ch32v003fun/ch32v003fun.h#L55
- https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/ch32v003fun/ch32v003fun.c#L1568-L1580

> RCC->APB2PCENR: 0x4231 on noneos meaning I/O auxiliary function module clock is enable RCC->APB2PCENR: 0x4230 on ch32v003fun meaning I/0 auxiliary function module clock is disabled.
> 
> I don't know how to fix this difference

Maybe by just hacking `ch32v003fun` temporarily if needed?

- https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/ch32v003fun/ch32v003fun.c#L1327-L1334

> By the way, 1 significant difference between 2 frameworks are interrupt void ADC1_IRQHandler() **attribute**((interrupt)); in ch32v003fun void ADC1_IRQHandler() **attribute**((interrupt("WCH-Interrupt-fast"))); in noneos Not sure if this can be the reason of the issue?

Could be a factor alright. Again, try modifying `ch32v003fun` to use the fast interrupt option temporarily and see if it makes a difference. 

You should probably try these three changes one by one and then maybe in combination to see if they make any difference.

I would try the fast interrupt change first and on its own. 
Then the clocking change on its own and then with fast interrupt if necessary. 
Then the other one and other combinations of the three.

hientv1999, 2024-09-23:
Thank you for the detail answer but honestly I don't know how to try the fast interrupt option. Simply doing interrupt("WCH-Interrupt-fast") will not work on ch32v003fun framework

TommyMurphyTM1234, 2024-09-23:
> Thank you for the detail answer but honestly I don't know how to try the fast interrupt option. Simply doing interrupt("WCH-Interrupt-fast") will not work on ch32v003fun framework

As far as I know only the custom GCC toolchain from WCH/MounRiver supports that non-standard attribute. And I'm not sure that they ever made the sources for their changes available or tried to upstream them to the master GCC project (where they probably wouldn't be accepted anyway).

hientv1999, 2024-09-23:
By the way, in funconfig.h, I add
#define FUNCONF_USE_HSI             1        
#define FUNCONF_USE_PLL             0

And on line 1329 of ch32v003func.c, I modify into
RCC->APB2PCENR |= RCC_APB2Periph_GPIOD | RCC_APB2Periph_USART1 | RCC_APB2Periph_AFIO;

These make the register values the same as noneos framework now. However, the issue of ADC reading is still there.

hientv1999, 2024-09-23:
> As far as I know only the custom GCC toolchain from WCH/MounRiver supports that non-standard attribute. And I'm not sure that they ever made the sources for their changes available or tried to [...]

Does that mean I have no solution for this problem? I didn't expect to encounter this awkward issue.

hientv1999, 2024-09-23:
However, 1 positive thing after using HSI instead of PLL is that ADC reading is much more stable. The difference between min and max ADC reading in 50ms is way less.

TommyMurphyTM1234, 2024-09-23:
> Does that mean I have no solution for this problem? I didn't expect to encounter this awkward issue.

If the issue is that you definitely need the custom fast interrupt mechanism in order to do what you're trying to do (and that doesn't seem to have been ascertained for sure so far as far as I can see) then you'll need a toolchain with support for it. And, as far as I know, the only toolchain with that support is the custom WCH/MounRiver one.

TommyMurphyTM1234, 2024-09-23:
> By the way, in funconfig.h, I add #define FUNCONF_USE_HSI 1 #define FUNCONF_USE_PLL 0
>
> And on line 1329 of ch32v003func.c, I modify into RCC->APB2PCENR |= RCC_APB2Periph_GPIOD | [...]

What if you try these changes one at a time as I previously suggested? Maybe only one of them makes a difference?

hientv1999, 2024-09-23:
> If the issue is that you definitely need the custom fast interrupt mechanism in order to do what you're trying to do (and that doesn't seem to have been ascertained for sure so far as far as I can [...]

I think I have to accept the issue for now. Normally my mic range reading will vary from 10 to 1023. Now it starts from 670 to 1023. I will have less dynamic range, but the good thing is that it looks like the signal is more stable than in noneos framework, so the same resolution may be achieved.

hientv1999, 2024-09-23:
> What if you try these changes one at a time as I previously suggested? Maybe only one of them makes a difference?

I did try each fix. Still doesn't solve my ADC problem.

hientv1999, 2024-10-19:
I found the problem. Turn out I set the pin mode to be **input float** instead of **input analog**