cnlohr/ch32fun #398: Store more than 2 bytes in optiondata

hientv1999, 2024-09-23:
Hi, I checked the example project optiondata and had it run successfully. However, I want to store more than 2 bytes. How can I achieve this?

TommyMurphyTM1234, 2024-09-23:
Aren't there only two bytes available for user data? 

- https://github.com/openwch/arduino_core_ch32/issues/63#issuecomment-2040703583

> For other readers: That particular address is mentioned at the top of page 179 of the CH32V003 Reference Manual. At the bottom of that page it's stated to be the reserved byte for WRP2, one of the 8 user-option bytes that are mentioned on page 178:
> 
>> "The user-option bytes is solidified in FLASH and will be reloaded into the corresponding register after system reset, and can be erased and programmed by the user at will. The user option bytes information block has a total of 8 bytes (4 bytes for write protection, 1 byte for read protection, 1 byte for configuration options, **and 2 bytes for storing user data**), and each bit has its inverse code bit for checksum during loading."

hientv1999, 2024-09-23:
> Aren't there only two bytes available for user data?
> 
> * [Flash example openwch/arduino_core_ch32#63 (comment)](https://github.com/openwch/arduino_core_ch32/issues/63#issuecomment-2040703583)
> 
> > For other readers: That particular address is mentioned at the top of page 179 of the CH32V003 Reference Manual. At the bottom of that page it's stated to be the reserved byte for WRP2, one of the 8 user-option bytes that are mentioned on page 178:
> > > "The user-option bytes is solidified in FLASH and will be reloaded into the corresponding register after system reset, and can be erased and programmed by the user at will. The user option bytes information block has a total of 8 bytes (4 bytes for write protection, 1 byte for read protection, 1 byte for configuration options, **and 2 bytes for storing user data**), and each bit has its inverse code bit for checksum during loading."

This guy claims to make an EEPROM library for 26 bytes, which made me think I can achieve the same
https://github.com/openwch/arduino_core_ch32/issues/63#issuecomment-2039607314

TommyMurphyTM1234, 2024-09-23:
> This guy claims to make an EEPROM library for 26 bytes, which made me think I can achieve the same [openwch/arduino_core_ch32#63 (comment)](https://github.com/openwch/arduino_core_ch32/issues/63#issuecomment-2039607314)

Then try using their code?

hientv1999, 2024-09-23:
> Then try using their code?
>
> * https://github.com/maxint-rd/arduino_core_ch32/tree/main/libraries/EEPROM

Unfortunately there link is broken. But my original thought was there are more than 2 bytes and I just didn't know how to use the remaining bytes. But now you said the limit is 2 and I don't really need more than 2 bytes for my project, I'm ok with optiondata for now

1 question though: Why each data slot is uint16_t, I cannot use it as 2 bytes, to get a total of 4 bytes? I tried but after saving to flash, reading it back mess up my byte data.

TommyMurphyTM1234, 2024-09-23:
> Unfortunately there link is broken.

The link that I posted isn't broken: 

- https://github.com/maxint-rd/arduino_core_ch32/tree/main/libraries/EEPROM

> 1 question though: Why each data slot is uint16_t, I cannot use it as 2 bytes, to get a total of 4 bytes? I tried but after saving to flash, reading it back mess up my byte data.

I'm not sure what exactly you're referring to but it looks like the two user data bytes are packed into a single `uint16_t` halfword:

- https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/examples/optiondata/optiondata.c#L28

You should probably refer to the datasheet/reference manual for specific details on how the user option bytes non-volatile memory block works and the mechanisms that the `optionbytes` example program is using to interact with it.

- https://www.wch-ic.com/downloads/CH32V003RM_PDF.html

hientv1999, 2024-09-23:
> I'm not sure what exactly you're referring to but it looks like the two user data bytes are packed into a single `uint16_t` halfword:
No, I think you are wrong. They pack 2 uint8_t into 1 uint32_t, which isn't efficient. I tried pack 4 uint8_t into 1 uint32_t and read back results in wrong data. Actually they pack 2 uint8_t into 2 uint16_t, but they use uint32_t to access 2 uint16_t at once.

TommyMurphyTM1234, 2024-09-23:
> No, I think you are wrong. They pack 2 uint8_t into 1 uint32_t, which isn't efficient. 

No. `hold` is an array of `uint16_t` that is, for some reason presumably related to the register interface for the non-volatile memory, just being referenced using a `uint32_t *` pointer:

- https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/examples/optiondata/optiondata.c#L14
- https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/examples/optiondata/optiondata.c#L25
- https://github.com/cnlohr/ch32v003fun/blob/bbad78a904a952a04e9edbd13863f8b0331a26fd/examples/optiondata/optiondata.c#L28

closed 2024-09-23 by hientv1999

TommyMurphyTM1234, 2024-09-23:
Thanks a lot @mengstr - that's useful info in the context of this thread. 👍

maxint-rd, 2025-07-15:
Perhaps a bit late to the party, but still...
My code for the EEPROM library as part of the Arduino core was submitted as a PR and later merged:
 -  https://github.com/openwch/arduino_core_ch32/tree/main/libraries/EEPROM
(the original branch was removed after merging).

Since then I've often used it on the V003 to reliably write more than 2 bytes. Thanks @mengstr for contributing that code. I based my library largely on the CH32fun example. I share your feelings regarding compactness and gladly used your creative data-type memory manipulation technique.

@TommyMurphyTM1234 - If you look at my code for the [`commit()`](https://github.com/openwch/arduino_core_ch32/blob/main/libraries/EEPROM/src/EEPROM.cpp#L96) function, you can see quite a bit of comments to explain what's going on. The inverse values are generated automatically and indeed, the whole 64 byte area needs to be written as a whole, so first thing is to hold the first 16 bytes. I used 16 instead of 12 since other CH32 chips use WRPR2 and WRPR3, while V003 marks it reserved.