cnlohr/ch32fun #406: SPI example issue!

Rockwell1799, 2024-10-05:
Hello guys
I wrote this small SPI example, just to make sure that it's working before I get a proper peripheral hooked up, but to my surprise, the SPI is not working, and I don't get anything on my oscilloscope. Any help would be appreciated!

```
#include "ch32v003fun.h"
void SPI_Init(void);
uint8_t SPI_Tx(uint8_t data);
#define CS_High         3
#define CS_Low        19

int main()
{
    SystemInit();
    SPI_Init();

    while(1)
    {
        SPI_Tx(0x80);

        Delay_Ms(1000);

        SPI_Tx(0x1);

    }
    }

void SPI_Init(void)
{

// Enable GPIOC and SPI

   RCC->APB2PCENR |= RCC_APB2Periph_GPIOC | RCC_APB2Periph_SPI1;

   //PC3(CS) IS OUTPUT PP
   GPIOC->CFGLR &= ~(0xf<<(4*3));
   GPIOC->CFGLR |= (GPIO_Speed_10MHz | GPIO_CNF_OUT_PP)<<(4*3);
   // PC5 is SCK, 10MHz Output, alt func, p-p
   GPIOC->CFGLR &= ~(0xf<<(4*5));
   GPIOC->CFGLR |= (GPIO_Speed_10MHz | GPIO_CNF_OUT_PP_AF)<<(4*5);

   // PC6 is MOSI, 10MHz Output, alt func, p-p
   GPIOC->CFGLR &= ~(0xf<<(4*6));
   GPIOC->CFGLR |= (GPIO_Speed_10MHz | GPIO_CNF_OUT_PP_AF)<<(4*6);

    // Configure SPI
        SPI1->CTLR1 =
        SPI_NSS_Soft | SPI_CPHA_1Edge | SPI_CPOL_Low | SPI_DataSize_8b |
        SPI_Mode_Master | SPI_Direction_1Line_Tx |
        SPI_BaudRatePrescaler_2 ;

    // enable SPI port
    SPI1->CTLR1 |= CTLR1_SPE_Set;

     GPIOC->BSHR =  (1<<CS_High);   /* make CS  pins high */

}

uint8_t SPI_Tx(uint8_t data)
{
    GPIOC->BSHR = (1 << CS_Low);

    SPI1->DATAR = data; /* send the contents of d */
    while((SPI1->STATR&(1<<SPI_STATR_RXNE)==0));   /* wait until RXNE is set */

    GPIOC->BSHR = (1 << CS_High);
    return SPI1->DATAR;
}

```

cnlohr, 2024-10-05:
Which chip are you using? Can you add some printf's and do `make clean all monitor`?

Rockwell1799, 2024-10-05:
> Which chip are you using (the SOIC-8?) Can you add some printf's and do `make clean all monitor`?

I'm using the ch32v003f4p6 (SOIC-20). The problem is the clock is not generated at all. Normally, when the SPI master is configured properly, the clock is generated on Pin PC5 (SCK), but in my case the clock is always low.

cnlohr, 2024-10-05:
I just copy-pasted your code, I am probing pins 3, 5 and 6...
[image]

Looks good, but your CS doesn't seem to stay high.

Rockwell1799, 2024-10-05:
So, something is really weird is happening in my end, which I can't figure out yet! But I made a little progress after I started the project from scratch on mounriver studio.
With prescaler of 2 I get the following waveform:

[image]
with prescaler of 4 I get:
[image]
the weirdest thing is my CS line never comes low!

Rockwell1799, 2024-10-06:
This issue is getting me crazy! This is what I'm getting on the logic analyzer with the same code
[image]
if I change the clock prescaler, I get more clock pulses, but completely not like yours.
This is my funconfig.h
```
#ifndef _FUNCONFIG_H
#define _FUNCONFIG_H

#define CH32V003           1
#define FUNCONF_USE_UARTPRINTF  1
#define FUNCONF_SYSTEM_CORE_CLOCK 48000000
#define FUNCONF_UART_PRINTF_BAUD 115200
#endif
```
I'm using these cheap ch32v003 from AliExpress
[image]

TommyMurphyTM1234, 2024-10-06:
Faulty hardware perhaps?

Rockwell1799, 2024-10-06:
> Faulty hardware perhaps?

I already tried two working boards that I used on other projects (ADC ,I2C...) ,and I got the same result! 
I've used SPI quite extensively with stm32 chips and never had a problem like this.

Rockwell1799, 2024-10-06:
After modifying my code a bit, I started seeing some improvements, but I haven't solved the problem yet.
First screenshot : 0xf0 is sent, and I did indeed receive 0xf0 and chip select seems to be alright.
[image]

Second screenshot: I only modified the value from 0xf0 to 0x0f, and things start to look weird again.
[image]

Rockwell1799, 2024-10-07:
Finally, I got the example working, but there is still some weirdness which I will talk about next.I had to check the busy flag, before I pull the CS line down.
the working code is :
```
#include "ch32v003fun.h"
void SPI_Init(void);
void SPI_Tx(uint8_t data);
#define CS_High         3
#define CS_Low        19

int main()
{
    SystemInit();
    SPI_Init();

    while(1)
    {

        SPI_Tx(0x0f);
        Delay_Us(10);
        SPI_Tx(0x80);

    }
    }

void SPI_Init(void)
{

// Enable GPIOC and SPI

   RCC->APB2PCENR |= RCC_IOPCEN | RCC_SPI1EN;

GPIOC->CFGLR =0xBB03000;

    // Configure SPI
       SPI1->CTLR1 |=
        SPI_NSS_Soft  | SPI_CPHA_1Edge | SPI_CPOL_Low | SPI_DataSize_8b |
        SPI_Mode_Master | SPI_Direction_1Line_Tx | SPI_BaudRatePrescaler_8
        ;

    // enable SPI port

SPI1->CTLR1 |= CTLR1_SPE_Set;

    GPIOC->BSHR = (1 << CS_High);

}

void SPI_Tx(uint8_t data)
{
    GPIOC->BSHR = (1 << CS_Low);

    SPI1->DATAR = data;
    while(!(SPI1->STATR & SPI_STATR_TXE)) {}
    while((SPI1->STATR & SPI_STATR_BSY) != 0) {}//with check, the code is working
    GPIOC->BSHR = (1 << CS_High);

}
```
[image]

Rockwell1799, 2024-10-07:
Now, the only thing that bugs me, that I can't assign a value to the SPI1->CTLR1 register.Let's when assigning a direct value to that register, the spi shutsdown completely.
expl:
SPI1->CTLR1= 0x4254; (is equivalent to SPI_NSS_Soft  | SPI_CPHA_1Edge | SPI_CPOL_Low | SPI_DataSize_8b |
        SPI_Mode_Master | SPI_Direction_1Line_Tx | SPI_BaudRatePrescaler_8 ), but for some reasons , when assigning the value 0x4254 to SPI1->CTLR1, the spi shuts off completely.

Rockwell1799, 2024-10-07:
It turns out that the line  while(!(SPI1->STATR & SPI_STATR_TXE)) {} can be ommitted.The code is working with only while((SPI1->STATR & SPI_STATR_BSY) != 0) {}
I'm going to leave the issue open for furher comments, then I will close it as resolved!

cnlohr, 2024-10-07:
@Rockwell1799 Your quanta on your logic analyzer looks like it's the primary culprit.  To capture SPI, your logic analyzer must be able to sample at roughly 3-4x the SPI clock rate.

Rockwell1799, 2024-10-07:
I could be my cheap logic analyzer.I have an oscilloscope and I'm going to use it and report back

Rockwell1799, 2024-10-07:
> @Rockwell1799 Your quanta on your logic analyzer looks like it's the primary culprit.  To capture SPI, your logic analyzer must be able to sample at roughly 3-4x the SPI clock rate.
> 
> EDIT: Also - also TXE and BSY have different meanings. Depending on what your goals are.  One lets you load more data in, one says it's done.

[image]

closed 2024-10-08 by Rockwell1799

cnlohr, 2024-10-08:
@Rockwell1799 what was it?