cnlohr/ch32fun #412: blink not working on WeAct Studio CH32X035F8U6 board

nanikjava, 2024-10-11:
Hi,

I've got this board and compiled `ch32v003fun/examples_x035/blink/blink.c` modified the code a bit like the following:

```
#include "ch32v003fun.h"
#include <stdio.h>

int main()
{
	SystemInit();

	funGpioInitAll();

	funPinMode( PB12, GPIO_CFGLR_OUT_10Mhz_PP );

	while(1)
	{
		funDigitalWrite( PB12, FUN_HIGH );	 // Turn on GPIO
		Delay_Ms( 1000 );
		funDigitalWrite( PB12, FUN_LOW );	 // Turn off GPIO
		Delay_Ms( 1000 );
	}
}
```

I've cross-compiled the toolchain on my Linux machine and was able to compile the sample code and got the following output:

```
riscv32-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32X035 -DMCU_PACKAGE=1 -DTARGET_MCU_LD=4		 -DTARGET_MCU_MEMORY_SPLIT= ../../ch32v003fun//ch32v003fun.ld > ../../ch32v003fun//generated_CH32X035F8U6_.ld
riscv32-unknown-elf-gcc -o blink.elf ../../ch32v003fun//ch32v003fun.c blink.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32imac -mabi=ilp32 -DCH32X03x=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun//../extralibs -I../../ch32v003fun/ -nostdlib -I. -Wall  -Wl,--print-memory-usage -lgcc -T ../../ch32v003fun//generated_CH32X035F8U6_.ld -Wl,--gc-sections
Memory region         Used Size  Region Size  %age Used
           FLASH:         524 B        62 KB      0.83%
             RAM:           0 B        20 KB      0.00%
riscv32-unknown-elf-objdump -S blink.elf > blink.lst
riscv32-unknown-elf-objdump -t blink.elf > blink.map
riscv32-unknown-elf-objcopy -O binary blink.elf blink.bin
riscv32-unknown-elf-objcopy -O ihex blink.elf blink.hex
make -C ../../ch32v003fun//../minichlink all
make[1]: Entering directory '/home/nanik/Downloads/devboards/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/nanik/Downloads/devboards/ch32v003fun/minichlink'
../../ch32v003fun//../minichlink/minichlink -w blink.bin flash -b
WARNING: You are not in the plugdev group, the canned udev rules will not work on your system.
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
make: *** [../../ch32v003fun/ch32v003fun.mk:241: cv_flash] Error 224
````

The changes made was the pin `PB12` as per the [schematic](https://github.com/WeActStudio/WeActStudio.CH32X035CoreBoard/blob/master/Hardware/WeAct-CH32X035_CoreBoard_V10%20SchDoc.pdf) the LED is at PB12 (not sure if I did it right or wrong, appreciate any feedback) on the code.

I'm using  the `wchisp` [project](https://github.com/ch32-rs/wchisp) to flash the board using the following command

```
wchisp  -v flash ./blink.elf
```

which I got the following response

```
12:28:45 [INFO] Opening USB device #0
12:28:45 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 001 Device 085: ID 1a86:55e0
12:28:45 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
12:28:45 [DEBUG] (1) wchisp::transport: <= a1000200 5e23
12:28:45 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
12:28:45 [DEBUG] (1) wchisp::transport: <= a1000200 5e23
12:28:45 [DEBUG] (1) wchisp::flashing: found chip: CH32X035F8U6[0x5e23]
12:28:45 [DEBUG] (1) wchisp::transport: => a70200   1f00
12:28:45 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ff001fe000ff00ffffffffff00020600cdaba1716cbcdad9
12:28:45 [DEBUG] (1) wchisp::flashing: read_config: ff001fe000ff00ffffffffff00020600cdaba1716cbcdad9
[... 62 lines cut ...]
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 672/102412:28:46 [DEBUG] (1) wchisp::transport: => a53d00   a00200005d65656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c3
12:28:46 [DEBUG] (1) wchisp::transport: <= a5000200 0000
████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 728/102412:28:46 [DEBUG] (1) wchisp::transport: => a53d00   d80200001665656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c3
12:28:46 [DEBUG] (1) wchisp::transport: <= a5000200 0000
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 784/102412:28:46 [DEBUG] (1) wchisp::transport: => a53d00   100300004265656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c3
12:28:46 [DEBUG] (1) wchisp::transport: <= a5000200 0000
████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 840/102412:28:46 [DEBUG] (1) wchisp::transport: => a53d00   480300004965656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c3
12:28:46 [DEBUG] (1) wchisp::transport: <= a5000200 0000
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░ 896/102412:28:46 [DEBUG] (1) wchisp::transport: => a53d00   800300002265656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c3
12:28:46 [DEBUG] (1) wchisp::transport: <= a5000200 0000
████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████░░░░░░░░░░░░░ 952/102412:28:46 [DEBUG] (1) wchisp::transport: => a53d00   b80300008465656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c365656565656565c3
12:28:46 [DEBUG] (1) wchisp::transport: <= a5000200 0000
█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████░░░ 1008/102412:28:46 [DEBUG] (1) wchisp::transport: => a51500   f00300006e65656565656565c365656565656565c3
12:28:46 [DEBUG] (1) wchisp::transport: <= a5000200 0000
████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 1024/102412:28:46 [DEBUG] (1) wchisp::transport: => a50500   0004000086
12:28:46 [DEBUG] (1) wchisp::transport: <= a5000200 0000
████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 1024/102412:28:46 [INFO] Code flash 1024 bytes written
12:28:47 [WARN] Skipping verify
12:28:47 [INFO] Now reset device and skip any communication errors
12:28:47 [DEBUG] (1) wchisp::transport: => a20100   01
12:28:47 [DEBUG] (1) wchisp::transport: <= a2000200 0000
12:28:47 [INFO] Device reset
```

The problem is the LED is not flashing as per what the code is doing

cnlohr, 2024-10-11:
Would you mind trying to use minichlink to flash instead?  There is a canned windows .exe, and if you add yourself to the plugdev group and use `sudo make install_udev_rules` you can try that.

nanikjava, 2024-10-11:
> Would you mind trying to use minichlink to flash instead?  There is a canned windows .exe, and if you add yourself to the plugdev group and use `sudo make install_udev_rules` you can try that.

I'm using Linux machine does it has Linux version ?

cnlohr, 2024-10-11:
Yes.  `make` by default will flash the part as long as your user is allowed to talk to the device.

nanikjava, 2024-10-11:
Running `make` it shows the following errors

```
WARNING: You are not in the plugdev group, the canned udev rules will not work on your system.
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
```

TommyMurphyTM1234, 2024-10-11:
> Running `make` it shows the following errors
> 
> ```
> WARNING: You are not in the plugdev group, the canned udev rules will not work on your system.
> Error: Could not initialize any supported programmers
> Error: Could not initialize any supported programmers
> ```
> 
> The board is connected only via USB there is no external programmer available btw

You need to install the udev rules and add the user to the `plugdev` group as @cnlohr said here: 

- https://github.com/cnlohr/ch32v003fun/issues/412#issuecomment-2408059761

cnlohr, 2024-10-11:
You need to add your user to plugdev, 
```
sudo usermod -aG plugdev YOUR_USERNAME
```
If you get an error about plugdev not being a group, you may need to add it:
```
sudo groupadd plugdev
```
But you will definitely need to have the rules install.
```
sudo make install_udev_rules 
```
Only do this after your user is in the plugdev group.

You may need to log out and back in before the OS picks up you're in plugdev.

nanikjava, 2024-10-11:
Thanks. I've followed the [instruction](https://github.com/cnlohr/ch32v003fun/issues/412#issuecomment-2408089578) and getting the following error.
 

```
riscv32-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32X035 -DMCU_PACKAGE=1 -DTARGET_MCU_LD=4		 -DTARGET_MCU_MEMORY_SPLIT= ../../ch32v003fun//ch32v003fun.ld > ../../ch32v003fun//generated_CH32X035F8U6_.ld
riscv32-unknown-elf-gcc -o blink.elf ../../ch32v003fun//ch32v003fun.c blink.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32imac -mabi=ilp32 -DCH32X03x=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun//../extralibs -I../../ch32v003fun/ -nostdlib -I. -Wall  -Wl,--print-memory-usage -lgcc -T ../../ch32v003fun//generated_CH32X035F8U6_.ld -Wl,--gc-sections
Memory region         Used Size  Region Size  %age Used
           FLASH:         592 B        62 KB      0.93%
             RAM:           0 B        20 KB      0.00%
riscv32-unknown-elf-objdump -S blink.elf > blink.lst
riscv32-unknown-elf-objdump -t blink.elf > blink.map
riscv32-unknown-elf-objcopy -O binary blink.elf blink.bin
riscv32-unknown-elf-objcopy -O ihex blink.elf blink.hex
make -C ../../ch32v003fun//../minichlink all
make[1]: Entering directory '/home/nanik/Downloads/devboards/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/nanik/Downloads/devboards/ch32v003fun/minichlink'
../../ch32v003fun//../minichlink/minichlink -w blink.bin flash -b
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
make: *** [../../ch32v003fun/ch32v003fun.mk:241: cv_flash] Error 224
```

nanikjava, 2024-10-11:
Following info from `dmesg` 

```
[122202.795514] usb 1-2: new full-speed USB device number 107 using xhci_hcd
[122203.072929] usb 1-2: New USB device found, idVendor=1a86, idProduct=55e0, bcdDevice=26.00
[122203.072938] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```

and from  `lsusb`

```
Bus 001 Device 108: ID 1a86:55e0 QinHeng Electronics 
```

cnlohr, 2024-10-12:
What distro are you on @nanikjava ?

nanikjava, 2024-10-12:
> What distro are you on @nanikjava ?

Here you go 

[image]

maxgerhardt, 2024-10-14:
As I understand, `minichlink` does not yet have the capability to flash an CH32X035 via the built-in USB bootloader. It's for using a programmer like the WCH-LinkE. So best to modify the makefile to really invoke `wchisp`.

closed 2024-10-21 by eeucalyptus