cnlohr/ch32fun #417: Compile failure for CH32V208WBU6

hexaedron, 2024-10-13:
Hi, i am not sure if i do everything right, but i get a `redeclaration of enumerator 'ETH_IRQn'` error.

I define both `CH32V20x` and `CH32V20x_D8W`.
But if both are defined, i get an error, because [line 369](https://github.com/cnlohr/ch32v003fun/blob/f41db0dc304fd855e8e806282a9a250661708a72/ch32v003fun/ch32v003fun.h#L369) conflicts with [line 408](https://github.com/cnlohr/ch32v003fun/blob/f41db0dc304fd855e8e806282a9a250661708a72/ch32v003fun/ch32v003fun.h#L408).
If i comment out one of the lines, everything compiles fine, which leads to question what line is correct.

hexaedron, 2024-10-14:
According to RM, I guess that correct line is
```c
ETH_IRQn                    = 77,      /* ETH global Interrupt               	               */
```

The whole following block looks strange:
```c
#elif defined(CH32V20x_D8W)
    ETH_IRQn = 61,           /* ETH global Interrupt               	                 */
    ETHWakeUp_IRQn = 62,     /* ETH WakeUp Interrupt                       			 */
    BB_IRQn = 63,            /* BLE BB global Interrupt                              */
    LLE_IRQn = 64,           /* BLE LLE global Interrupt                             */
    TIM5_IRQn = 65,          /* TIM5 global Interrupt                                */
    UART4_IRQn = 66,         /* UART4 global Interrupt                               */
    DMA1_Channel8_IRQn = 67, /* DMA1 Channel 8 global Interrupt                      */
    OSC32KCal_IRQn = 68,     /* OSC32K global Interrupt                              */
    OSCWakeUp_IRQn = 69,     /* OSC32K WakeUp Interrupt                              */
#endif
```
All the numbers here are different from RM.

cnlohr, 2024-10-14:
@hexaedron I think the issue is no one has any CH32V208's to check with, I wonder if the CH32V208 is different from the V203/V205.  Would you mind doing a little more legwork here to try to figure it out?  I have no V208's to check with.

Also, do you have Discord, may be best to discuss there?

hexaedron, 2024-10-14:
Moved to Discord

closed 2024-10-14 by hexaedron