cnlohr/ch32fun #424: InterruptVectorDefault() issues on ch32v20x

hexaedron, 2024-10-21:
Hi, there are some inaccuracies in `InterruptVectorDefault()` when i compile ch32v003fun with ch32v208. They lead to wrong interrupt vector layout.
These inaccuracies seem to have similar source as #417.
I can fix that and make a PR (and i have already fixed it for me), but there is a code style question.

For now, `InterruptVectorDefault()` is full of conditional defines, which makes it hard to understand and to maintain. I propose to alter it a little in the following way:

```c
#elif defined(CH32V10x) || defined(CH32V20x) || defined(CH32V30x)

void InterruptVectorDefault()
{
	asm volatile( "\n\
	.align	1 \n\
	.option norvc; \n\
	j handle_reset \n"
#if !defined(FUNCONF_TINYVECTOR) || !FUNCONF_TINYVECTOR
"	.word   0 \n\
	.word   NMI_Handler                /* NMI */ \n\
	.word   HardFault_Handler          /* Hard Fault */ \n\
	.word   0 \n"
#if defined(CH32V10x)
// Here goes the complete interrupt vector for CH32V10x.
#else if defined(CH32V20x)
// Here goes the complete interrupt vector for CH32V20x with conditional defines only for 20x variants
#else if defined(CH32V30x)
// The same for CH32V30x
#endif
```

If @cnlohr you are OK with that, i am ready to start working on my PR.

eeucalyptus, 2024-10-21:
Go for it! Contribution is much appreciated!

cnlohr, 2024-11-06:
Is the issue resolved?

hexaedron, 2024-11-06:
Absolutely!

closed 2024-11-06 by hexaedron