cnlohr/ch32fun #426: If not use DEBUGPRINTF - error undefined reference to `funAnalogInit'  and `funAnalogRead'

dzanis, 2024-10-23:
Hi. If I use these settings and code, I get a compilation  error
**undefined reference to funAnalogInit  and funAnalogRead**

 _funconfig.h_
```c
#ifndef _FUNCONFIG_H
#define _FUNCONFIG_H

#define CH32V003           1

 #define FUNCONF_USE_DEBUGPRINTF 0
 #define FUNCONF_USE_UARTPRINTF  1

#endif
```

_main.c_
```c
#include "ch32v003fun.h"
#include <stdio.h>

int main()
{
	SystemInit();

	// Enable GPIOs
	funGpioInitAll();
        funAnalogInit();

	funPinMode( PA2, GPIO_CFGLR_IN_ANALOG );    // Corresponds to analog in A0

	while(1)
	{
        printf( "PA2: %d \n",funAnalogRead( ANALOG_0 ) );
		Delay_Ms( 1000 );
	}
}
```

eeucalyptus, 2024-10-23:
Hi, you found a proper bug. The ADC code is within the debugprintf stuff and disabled with uart printf. I submitted a PR for it: #427

cnlohr, 2024-10-28:
Should be resolved.  PR Merged.

closed 2024-10-28 by cnlohr