cnlohr/ch32fun #441: Timer Issue

RhonyD, 2024-11-10:
if Delay_Ms() is used in side main while loop and TIM2 update interrupt is also used simultaneously then complete malfunction is taking place. TIM2 interrupt works perfectly but Delay_Ms() completely malfunctioning.

maxgerhardt, 2024-11-11:
Do you have a specific code example? Since Delay_Ms works with the SysTick, it should not affect anything in TIM2. 

You're not using a `Delay_Ms()` **inside** the interrupt for TIM2, are you?

https://github.com/cnlohr/ch32v003fun/blob/f18f2e9fb6078501f92f0e1a1d4308c36a374e8a/ch32v003fun/ch32v003fun.c#L1730-L1744

https://github.com/cnlohr/ch32v003fun/blob/f18f2e9fb6078501f92f0e1a1d4308c36a374e8a/ch32v003fun/ch32v003fun.h#L13718-L13719

RhonyD, 2024-11-15:
Here is my code

#include "ch32v003fun.h"

#define LED_PORT		GPIOD
#define LED_PIN			2
#define	LED_PIN_1		3

void TIM2_IRQHandler() __attribute__((interrupt("WCH-Interrupt-fast")));

void TIMER2_INIT(u16 psclr, u16 atlr)
{
	
	RCC->APB1PCENR|=(RCC_APB1Periph_TIM2);
  	TIM2->CTLR1|= TIM_CKD_DIV1;
	TIM2->PSC=psclr;
	TIM2->ATRLR=atlr;
	TIM2->SWEVGR |= TIM_UG;
	TIM2->DMAINTENR|=(TIM_IT_Update);
	TIM2->CTLR1 |= TIM_CEN;
}

void TIM2_IRQHandler()
{

if(TIM2->INTFR&(TIM_IT_Update))
{
	LED_PORT->OUTDR^=(1<<LED_PIN);

	TIM2->INTFR &=~(u16)TIM_IT_Update;
}

	
}

int main()
{
	SystemInit();

        RCC->APB2PCENR&=~(RCC_APB2Periph_GPIOD);	
	LED_PORT->CFGLR &= ~(0xf<<(4*LED_PIN));
	RCC->APB2PCENR|=(RCC_APB2Periph_GPIOD);	

	LED_PORT->CFGLR &= ~(0xf<<(4*LED_PIN));
	LED_PORT->CFGLR |=(GPIO_CNF_OUT_PP|GPIO_Speed_50MHz)<<(4*LED_PIN);
        
	LED_PORT->CFGLR &= ~(0xf<<(4*LED_PIN_1));
	LED_PORT->CFGLR |=(GPIO_CNF_OUT_PP|GPIO_Speed_50MHz)<<(4*LED_PIN_1);
	
	TIMER2_INIT(28000-1,999);
	NVIC_SetPriority(TIM2_IRQn,5);
	NVIC_EnableIRQ(TIM2_IRQn);

while(1)
{
                      LED_PORT->OUTDR^=(1<LED_PIN_1);
		      Delay_Ms(1000);

}
return 0;
}

TommyMurphyTM1234, 2024-11-15:
What exactly does this mean? 

> `Delay_Ms()` completely malfunctioning.

What is the nature of the malfunction?

maxgerhardt, 2024-11-15:
Do **not** use WCH fast interrupt with ch32v003fun
```cpp
void TIM2_IRQHandler() attribute((interrupt("WCH-Interrupt-fast")));
```
use instead
```
void TIM2_IRQHandler() __attribute__( ( interrupt() ) );
```
I tested this inside PlatformIO and it fixed the problem for me. It does not crash in the DefaultHandler anymore.

maxgerhardt, 2024-11-15:
Specifically, this was the code that worked for me (aka, did not crash)
```cpp
#include "ch32v003fun.h"

#define LED_PORT GPIOD
#define LED_PIN 2
#define LED_PIN_1 3

void TIM2_IRQHandler() __attribute__( ( interrupt() ) );

void TIMER2_INIT( u16 psclr, u16 atlr ) {
	RCC->APB1PCENR |= ( RCC_APB1Periph_TIM2 );
	TIM2->CTLR1 |= TIM_CKD_DIV1;
	TIM2->PSC = psclr;
	TIM2->ATRLR = atlr;
	TIM2->SWEVGR |= TIM_UG;
	TIM2->DMAINTENR |= ( TIM_IT_Update );
	TIM2->CTLR1 |= TIM_CEN;
}

void TIM2_IRQHandler() {
	if ( TIM2->INTFR & ( TIM_IT_Update ) ) {
		LED_PORT->OUTDR ^= ( 1 << LED_PIN );

		TIM2->INTFR &= ~(u16)TIM_IT_Update;
	}
}

int main() {
	SystemInit();

	RCC->APB2PCENR &= ~( RCC_APB2Periph_GPIOD );
	LED_PORT->CFGLR &= ~( 0xf << ( 4 * LED_PIN ) );
	RCC->APB2PCENR |= ( RCC_APB2Periph_GPIOD );

	LED_PORT->CFGLR &= ~( 0xf << ( 4 * LED_PIN ) );
	LED_PORT->CFGLR |= ( GPIO_CNF_OUT_PP | GPIO_Speed_50MHz ) << ( 4 * LED_PIN );

	LED_PORT->CFGLR &= ~( 0xf << ( 4 * LED_PIN_1 ) );
	LED_PORT->CFGLR |= ( GPIO_CNF_OUT_PP | GPIO_Speed_50MHz ) << ( 4 * LED_PIN_1 );

	TIMER2_INIT( 28000 - 1, 999 );
	NVIC_SetPriority( TIM2_IRQn, 5 );
	NVIC_EnableIRQ( TIM2_IRQn );

	while ( 1 ) {
		LED_PORT->OUTDR ^= ( 1 < LED_PIN_1 );
		Delay_Ms( 1000 );
	}
	return 0;
}
```

RhonyD, 2024-11-15:
Thanks for your reply. I will make the changes and come back to you. Thanks again.

RhonyD, 2024-11-16:
I have implemented your suggested solution and it worked like a charm. Thanks again.

maxgerhardt, 2024-11-17:
Then I think you can close this issue.

closed 2024-11-17 by cnlohr