cnlohr/ch32fun #446: minichlink bricked my Ch32x033!

Rockwell1799, 2024-11-15:
Hello
My ch32x033 was working fine, until I decided to enable the NRST by using the command ./minichlink -d
Now, my chip is bricked, and I can't un brick it even after using the command ./minichlink -u
Any help would be very much appreciated!
TIA

cnlohr, 2024-11-15:
How are you connecting the power to the chip/programmer?  For unbrick to work, it must be powered off the 3.3v rail of the programmer.

Rockwell1799, 2024-11-15:
my connection is as follow:
Mcu                    programmer
VCC                   3.3V
GND                 GND
CLK (PIN C19)   SWC
DIO (PIN C18)   SWD

cnlohr, 2024-11-16:
I can test tonight in case no one else gets a chance before then.

cnlohr, 2024-11-16:
@Rockwell1799 
Wow!  Apparently the official WCH Link-E "unbrick" command does not really do a full unbrick on X-series devices. I commented it out and used ours and it seems to work just fine.

I am assuming you using a WCH Link-E?  If so, we are using the unbrick command from them verbatim.  Though, I will admit, I am seeing a similar behavior / failure. 

If you're up for trying it, I've tacked it on the current gdb_fix branch here: https://github.com/cnlohr/ch32v003fun/pull/444

Rockwell1799, 2024-11-17:
> @Rockwell1799 Wow! Apparently the official WCH Link-E "unbrick" command does not really do a full unbrick on X-series devices. I commented it out and used ours and it seems to work just fine.
>
> I am [...]

Thank cnlohr for testing!
Yes! I'm using a WCH LINK-E, and I'm wondering why did the command ./minichlink -d bricked the chip in the first place? It's working fine with the CH32V003.
I'll test the gdb_fix tomorrow and report back!

Rockwell1799, 2024-11-18:
With the gdb_fix branch, this is what I get after executing the command ./minichlink -u

```
Found WCH Link
Entering Unbrick Mode
Connection starting
DMStatus After Halt: /0/00000c82
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 00000000
HARTINFO: 00312380
Chip Type: 0
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 00000000
HARTINFO: 00312380
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Fault on DefaultReadBinaryBlob
Error: Critical memory zone is still locked out
Warning: Flash timed out
Error: Option Byte Unlock Failed (FLASH_CTRL=cdef89ab)
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Fault on DefaultWriteHalfWord
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 00005aa5
HARTINFO: 00312380
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff800, 5aa5 != f800
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 1ffff800
HARTINFO: 00312380
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Fault on DefaultWriteHalfWord
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 0000c03f
HARTINFO: 00312380
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff802, c03f != f802
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 1ffff802
HARTINFO: 00312380
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Fault on DefaultWriteHalfWord
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 0000ff00
HARTINFO: 00312380
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff804, ff00 != f804
[... 75 lines cut ...]
Fault on DefaultWriteHalfWord
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 000000ff
HARTINFO: 00312380
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff80e, 00ff != f80e
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 1ffff80e
HARTINFO: 00312380
Whole-chip erase
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
Chip Type: 1ff
Unknown chip type.  Report as bug with picture of chip.
Vendored: 1ffff704
marchid : 00000000
HARTINFO: 00312380
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 00000c82
```

cnlohr, 2024-11-18:
Please perform the following operation:

1. Disconnect power.
2. Make sure SWD and SWCLK are connected.
2. Run minichlink -u
3. Apply power.

Rockwell1799, 2024-11-18:
> Please perform the following operation:
> 
> 1. Disconnect power.
> 2. Make sure SWD and SWCLK are connected.
> 3. Run minichlink -u
> 4. Apply power.

When performing the step above, I get this:
```
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
```

cnlohr, 2024-11-20:
Please provide the full log.  Also, please do it at least 4-5 times.

cnlohr, 2024-11-25:
I hate to bother you again but the behavior may yet be different now that everything has been merged to master.

Rockwell1799, 2024-11-25:
> I hate to bother you again but he behavior may yet be different now that everything has been merged to master.

Sorry! I was away for a few days.
Unfortenaltley, I tried more than 10 times and I have the same issue!
For the log is repeating all these lines, and putting it here would be useless.

```
RR: 9 :82 08 06 7f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 7f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 7f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 7f ff ff ff ff 03 
Timed out trying to unbrick

```

cnlohr, 2024-11-25:
I want to see the output leading up to the spew of sad.

Rockwell1799, 2024-11-25:
> I want to see the output leading up to the spew of sad.

```
[I] ~/G/c/minichlink ❯❯❯ ./minichlink -u                                                                                                                        ✘ 250 master ✖ ◼
Found WCH Link
Entering Unbrick Mode
Connection starting
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 7f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 7f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 7f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 7f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
[... 402 lines cut ...]
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 7f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 1f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Error setting write reg. Tell cnlohr. Maybe we should allow retries here?
RR: 9 :82 08 06 3f ff ff ff ff 03 
Timed out trying to unbrick
```

Rockwell1799, 2024-11-25:
> here
[image]
I think my chip is completely faulty beyond repair!

TommyMurphyTM1234, 2024-11-26:
That looks like TDO stuck at 1?

monte-monte, 2024-11-26:
@Rockwell1799 sorry for asking obvious thing, but did you try ``minichlink -D``?
``-d`` and ``-D`` are just sending command to do stuff to LinkE and then it does whatever sequence is programmed for it in its firmware, so it's not minichlink's fault if it is really bricked. Maybe the sequence WCH has programmed in the firmware version isn't compatible with x033? Then I think we should check chip ID before running commands that can potentially brick different chips.

monte-monte, 2024-11-26:
@Rockwell1799 also also, what is connected to the NRST pin itself? Can you pull it high and try programming? Maybe it's constantly shutdown by reset?

Rockwell1799, 2024-11-28:
Sorry for my late reply, I couldn't test what you guys suggested,because I accidentally reversed the polarity on the chip and my board doesn't have a protection diode,so the chip now is dead.I'm going to solder another one and I need to know how to enable the NRST pin without bricking the chip.

NeonFrost, 2024-11-29:
Hi, I was looking through the output Rockwell posted and noticed that in the screenshot it says 'Detected: CH32V103' however his controller is Ch32x033. 
Could this cause the un-bricking process to not work or otherwise run into issues?

cnlohr, 2024-12-03:
Generally, that just means the link-E didn't autodetect, which could be because of all sorts of things ranging from the SWIO interface being off at the time of the attempted unbrick, to being miswired, to there being too much load on those pins or they are being used as an input.

Rockwell1799, 2024-12-03:
After soldering a new chip to the board and run ./minichlink -d I was able to change PB7 pin as Nrst without any issue.
I think the issue was related to my wch-linkE, because a few days ago I had to reflash it due to some disconnection problems. Anyway, thank you all for your help, especially chlohr.
[image]

cnlohr, 2024-12-03:
_still shouldn't have needed to do that_ but, I am glad you are resolved, I guess we should close this issue?

cnlohr, 2024-12-03:
Actually, I am going to close, anything new should be a new issue.

closed 2024-12-03 by cnlohr