cnlohr/ch32fun #451: DMA1_Channel3_IRQn not firing when using ws2812b_dma_spi_led_driver.h

cad435, 2024-11-23:
Hello there,

Im currently tinkering around with the ws2812b DMA Driver.

For this I made a platformIO Project with the following .ini
```[env:genericCH32V003F4P6]
platform = ch32v
board_build.core = ch32v003
board = genericCH32V003F4P6
framework = arduino
```

It works straight away. The one WS2812B LED gets one set of Color data transmitted.

But than no other transmits are possible.
I debugged it and traced it down to the fact that the `DMA1_Channel3_IRQHandler` is never firing. this causes the code in line 199 `WS2812FillBuffSec( WS2812dmabuff, DMA_BUFFER_LEN / 2, 1 );` never gets called.

Therefore the function `WS2812FillBuffSec` never gets called with `tce = 1`, which in terms at least never ends the transmission 

( `if( tce )` in Line 114 never returns "true" which means the flag WS2812BLEDInUse is never reset )
I also confirmed this with a debugger and also a LED which would light up when the IRQ is fired. 

Unfortunately I now has hit my competence-limit and I do not know why this is happening and how I can fix this issue

This is my test-Code:

```
#include <Arduino.h>

#define WS2812DMA_IMPLEMENTATION
#define WSGRB
//#define WS2812B_ALLOW_INTERRUPT_NESTING
#include "ws2812b_dma_spi_led_driver.h"

static volatile uint32_t i = 0xFF;

#define NRLEDS 30

void setup() {

  pinMode(D2, OUTPUT);
  digitalWrite(D2, LOW);

  Delay_Ms(100);

  WS2812BDMAInit( );
  while( WS2812BLEDInUse );

  Delay_Ms(100);

  WS2812BDMAStart(NRLEDS);
}

void loop() {
  // put your main code here, to run repeatedly: 
    if (!WS2812BLEDInUse)
    {
      digitalWrite(D2, HIGH);
    }

}

uint32_t WS2812BLEDCallback(int ledno)
{
  return 0xFF; //just write some value into the led...
}
```
 
I figured that it might has something todo with enabeling interrup-Nesting, but I I use that macro I'll get a compiler error:

```
In file included from src\main.cpp:6:
src\ws2812b_dma_spi_led_driver.h: In function 'void WS2812BDMAInit()':
src\ws2812b_dma_spi_led_driver.h:272:18: error: '__get_INTSYSCR' was not declared in this scope
  __set_INTSYSCR( __get_INTSYSCR() | 2 ); // Enable interrupt nesting.
                  ^~~~~~~~~~~~~~
src\ws2812b_dma_spi_led_driver.h:272:18: note: suggested alternative: '__get_MTVEC'
  __set_INTSYSCR( __get_INTSYSCR() | 2 ); // Enable interrupt nesting.
                  ^~~~~~~~~~~~~~
                  __get_MTVEC
src\ws2812b_dma_spi_led_driver.h:272:2: error: '__set_INTSYSCR' was not declared in this scope
  __set_INTSYSCR( __get_INTSYSCR() | 2 ); // Enable interrupt nesting.
  ^~~~~~~~~~~~~~
src\ws2812b_dma_spi_led_driver.h:272:2: note: suggested alternative: '__set_MTVEC'
  __set_INTSYSCR( __get_INTSYSCR() | 2 ); // Enable interrupt nesting.
  ^~~~~~~~~~~~~~
  __set_MTVEC
*** [.pio\build\genericCH32V003F4P6\src\main.cpp.o] Error 1
```

I also searched the github repo on the occourances of "__get_INTSYSCR" and " __set_INTSYSCR" but github would only return the WS2812 DMA Library of the only occourance of those symbols. So I don't know where they are from.

Any Help would be greatly appreciated,

 - cad435

maxgerhardt, 2024-11-27:
With that `platformio.ini` you would be running against https://github.com/AlexanderMandera/arduino-wch32v003. This uses an older version of ch32v003fun, over 1 year old to be exact.

I would recommend the following this:
* Wait until @AlexanderMandera has updated their arduino core.
* Rewrite your sketch to work against the regular [OpenWCH Arduino](https://github.com/openwch/ch32v003) by setting `board_build.core = openwch`
* Rewrite your firmware to use the pure `framework = ch32v003fun` framework. This you way you get the most up-to-date ch32v003fun project version and can use the [examples/ws2812bdemo/ws2812bdemo.c](https://github.com/cnlohr/ch32v003fun/blob/master/examples/ws2812bdemo/ws2812bdemo.c) example without problems.

cad435, 2024-11-27:
Hello!

 I didn't realize that the core is so old...

I tinkered around with it more and realized, that (I guess) the real problem lies a bit deeper. In a sense that whatever I'm doing I can't get any interrupts to fire at all.

It always ends up in ch32v003fun.c

```
void DefaultIRQHandler( void )
{
	// Infinite Loop
	asm volatile( "1: j 1b" );
}

```

As this is supposed to be just a "small" christmas gift I'll therefore have to do it with the "primitive" BitBang Driver (which works just fine)

I suppose I could compile against the chv003fun framework, but due to time constraints (well, christmas) this will have to come later.

thank you for pointing it out!

I'll close the issue as this is most likely not a problem of the ch32v003fun code.

closed 2024-11-27 by cad435