cnlohr/ch32fun #458: Inquiry About Support for Additional Models

recalci, 2024-11-30:
I'm curious if there are any plans to support additional models. Some models, although not starting with "ch32," still use the same Qingke series core.
[image]

For example, the CH643's datasheet includes all the content of the CH32X035's datasheet verbatim, along with additional content related to LEDPWM that is not present in the CH32X035.
It would be great to know what’s on the roadmap regarding support for these models. Thank you!

cnlohr, 2024-11-30:
If you have a CH643, pretend it's an x035 - can you try using it and seeing where there's resistance?  It should be pretty easy to bring up more chips!

recalci, 2024-12-01:
I have CH643 chips, and I will start testing them once I get the pcb in a few days. 
I’ve used the CH641 as a CH32V003 and didn’t find any issues so far.

cnlohr, 2024-12-03:
Please let me know what you find!

recalci, 2024-12-05:
I've tested it. Aside from electrical characteristics, LEDPWM, gpio number, remapping, and adc number, they are essentially the same.

cnlohr, 2024-12-06:
It would take some thinking to select appropriate #define's for this... Does it make sense to just #define ch32x035 and go from there or should there be direct support for that in ch32v003fun?

TommyMurphyTM1234, 2024-12-06:
> I'm curious if there are any plans to support additional models. Some models, although not starting with "ch32," still use the same Qingke series core. ![屏幕截图 2024-11-27 153338](https://private-user-images.githubusercontent.com/59434394/391227876-02e642e1-ddd3-4d70-95d5-982b36b2bb21.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM0NzE2NDUsIm5iZiI6MTczMzQ3MTM0NSwicGF0aCI6Ii81OTQzNDM5NC8zOTEyMjc4NzYtMDJlNjQyZTEtZGRkMy00ZDcwLTk1ZDUtOTgyYjM2YjJiYjIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA2VDA3NDkwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg3MzU2YjUyZmI3OGRmM2MyNTRjNjI4ZmY4NWY5ZjQwMWE3NGQ3ZTkxZDNlZjg3MTE0Mzg1ZTE3ZjFhMzBjZjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pNR91sxrewFieR9UDz72UMAzXEdDANbj9e-PpuyEccw)
> 
> For example, the CH643's register map primarily differs from the CH32X035 in that it includes LEDPWM functionality. It would be great to know what’s on the roadmap regarding support for these models. Thank you!

Hi @recalci - do you have a link to the source for that graphic by any chance please? Thanks.

recalci, 2024-12-06:
> Hi @recalci - do you have a link to the source for that graphic by any chance please? Thanks.

Sure! The graphic can be found on the WCH website in Chinese: [https://www.wch.cn/products/productsCenter?categoryId=5](https://www.wch.cn/products/productsCenter?categoryId=5)

TommyMurphyTM1234, 2024-12-09:
> Sure! The graphic can be found on the WCH website in Chinese: https://www.wch.cn/products/productsCenter?categoryId=5

Thanks a lot @recalci - I had looked at their website but failed to find that page. 👍

biemster, 2025-01-05:
I too would love to see support for additional models, especially the CH592 because I'm trying to go vertical in https://github.com/biemster/qible but also because this line of chips is quickly becoming my favorite!

Is there maybe some guide or advise of sorts how to port other chips to this toolchain? I can definitely spend some time on that, and I also realize that on the BLE part I'm likely on my own, at first at least, but that should not be a problem.

Also I've read in another ticket that there is some refactoring going on to split the different chips in separate headers, maybe that's a good moment to keep notes on this?

biemster, 2025-01-28:
> Also I've read in another ticket that there is some refactoring going on to split the different chips in separate headers,

I might have dreamt this, I can't find any reference to this anymore..

cnlohr, 2025-02-07:
I am very excited to now get new models supported!!!! Sorry it took so long for me to rearchitect to make it much simpler to add new models.

cnlohr, 2025-02-08:
If you are going to RE this, there's already some others working on it in my Discord.  If you have Discord, ping me  at `cnlohr` and I can give you an invite.

monte-monte, 2025-11-25:
Since we've added most of the lineup to ch32fun, I think this issue can be closed. Only *old* chip that is missing support for is ch569. But it's expensive and very niche, so unless there is someone really interested in it it will stay like this. ch32h41x support is being added and ch32v205 will probably be supported too once it becomes available.

cnlohr, 2025-11-25:
Thanks @monte-monte

closed 2025-11-25 by cnlohr