cnlohr/ch32fun #469: Can't run programs on some chips

maxpowel, 2024-12-04:
Hello, I successfully compiled and run programs using some boards and barebone chips (just the chip without any other component) I bought at Aliexpress. The chips I bought was CH32V003F4U6 and CH32V003F4P6

Then I made my own board using exactly this CH32V003F4U6 chip https://jlcpcb.com/partdetail/wch_jiangsu_Qin_heng-CH32V003F4U6/C5299908 but the programs does not run. I can flash it, query information, etc. But the programs does not run.
The boad is nothing else but the chip, a capacitor for VCC and some pins exposed to connect dupont cables.

If I run minichlink I get this
```
Found WCH Link
WCH Programmer is LinkE version 2.15
Chip Type: 003
Part Type (A): 0x0010 (This is the capacity code, in KB)
Part UUID    : 45-ad-ab-cd-ad-81-bc-07
PFlags       : ff-ff-ff-ff
Part Type (B): 00-31-05-10
```
I can run all minichlink commands but these two:
```
 -D Configure NRST as GPIO
 -d Configure NRST as NRST
```
(but I need to set NRST as GPIO in my program)

I tried many things:
 * Clock configuration
 * Wch link utility (all chips display the same information)
 * NRST pin disconnected and connected to VCC
 * 3.3V and 5V
 * FUNCONFIG options

I dont know if this matters but the chips that work have `Part Type (B): 00-30-05-00` while the chips from jlcpb have `Part Type (B): 00-31-05-10`
This is the only difference I found between the chips that "works" and ones that don't.

I also tried to debug using "print". With my first chips it worked but it does not with the ones from jlcpb.
I repeated from scratch the same process (compile, flash, debug) on both chips and I dont understand why it works only on some chips but not with the others.

The weird thing is that I already tested it with a CH32V003F4U6 chip so I'm starting to think that the ones from jlcpb are a different version of chips.

What do you think? Should I dump these chips or maybe is there some process I am missing?

Thank you.

ned-pcs, 2024-12-04:
I'm receiving some JLCPCB boards with these chips today and will try them soon.

AlexanderMandera, 2024-12-04:
Can you check the marking on the ICs that work and don't work?
They may be two different batches or revisions.

maxgerhardt, 2024-12-04:
What would be interesting to is whether the program can be debugged or not. Then you can watch execution flow from the Reset handler hopefully into main without ending up in a crash handler. It could just unexpectedly crash at some point in the firmware. Or, it's not able to write the firmware into flash correctly. Minichlink is able to debug the chip, either on the commandline with GDB or inside VSCode. See the https://github.com/cnlohr/ch32v003fun/tree/master/examples/template project for that. (It can also be done if you open the project in VSCode with the PlatformIO extension installed).

ned-pcs, 2024-12-04:
What do you get when you run `minichlink -i`?

maxpowel, 2024-12-05:
> Can you check the marking on the ICs that work and don't work? They may be two different batches or revisions.

Of course, regarding the CH32V003F4U6:
The ones that work: WH V003F4 02D07
The ones that don't work: WH v003F4 12E22

> What would be interesting to know is whether the program can be debugged or not. Then you can watch execution flow from the Reset handler hopefully into main without ending up in a crash handler. It could just unexpectedly crash at some point in the firmware. Or, it's not able to write the firmware into flash correctly. Minichlink is able to debug the chip, either on the commandline with GDB or inside VSCode. See the https://github.com/cnlohr/ch32v003fun/tree/master/examples/template project for that. (It can also be done if you open the project in VSCode with the PlatformIO extension installed).
> 
> Also, minichlink isn't the only program that can flash the CH32V003. You can try it with the official MounRiver IDE that bundles a modified OpenOCD version. It would be interesting to know whether programs built and uploaded that way are able to run on your chip or not. (http://www.mounriver.com/)

Ok I'm going to test this debug. And I'm using VSCode platformio to flash in case this is relevant. I will upload the programs too (basically a hello world).

> What do you get when you run `minichlink -i`?
This command returns `Error: Unknown command i`, probably I have an outdated version. I will try the latest and tell you

cnlohr, 2024-12-06:
@maxpowel Can I pay you to mail me some that do not work?

Please message me on Discord at CNLohr.

But, also, please include the **entire** operation, from the command you are executing to the output of minichlink.

ned-pcs, 2024-12-06:
OK, I just received my boards from JLCPCB with the same CH32V003F4U6 (same LCSC number and markings as @maxpowel ). On mine I'm getting no response at all from the chip (that is: the response from `minichlink` is this, whether or not connected to the board):
```
[ned@Neds-mini wireless_driver/minichlink] % ./minichlink -b -i
Found WCH Link
WCH Programmer is LinkE version 2.12
Already Connected
Full Chip Type Reply: [4] 81-55-01-01-11-00-00-00-01
Detected: CH32V10x
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on setup -9
Could not setup interface.
```
@cnlohr is there something you'd like me to do to check this? Of course I'm going to check continuity, etc. on the board first.

dwillmore, 2024-12-06:
nRST is active low, so it looks like you're holding the chip in reset by grounding that pin.

ned-pcs, 2024-12-06:
OK, after jumpering `PD7 / NRST` to `VCC` I get this from `minichlink -i`:
```
[ned@Neds-mini wireless_driver/minichlink] % ./minichlink -i
Found WCH Link
WCH Programmer is LinkE version 2.12
Full Chip Type Reply: [8] 82-0d-05-09-00-31-05-10-03
Detected: CH32V003
Setup success
Flash Storage: 16 kB
Part UUID    : 41-d2-ab-cd-a9-a6-bc-07
PFlags       : ff-ff-ff-ff
Part Type (B): f8-07-67-13
Read protection: disabled
Interface Setup
USER/RDPR  : e817/5aa5
DATA1/DATA0: ff00/ff00
WRPR1/WRPR0: 00ff/00ff
WRPR3/WRPR2: 00ff/00ff
Flash Size: 16 kB
R32_ESIG_UNIID1: 41d2abcd
R32_ESIG_UNIID2: a9a6bc07
R32_ESIG_UNIID3: ffffffff
```

So I ran `minichlink -D` to get `PD7` back as GPIO.

After doing that, I can debug normally.

So no issues here with these chips, other than that they apparently come from the factory with `PD7` set as `NRST`.

cnlohr, 2024-12-06:
You should always be able to do `minichlink -u` to unbrick those... BUT not if NRST is grounded.

I believe the documented default is for PD7 == NRST.  So it is unusual that you have some where that is not the case.

ned-pcs, 2024-12-06:
We still haven't heard back from the OP who had problems with these chips.

cnlohr, 2024-12-07:
oops my b

maxpowel, 2024-12-07:
Hello to all, it's a little embarrassing but I downloaded the `ch32v003fun` git repository from scratch, compiled `minichlink` again and now it works like a charm.

Now I can configure NRST pin as GPIO without any problem and the uploaded program runs as expected.
Don't know why I did not try this before (maybe because it worked with my initial chips). I thinks that there is something "different" between the chips but definitely `minichlink` can handle it without any problem.

I have tested it also in a little more complex board (basically a RGB PWM led controller with some mosfets, this is why I need NRST as GPIO) and a second version for ws2812b led strips and everything is ok.

I will publish it in my github and release it under a free license both software and hardware design so in case anyone need more details I can provide anything I have.

In case this is useful, the `minichlnk -i` output is:

```
Found WCH Link
WCH Programmer is LinkE version 2.15
Full Chip Type Reply: [8] 82-0d-05-09-00-31-05-10-03
Detected: CH32V003
Setup success
Flash Storage: 16 kB
Part UUID    : 45-bd-ab-cd-ad-91-bc-07
PFlags       : ff-ff-ff-ff
Part Type (B): 00-31-05-10
Read protection: disabled
Interface Setup
USER/RDPR  : 00ff/5aa5
DATA1/DATA0: 00ff/00ff
WRPR1/WRPR0: 00ff/00ff
WRPR3/WRPR2: 00ff/00ff
Flash Size: 16 kB
R32_ESIG_UNIID1: 45bdabcd
R32_ESIG_UNIID2: ad91bc07
R32_ESIG_UNIID3: ffffffff
```
The debug an basically everything is working fine now.
The conclusion is that everything was fine from the beginning and I was just a little clumsy.

Thank you so much for your help!

closed 2024-12-07 by maxpowel