cnlohr/ch32fun #474: ADC read in interrupt mode is not working properly. Need help

RhonyD, 2024-12-12:
Here is my code:-
[image]

The problem is it's Showing the ADC value just after starting the program then it's showing the "ERROR" message in the while loop. What mistake am I doing? Need suggestion.

TommyMurphyTM1234, 2024-12-12:
Doesn't your code have an inherent race condition with regard to `adcFlag` getting set to 1? Perhaps your `while` loop simply iterates before the interrupt handler has had a chance to set `adcFlag` to 1? If you're going to use this sort of handshaking between the main loop and an interrupt handler then you probably need to use a different approach and ideally atomic set/clear operations.

RhonyD, 2024-12-12:
The adcFlag is set when conversion is complete and this flag is cleared in main loop. where is the problem I am unable to find the problem.

TommyMurphyTM1234, 2024-12-12:
The setting and clearing of `adcFlag` is almost certainly not an atomic operation and there is nothing to enforce a critical section on access to flag by the two "threads" (main loop and interrupt handler).

- https://en.wikipedia.org/wiki/Critical_section

maxgerhardt, 2024-12-12:
You need 
```cpp
volatile u8 adcFlag = 0;
volatile u16 ADC_VALUE = 0;
```
since these are variables that normal code and interrupt code shares. The compiler otherwise optimizes away reading the adcFlag multiple times.

https://docs.arduino.cc/language-reference/en/variables/variable-scope-qualifiers/volatile/

cnlohr, 2024-12-12:
Also @RhonyD in the future, please do not use code-to-image conversions in github issues. It prevents search indexing and copy-pasting.

RhonyD, 2024-12-13:
@maxgerhardt,@cnlohr,@TommyMurphyTM1234  thanks to all of you for your suggestions and guidance. By the way my issue is solved. I have tried adcFlag and ADC_VALUE with volatile key word and without it,and things worked perfectly. My final code is as follows:-

#include "ch32v003fun.h"
#include <stdio.h>
#include "GPIO_CONFIG.h"
#include "I2C.h"
#include "I2C_LCD.h"
#include "IT_CONFIG.h"
#include "ADC_CONFIG.h"

#define LED_PORT GPIO_PORT
#define LED_PIN 2

u8 adcFlag = 0;
u16 voltage = 0;

#if ADC_INJECTION
u16 ADC_VAL2 = 0, ADC_VAL0 = 0, ADC_VAL1 = 0;
#endif

#if ADC_POLL || ADC_INTERRUPT
u16 ADC_VAL = 0;
#define ADC_CHANNEL 1
#endif

void TIMER2_INIT(u16 psclr, u16 atlr)
{

	RCC->APB1PCENR |= (RCC_APB1Periph_TIM2);
	TIM2->CTLR1 |= TIM_CKD_DIV1;
	TIM2->PSC = psclr;
	TIM2->ATRLR = atlr;
	TIM2->SWEVGR |= TIM_UG;
	TIM2->DMAINTENR |= (TIM_IT_Update);
	TIM2->CTLR1 |= TIM_CEN;
}

void TIM2_IRQHandler()
{

	if (TIM2->INTFR & (TIM_IT_Update))
	{
		// TOGGLE_GPIO(LED_PIN);
		TIM2->INTFR &= ~(u16)TIM_IT_Update;
	}
}

#if ADC_INTERRUPT
void ADC1_IRQHandler()
{

	if (ADC1->CTLR1 & (1 << 5))
	{
		adcFlag = 1;
		ADC_VAL = ADC1->RDATAR;
	}
	/*
	if (ADC1->STATR& ADC_FLAG_EOC)
	{
		adcFlag = 1;
		ADC_VAL = ADC1->RDATAR;
	}
	*/
}
#endif

int main()
{
	SystemInit();
	GPIO_INIT(LED_PIN);
[... 247 lines cut ...]
    ADC1->RSQR2 = 0;
    ADC1->RSQR3 = ch; // 0-9 for 8 ext inputs and two internals

    // set sampling time for chl 7
    ADC1->SAMPTR2 &= ~(ADC_SMP0 << (3 * ch));
    ADC1->SAMPTR2 |= 7 << (3 * ch); // 0:7 => 3/9/15/30/43/57/73/241 cycles

    // turn on ADC and set rule group to sw trig

    ADC1->CTLR2 |= ADC_ADON | ADC_EXTSEL;
    ADC1->CTLR1 |= ADC_IT_EOC;
    ADC1->CTLR2 |= ADC_SWSTART;

    // Reset calibration
    ADC1->CTLR2 |= ADC_RSTCAL;
    while (ADC1->CTLR2 & ADC_RSTCAL);
        

    // Calibrate
    ADC1->CTLR2 |= ADC_CAL;
    while (ADC1->CTLR2 & ADC_CAL);
        
}

#endif

maxgerhardt, 2024-12-13:
Can you edit your last post to put that code in a tripple-backtick C code block? https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

RhonyD, 2024-12-13:
'''
#include "ch32v003fun.h"
#include <stdio.h>
#include "GPIO_CONFIG.h"
#include "I2C.h"
#include "I2C_LCD.h"
#include "IT_CONFIG.h"
#include "ADC_CONFIG.h"

#define LED_PORT GPIO_PORT
#define LED_PIN 2

u8 adcFlag = 0;
u16 voltage = 0;

#if ADC_INJECTION
u16 ADC_VAL2 = 0, ADC_VAL0 = 0, ADC_VAL1 = 0;
#endif

#if ADC_POLL || ADC_INTERRUPT
u16 ADC_VAL = 0;
#define ADC_CHANNEL 1
#endif

void TIMER2_INIT(u16 psclr, u16 atlr)
{

	RCC->APB1PCENR |= (RCC_APB1Periph_TIM2);
	TIM2->CTLR1 |= TIM_CKD_DIV1;
	TIM2->PSC = psclr;
	TIM2->ATRLR = atlr;
	TIM2->SWEVGR |= TIM_UG;
	TIM2->DMAINTENR |= (TIM_IT_Update);
	TIM2->CTLR1 |= TIM_CEN;
}

void TIM2_IRQHandler()
{

	if (TIM2->INTFR & (TIM_IT_Update))
	{
		// TOGGLE_GPIO(LED_PIN);
		TIM2->INTFR &= ~(u16)TIM_IT_Update;
	}
}

#if ADC_INTERRUPT
void ADC1_IRQHandler()
{

	if (ADC1->CTLR1 & (1 << 5))
	{
		adcFlag = 1;
		ADC_VAL = ADC1->RDATAR;
		ADC1->CTLR2 |= ADC_SWSTART;
	}
	/*
	if (ADC1->STATR& ADC_FLAG_EOC)
	{
		adcFlag = 1;
		ADC_VAL = ADC1->RDATAR;
	}
	*/
}
#endif

int main()
{
	SystemInit();
	GPIO_INIT(LED_PIN);
[... 262 lines cut ...]
    // turn on ADC and set rule group to sw trig

    ADC1->CTLR2 |= ADC_ADON | ADC_EXTSEL;
    ADC1->CTLR1 |= ADC_IT_EOC;
    ADC1->CTLR2 |= ADC_SWSTART;
    ADC1->CTLR1 &= ~(uint32_t)(3 << 25);
#if CAL_VOL == 50
    ADC1->CTLR1 |= ADC_CALVOL_50PERCENT;
#endif
#if CAL_VOL == 75
    ADC1->CTLR1 |= ADC_CALVOL_75PERCENT;
#endif
    // Reset calibration
    ADC1->CTLR2 |= ADC_RSTCAL;
    while (ADC1->CTLR2 & ADC_RSTCAL)
        ;

    // Calibrate
    ADC1->CTLR2 |= ADC_CAL;
    while (ADC1->CTLR2 & ADC_CAL)
        ;
}

#endif
'''

RhonyD, 2024-12-13:
@[maxgerhardt](https://github.com/maxgerhardt) I have tried to do as per your instructions. Please see if this is what you wanted.

maxgerhardt, 2024-12-13:
>and things worked perfectly. My final code is as follows:-

Okay, so the whole program is working? I'm still of the opinion that it's safer and correct to use `volatile` for variables shared with interrupts; It working may just be the compiler happening to emit the right code.

RhonyD, 2024-12-13:
@maxgerhardt, yes you are right. This is just to inform you that the compiler is working fine. I will certainly use the volatile keyword. Thanks for your suggestion and guidance.

TommyMurphyTM1234, 2024-12-13:
> > and things worked perfectly. My final code is as follows:-
> 
> Okay, so the whole program is working? I'm still of the opinion that it's safer and correct to use `volatile` for variables shared with interrupts; It working may just be the compiler happening to emit the right code.

I agree. If the code is compiled with optimisation enabled the behaviour may change. It may just be luck that the non-optimised version works right now. I also still think that the atomicity and critical section issues that I mentioned above should be addressed for similar reasons. As things stand the code doesn't look safe to me from a multiple "thread" point of view.

RhonyD, 2024-12-13:
> > > and things worked perfectly. My final code is as follows:-
> > 
> > 
> > Okay, so the whole program is working? I'm still of the opinion that it's safer and correct to use `volatile` for variables shared with interrupts; It working may just be the compiler happening to emit the right code.
> 
> I agree. If the code is compiled with optimisation enabled the behaviour may change. It may just be luck that the non-optimised version works right now. I also still think that the atomicity and critical section issues that I mentioned above should be addressed for similar reasons. As things stand the code doesn't look safe to me from a multiple "thread" point of view. But none of these issues are specific to the ch32v003fun project/framework - they are general computer science/coding issues.

Yes Sir you are right. that's why I have already impelmented the volaite keyword in the code above. By the way thanks for taking time to read the issue and comment.