cnlohr/ch32fun #475: optimization for size in vscode

RhonyD, 2024-12-14:
I am new to VS code, but I have developed avr mircocontoller in eclipse where there is an option to set optimization. But I haven't seen such option in VS code. Please guide on optimization.
[image]

TommyMurphyTM1234, 2024-12-14:
You should probably ask about that here or wherever the VSCode extensions that you're using come from: 

- https://github.com/Community-PIO-CH32V/platform-ch32v

prosper00, 2024-12-14:
Generally, you need to exit the 'CFLAGS' line in ch32v003fun.mk:

CFLAGS?=-g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8

Right now, you can see that optimization is defaulted to '-Os' (size).

VScode wont modify this typically. No idea how/if platformio works

cnlohr, 2024-12-16:
Let us know by closing the issue if you're solved, @RhonyD

RhonyD, 2024-12-17:
No I am unable to figure it out. if any screen shot is provided then it would be helpful. can't find ch32v003fun.mk in any of my projects. it is there only in the github page in examples section.

TommyMurphyTM1234, 2024-12-17:
> No I am unable to figure it out. if any screen shot is provided then it would be helpful. can't find ch32v003fun.mk in any of my projects. it is there only in the github page in examples section.

If there's no `ch32v003fun.mk` in your project then it sounds like you're not using ch32v003fun at all and, as I said previously, you need to direct your questions to the originators of whatever you are using?

It might help if you clarified what actually is in your project.

RhonyD, 2024-12-20:
ok I have found the ch32v003fun.mk file. it's there in /home/abc/.platformio/packages/framework-ch32v003fun. And it's already set to optimize for size mode. Thanks for the guidance. Special thanks to @prosper00 .

closed 2024-12-20 by RhonyD