cnlohr/ch32fun #479: Problem with adding CH32x035 example (cap_touch_adc)

yves5141, 2024-12-21:
Hi,
I'm trying to make some of the examples work on a CH32x035.
But at the moment I'm struggling with the cap_touch_adc example.
I changed line 54 in ch32v003_touch.h, since CH32x035 has a 64-bit counter for SysTick.
For testing, I changed cap_touch_adc.c, in order to get values only from PA0.
However, the program crashes. I did some debugging, the code works until it gets to line 87 in ch32v003_touch.h, switching to assembly, it happens at 0x00000734:

`0x00000734: 1c 44           	lw	a5,8(s0)
0x00000736: 5c 44           	lw	a5,12(s0)
0x00000738: f9 bf           	j	0x716 <main+30>`

I'm not sure, how to solve this.
BTW: Not directly related, but I couldn't find a document for the CH32x035 containing the instruction set and listing all the registers. Where can I find that?

ch32v003_touch.h

`#ifndef _CH32V003_TOUCH_H
#define _CH32V003_TOUCH_H

/** ADC-based Capactive Touch Control.

	see cap_touch_adc.c for an example.

	// Enable GPIOD, C and ADC
	RCC->APB2PCENR |= RCC_APB2Periph_GPIOA | RCC_APB2Periph_GPIOD | RCC_APB2Periph_GPIOC | RCC_APB2Periph_ADC1;
	InitTouchADC();

	// Then do this any time you want to read some touches.
	sum[0] += ReadTouchPin( GPIOA, 2, 0, iterations );
	sum[1] += ReadTouchPin( GPIOA, 1, 1, iterations );
	sum[2] += ReadTouchPin( GPIOC, 4, 2, iterations );
	sum[3] += ReadTouchPin( GPIOD, 2, 3, iterations );
	sum[4] += ReadTouchPin( GPIOD, 3, 4, iterations );
	sum[5] += ReadTouchPin( GPIOD, 5, 5, iterations );
	sum[6] += ReadTouchPin( GPIOD, 6, 6, iterations );
	sum[7] += ReadTouchPin( GPIOD, 4, 7, iterations );
*/

#define TOUCH_ADC_SAMPLE_TIME 2  // Tricky: Don't change this without a lot of experimentation.

// Can either be 0 or 1.
// If 0: Measurement low and rises high.  So more pressed is smaller number.
// If 1: Higher number = harder press. Good to pair with TOUCH_FLAT.
// If you are doing more prox, use mode 0, otherwise, use mode 1.
#define TOUCH_SLOPE     1

// If you set this to 1, it will glitch the line, so it will only read
// anything reasonable if the capacitance can overcome that initial spike.
// Typically, it seems if you use this you probbly don't need to do
// any pre-use calibration.
#define TOUCH_FLAT      0

// Macro used for force-alingining ADC timing
#define FORCEALIGNADC \
	asm volatile( \
		"\n\
		.balign 4\n\
		andi a2, %[cyccnt], 3\n\
		c.slli a2, 1\n\
		c.addi a2, 12\n\
		auipc a1, 0\n\
		c.add  a2, a1\n\
		jalr a2, 1\n\
		.long 0x00010001\n\
		.long 0x00010001\n\
		"\
		:: [cyccnt]"r"(SysTick->CNTL) : "a1", "a2"\
	);

static void InitTouchADC( );
[... 207 lines cut ...]

		uint64_t start = SysTick->CNTL;
		start = start | ((uint64_t)SysTick->CNTH << 32);

		// Sampling all touch pads, 3x should take 6030 cycles, and runs at about 8kHz

		int iterations = 3;/*
		sum[0] += ReadTouchPin( GPIOA, 2, 0, iterations );
		sum[1] += ReadTouchPin( GPIOA, 1, 1, iterations );
		sum[2] += ReadTouchPin( GPIOC, 4, 2, iterations );
		sum[3] += ReadTouchPin( GPIOD, 2, 3, iterations );
		sum[4] += ReadTouchPin( GPIOD, 3, 4, iterations );
		sum[5] += ReadTouchPin( GPIOD, 5, 5, iterations );
		sum[6] += ReadTouchPin( GPIOD, 6, 6, iterations );
		sum[7] += ReadTouchPin( GPIOD, 4, 7, iterations );
		*/
		sum[0] += ReadTouchPin( GPIOA, 0, 0, iterations );

		uint64_t end = SysTick->CNTL;
		end = end | ((uint64_t)SysTick->CNTH << 32);
	
		printf( "%d %d %d %d %d %d %d %d %d\n", (int)sum[0], (int)sum[1], (int)sum[2],
			(int)sum[3], (int)sum[4], (int)sum[5], (int)sum[6], (int)sum[7], (int)(end-start) );
	}
}`

TommyMurphyTM1234, 2024-12-21:
> Hi, I'm trying to make some of the examples work on a CH32x035G8U6. But at the moment I'm struggling with the cap_touch_adc example. I changed line 54 in ch32v003_touch.h, since CH32x035 has a 64-bit counter for SysTick. For testing, I changed cap_touch_adc.c, in order to get values only from PA0. However, the program crashes and jumps to line 767 in ch32v003fun.c (`while( !DebugPrintfBufferFree() );`). I did some debugging, the code works until it gets to line 87 in ch32v003_touch.h, switching to assembly, it happens at 0x00000734:
> 
> ```
> 0x00000734: 1c 44           	lw	a5,8(s0)
> 0x00000736: 5c 44           	lw	a5,12(s0)
> 0x00000738: f9 bf           	j	0x716 <main+30>
> ```

What happens when it tries to execute the `lw` instruction at `0x00000734`?
Does it trap or something?
If so then what are the values of CSRs such as `mcause`, `mepc`, `mtval` etc.?

yves5141, 2024-12-21:
Of course, sorry I forgot: `MSTATUS:00000088 MTVAL:300027f3 MCAUSE:00000002 MEPC:00000216`

Right after executing `0x00000734` it jumps to `while( !DebugPrintfBufferFree() );`, prints the values of CSRs (above) and then goes to 
```
	// Infinite Loop
	asm volatile( "1: j 1b" );
```

TommyMurphyTM1234, 2024-12-21:
> Of course, sorry I forgot: `MSTATUS:00000088 MTVAL:300027f3 MCAUSE:00000002 MEPC:00000216`

As far as I can see...

- `mcause = 0x00000002` [means illegal instruction](https://five-embeddev.com/riscv-priv-isa-manual/Priv-v1.12/machine.html#sec:mcause)
- `mepc = 0x00000216` [is the address of the offending instruction](https://five-embeddev.com/riscv-priv-isa-manual/Priv-v1.12/machine.html#machine-exception-program-counter-mepc)
- `mtval = 0x300027f3` is the illegal instruction which [decodes](https://luplab.gitlab.io/rvcodecjs/#q=0x300027f3&abi=false&isa=AUTO) as `csrrs x15, mstatus, x0`

yves5141, 2024-12-21:
I had a similiar issue when trying to use parts the of systick_irq example in one of my projects, it crashed when calling `NVIC_EnableIRQ(SysTicK_IRQn);`. However, that problem was solved easily by adding `extern "C" { ... }` in my .cpp file.

TommyMurphyTM1234, 2024-12-22:
> I had a similiar issue when trying to use parts the of systick_irq example in one of my projects, it crashed when calling `NVIC_EnableIRQ(SysTicK_IRQn);`. However, that problem was solved easily by adding `extern "C" { ... }` in my .cpp file.

I don't see how that would help or how that's relevant here.

Did you check your disassembly against what I posted above?
Does your program have the `csrrs x15, mstatus, x0` instruction at address `0x00000216` and is this actually where the fault is occurring?
If it does then somehow your target is seeing `csrrs x15, mstatus, x0` as an illegal instruction.

TommyMurphyTM1234, 2024-12-22:
The odd thing is that I cannot see any `csrrs` instructions at all in the core ch32v003fun or cap_touch_adc example program source files:
```
git clone https://github.com/cnlohr/ch32v003fun
cd ch32v003fun

grep -i csrrs ch32v003fun/*
# Nothing

grep -i csrrs examples/cap_touch_adc/*
# Nothing
```
There are some `csrr` instructions:
```
grep -i csrr ch32v003fun/*
ch32v003fun/ch32v003fun.h:              "csrr %0," "mstatus": "=r"(result));
ch32v003fun/ch32v003fun.h:              "csrr %0," "mstatus": "=r"(result));
ch32v003fun/ch32v003fun.h:    "csrr %0," "mstatus": "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0, 0x804": "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0," "fflags" : "=r" (result) );
ch32v003fun/ch32v003fun.h:      "csrr %0," "frm" : "=r" (result) );
ch32v003fun/ch32v003fun.h:      "csrr %0," "fcsr" : "=r" (result) );
ch32v003fun/ch32v003fun.h:      "csrr %0," "mstatus": "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0,""misa" : "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0," "mtvec": "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0," "mscratch" : "=r"(result));
ch32v003fun/ch32v003fun.h:      __ASM volatile("csrr %0," "mepc" : "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0," "mcause": "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0," "mtval" : "=r" (result) );
ch32v003fun/ch32v003fun.h:      "csrr %0,""mvendorid": "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0,""marchid": "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0,""mimpid": "=r"(result));
ch32v003fun/ch32v003fun.h:      "csrr %0,""mhartid": "=r"(result));

grep -i csrr examples/cap_touch_adc/*
# Nothing
```

yves5141, 2024-12-22:
Well, there is no csrrs in the disassembled file. At address `0x00000216` is:
`216:	300027f3          	csrr	a5,mstatus`

The strange thing is, during debugging, it's executing 
`0x00000732: e7 80 20 8d     	jalr	-1838(ra) # 0x20000000 <ReadTouchPin>`, 
but it doesn't seem to perform the jump, next thing it does is executing 
`0x00000736: 1c 44           	lw	a5,8(s0)`
This address does not correspond with the address in the disassembled file. There it is `61a`
`61a:	441c                	lw	a5,8(s0)`

[firmeware.txt](https://github.com/user-attachments/files/18223271/firmeware.txt)

TommyMurphyTM1234, 2024-12-22:
> Well, there is no csrrs in the disassembled file. At address `0x00000216` is: `216: 300027f3 csrr a5,mstatus`

`csrr` is a pseudoinstruction that is actually encoded as `csrrs`.

[image]

> The strange thing is, during debugging, it's executing `0x00000732: e7 80 20 8d jalr -1838(ra) # 0x20000000 <ReadTouchPin>`, but it doesn't seem to perform the jump, next thing it does is executing [...]

Are you sure that you're compiling the program with something like `-g -Og` in order to facilitate debugging and accurate line/instruction tracking?

yves5141, 2024-12-23:
I added `g -Og -v` to the flags. The address changed, but the behaviour is the same.
`DEAD MSTATUS:00000088 MTVAL:300027f3 MCAUSE:00000002 MEPC:20000006`
When running the code step-by-step, the program halts at 
`736:	441c                	lw	a5,8(s0)`
and then is handling the exception and throwing the informations above.

[firmeware.txt](https://github.com/user-attachments/files/18227796/firmeware.txt)

yves5141, 2024-12-24:
Found this thread: [https://github.com/eugene-tarassov/vivado-risc-v/issues/181](url)
Might not be directly related to my problem, but this user has a similiar issue when trying to read CSR with mstatus. In his case, the problem is, he is running an OS so he has not enough privileges for mstatus and has to use ustatus.
However, this should not apply to my problem, since I'm running my code bare metal. How does the CPU decide, if code is running in machine mode or usermode?
Replacing mstatus with ustatus avoid mcause 2, but it leads to mcause 6, but this has something to do with the rest of the code which depends on working interrupts. Trying to access CSR with ustatus does nothing, I tried and it is not possible to change any values in CSR using ustatus.

yves5141, 2024-12-24:
> I presume that that's not relevant in your case? (Does the CH32x035 MCU even support modes other than machine mode? I don't know and can't seem to access the WCH datasheet page lately).

Good question. Since the Ch32x035 has a PMP, it seems it can work in different modes. See reference manual on page 51.
However, when using the ch32v003fun framework, everything should work in machine mode, right?

> If the `csrrs x15, mstatus, x0` is definitely the instruction that is causing the illegal instruction exception then you need to figure out why this is the case.

Well, I'm working on it.

[CH32X035DS0.PDF](https://github.com/user-attachments/files/18240385/CH32X035DS0.PDF)
[CH32X035RM.PDF](https://github.com/user-attachments/files/18240387/CH32X035RM.PDF)

yves5141, 2024-12-25:
I ended up adding:
`
#if defined ( CH32X03x )
asm volatile(
"	li t0, 0x1888\n\
	csrw mstatus, t0\n\
");
#endif
`
to ch32v003fun.c in the `handle_reset()` method. According to [https://github.com/WuxiProject-offical/CH32X035-HelperLibrary/blob/main/startup_file/startup_ch32x035_highcode_machine.S](url) this enters machine mode and enables global interrupts.

Seems to work, at least I don't get `MCAUSE:00000002`, now I get `MCAUSE:00000006` when leaving the for-loop in ch32v003_touch.h for some reason, but this is another problem.

TommyMurphyTM1234, 2024-12-25:
According to [https://github.com/WuxiProject-offical/CH32X035-HelperLibrary/blob/main/startup_file/startup_ch32x035_highcode_machine.S](url) this enters machine mode and enables global interrupts.

Link doesn't work. I presume that you meant this? 

- https://github.com/WuxiProject-offical/CH32X035-HelperLibrary/blob/e6ebd99c2cd487d9bceac6caabe9a09ee0ef2207/startup_file/startup_ch32x035_highcode_machine.S#L232-L235

TommyMurphyTM1234, 2024-12-25:
I doubt that the device boots into anything other than machine mode and, in any case, the startup/reset code in question is writing to `mstatus` which can only be done in machine mode so I'm not sure that the analysis/change here is actually adequately analysed.

FWIW this is what the WCH EVT does at startup: 

- https://github.com/openwch/ch32x035/blob/ed56d28751ca2749db1f829307b434e4682b7de4/EVT/EXAM/SRC/Startup/startup_ch32x035.S#L214-L216
```
/* Enable global interrupt and configure privileged mode */
   	li t0, 0x88           
   	csrw mstatus, t0
```

yves5141, 2024-12-25:
> I doubt that the device boots into anything other than machine mode and, in any case, the startup/reset code in question is writing to `mstatus` which can only be done in machine mode so I'm not sure that the analysis/change here is actually thoroughly understood.

Well, I can only draw conclusions from what the official documentation tells me. Assumptions lead nowhere.
Anyway, I don't claim my solution to be the best solution. Maybe there is something else missing.

> FWIW this is what the WCH EVT does at startup:
>
>     * https://github.com/openwch/ch32x035/blob/ed56d28751ca2749db1f829307b434e4682b7de4/EVT/EXAM/SRC/Startup/startup_ch32x035.S#L214-L216
>
>
> ```
> /* [...]

But where is the ch32v003fun framework doing this?
If the reference manual is right, the standard mode is user mode. If that's the case, there are areas in flash, which can execute code in machine mode by default. The reset handler seems to belong to this area, since there CSR instructions can be executed without raising exceptions.

> According to [https://github.com/WuxiProject-offical/CH32X035-HelperLibrary/blob/main/startup_file/startup_ch32x035_highcode_machine.S](url) this enters machine mode and enables global interrupts. [...]

That's it.

cnlohr, 2024-12-26:
0. Have you enabled interrupts in this project or not?  If not at all, then yes, let's dig in more.  If so, please disable them and debug from there.

1. If you are going to use the examples, please do not use C++.  You can use C++ only after you have all the basic stuff working.  You must take care when trying to use interrupts in C++.

2. I haven't tested cap_touch_adc on the x035.  It wasn't specifically designed to work with it, but should in principle.

@TommyMurphyTM1234 Re: mstatus, we set it right here: https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.c#L1012-L1020 - what is the issue there?

In general, why are there so many links here to WuxiProject? 

I am available now, but I am really confused, what is the current issue you are having?

TommyMurphyTM1234, 2024-12-27:
> @TommyMurphyTM1234 Re: mstatus, we set it right here: https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.c#L1012-L1020 - what is the issue there?

I haven't asserted that there's any issue with what ch32v003fun does with `mstatus`.

> but I am really confused

Me too. :⁠-⁠(

cnlohr, 2024-12-29:
@yves5141  I still can't make heads or tails of what you're saying. ch32v003fun performs this operation https://github.com/cnlohr/ch32v003fun/blob/master/ch32v003fun/ch32v003fun.c#L1012-L1020 

I don't think you're looking at ch32v003fun.

yves5141, 2025-01-21:
In general, there are no problems using interrupts or something like this. The cap_touch_exti example is working just fine.
Unfortunately, I don't have much time left at the moment, but I will look at this later, again.

cnlohr, 2025-04-01:
ONE OTHER NOTE.  Just FYI ... the cap_touch_adc is intended for use with the ch32v003, and likely will not work on the x035 because it uses the ADC in ADC mode.