cnlohr/ch32fun #482: if I compile the original blink it doesn't work

DanSkareda, 2025-01-01:
When I upload an originally compiled BIN file to ch32v003 using minichlink and restart ch32v003, the LED flashes. It is ok. But when I compile the original blink source codes using "make" to get a BIN file, this is what I get:

_make
riscv64-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32V003 -DMCU_PACKAGE= -DTARGET_MCU_LD=0 -DTARGET_MCU_MEMORY_SPLIT= ../../ch32v003fun/ch32v003fun.ld > ../../ch32v003fun/generated_ch32v003.ld
riscv64-unknown-elf-gcc -o blink.elf ../../ch32v003fun/ch32v003fun.c blink.c -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall -Wl,--print-memory-usage -Wl,-Map=blink.map -L../../ch32v003fun/../misc -lgcc -T ../../ch32v003fun/generated_ch32v003.ld -Wl,--gc-sections
Memory region Used Size Region Size %age Used
 FLASH: 1752 B 16 KB 10.69%
 RAM: 0 GB 2 KB 0.00%
riscv64-unknown-elf-objdump -S blink.elf > blink.lst
riscv64-unknown-elf-objcopy -O binary blink.elf blink.bin
riscv64-unknown-elf-objcopy -O ihex blink.elf blink.hex
make -C ../../ch32v003fun/../minichlink all
make[1]: Entering directory "/home/dan/ch32v003fun/minichlink"
make[1]: Nothing will be done for "all".
make[1]: Leaving "/home/dan/ch32v003fun/minichlink"
../../ch32v003fun/../minichlink/minichlink -w blink.bin flash -b
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
make: *** [../../ch32v003fun/ch32v003fun.mk:244: cv_flash] Error 224_

The error occurs because I am compiling on a machine that does not have a ch32 programmer (esp32s2) connected to it. When I then upload the blink.bin file using minichlink (the same way I uploaded the original blink.bin file) to the ch32v003, the led does not blink after reset. The original blink.bin is about 560 bytes, while my compiled one is 1752 bytes. Please advise why the compilation is not working properly? I upload it like this:
./minichlink -w blink.bin flash

TommyMurphyTM1234, 2025-01-01:
> ```
> Memory region         Used Size  Region Size  %age Used
>            FLASH:        1752 B        16 KB     10.69% 
>              RAM:           0 B         2 KB      0.00%
> ```

Yeah - I built the toolchain from scratch from the `riscv-gnu-toolchain` repo and got this:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:        1904 B        16 KB     11.62%
             RAM:           0 B         2 KB      0.00%
```
At a glance I can't see what in terms of recent PR activity might have caused this to happen...

In case any of this gives any clues...
```
riscv64-unknown-elf-size blink.elf
   text    data     bss     dec     hex filename
   1904       0       0    1904     770 blink.elf

riscv64-unknown-elf-nm --print-size --size-sort --reverse-sort blink.elf
0000046c 000001bc t mini_vpprintf
000002b2 000000e4 t _write.constprop.0
000001e8 000000c0 t mini_itoa.part.0
0000069e 0000009c T main
00000000 0000009c T InterruptVectorDefault
00000000 0000009c T InterruptVector
0000011e 0000008a t mini_pad
000003aa 0000007a T handle_reset
000000a0 00000058 T __divsi3
0000064a 00000054 T SystemInit
000001a8 00000040 t _puts
000000a8 0000002c T __udivsi3
000000a8 0000002c T __hidden___udivsi3
00000432 00000028 T WWDG_IRQHandler
00000432 00000028 T USB_LP_CAN1_RX0_IRQHandler
00000432 00000028 T USB_HP_CAN1_TX_IRQHandler
00000432 00000028 T USBWakeUp_IRQHandler
00000432 00000028 T USBPD_WKUP_IRQHandler
00000432 00000028 T USBPD_IRQHandler
00000432 00000028 T USBHS_IRQHandler
00000432 00000028 T USBHSWakeUp_IRQHandler
00000432 00000028 T USBFS_WakeUp_IRQHandler
00000432 00000028 T USBFS_IRQHandler
00000432 00000028 T USBFSWakeUp_IRQHandler
00000432 00000028 T USART4_IRQHandler
00000432 00000028 T USART3_IRQHandler
00000432 00000028 T USART2_IRQHandler
00000432 00000028 T USART1_IRQHandler
00000432 00000028 T UART8_IRQHandler
00000432 00000028 T UART7_IRQHandler
00000432 00000028 T UART6_IRQHandler
00000432 00000028 T UART5_IRQHandler
00000432 00000028 T UART4_IRQHandler
00000432 00000028 T TIM9_UP_IRQHandler
00000432 00000028 T TIM9_TRG_COM_IRQHandler
00000432 00000028 T TIM9_CC_IRQHandler
00000432 00000028 T TIM9_BRK_IRQHandler
00000432 00000028 T TIM8_UP_IRQHandler
00000432 00000028 T TIM8_TRG_COM_IRQHandler
00000432 00000028 T TIM8_CC_IRQHandler
00000432 00000028 T TIM8_BRK_IRQHandler
00000432 00000028 T TIM7_IRQHandler
00000432 00000028 T TIM6_IRQHandler
00000432 00000028 T TIM5_IRQHandler
00000432 00000028 T TIM4_IRQHandler
00000432 00000028 T TIM3_IRQHandler
00000432 00000028 T TIM2_TRG_IRQHandler
00000432 00000028 T TIM2_IRQHandler
00000432 00000028 T TIM2_CC_IRQHandler
[... 64 lines cut ...]
00000432 00000028 T DMA1_Channel5_IRQHandler
00000432 00000028 T DMA1_Channel4_IRQHandler
00000432 00000028 T DMA1_Channel3_IRQHandler
00000432 00000028 T DMA1_Channel2_IRQHandler
00000432 00000028 T DMA1_Channel1_IRQHandler
00000432 00000028 T CAN2_TX_IRQHandler
00000432 00000028 T CAN2_SCE_IRQHandler
00000432 00000028 T CAN2_RX1_IRQHandler
00000432 00000028 T CAN2_RX0_IRQHandler
00000432 00000028 T CAN1_SCE_IRQHandler
00000432 00000028 T CAN1_RX1_IRQHandler
00000432 00000028 T Break_Point_Handler
00000432 00000028 T BB_IRQHandler
00000432 00000028 T AWU_IRQHandler
00000432 00000028 T ADC1_IRQHandler
00000432 00000028 T ADC1_2_IRQHandler
000000f8 00000024 T __modsi3
00000628 00000022 W printf
00000396 00000014 t __puts_uart
0000045a 00000012 W strlen
00000424 0000000e T NMI_RCC_CSS_IRQHandler
00000424 0000000e T NMI_Handler
000002a8 0000000a t internal_handle_input.constprop.0
000000d4 00000008 T __umodsi3
```

TommyMurphyTM1234, 2025-01-01:
For a previous PR I checked the size details of all of the examples available at the time - see here:

- https://github.com/cnlohr/ch32v003fun/issues/265#issuecomment-1917597036

Here are the previous (first) and current (second) size details for a selection of the example programs:

```
riscv64-unknown-elf-size blink.elf
   text	   data	    bss	    dec	    hex	filename
    564	      0	      0	    564	    234	blink.elf
   1904       0       0    1904     770 blink.elf

riscv64-unknown-elf-size blink_raw.elf
   text	   data	    bss	    dec	    hex	filename
    568	      0	      0	    568	    238	blink_raw.elf
   1908       0       0    1908     774 blink_raw.elf

riscv64-unknown-elf-size bootload.elf
   text	   data	    bss	    dec	    hex	filename
    612	      0	      4	    616	    268	bootload.elf
    596       0       4     600     258 bootload.elf

riscv64-unknown-elf-size cap_touch_adc.elf
   text	   data	    bss	    dec	    hex	filename
   2352	      0	      0	   2352	    930	cap_touch_adc.elf
   2680       0       0    2680     a78 cap_touch_adc.elf

riscv64-unknown-elf-size direct_gpio.elf
   text	   data	    bss	    dec	    hex	filename
    560	      0	      0	    560	    230	direct_gpio.elf
   1904       0       0    1904     770 direct_gpio.elf

riscv64-unknown-elf-size external_crystal.elf
   text	   data	    bss	    dec	    hex	filename
    540	      0	      0	    540	    21c	external_crystal.elf
   1868       0       0    1868     74c external_crystal.elf

riscv64-unknown-elf-size exti_pin_change_isr.elf
   text	   data	    bss	    dec	    hex	filename
    520	      0	      0	    520	    208	exti_pin_change_isr.elf
   1892       0       0    1892     764 exti_pin_change_isr.elf

riscv64-unknown-elf-size GPIO.elf
   text	   data	    bss	    dec	    hex	filename
    384	      0	      0	    384	    180	GPIO.elf
    360       0       0     360     168 GPIO.elf

riscv64-unknown-elf-size iwdg.elf
   text	   data	    bss	    dec	    hex	filename
    660	      0	      0	    660	    294	iwdg.elf
   2008       0       0    2008     7d8 iwdg.elf

riscv64-unknown-elf-size run_from_ram.elf
   text	   data	    bss	    dec	    hex	filename
    424	      0	      4	    428	    1ac	run_from_ram.elf
    400       0       4     404     194 run_from_ram.elf

riscv64-unknown-elf-size spi_24L01_rx.elf
   text	   data	    bss	    dec	    hex	filename
   3780	      8	     24	   3812	    ee4	spi_24L01_rx.elf
   3912       8      24    3944     f68 spi_24L01_rx.elf

riscv64-unknown-elf-size spi_max7219.elf
   text	   data	    bss	    dec	    hex	filename
   5808	      0	      0	   5808	   16b0	spi_max7219.elf
   9356       0       0    9356    248c spi_max7219.elf

riscv64-unknown-elf-size standby_btn.elf
   text	   data	    bss	    dec	    hex	filename
   2036	      0	      0	   2036	    7f4	standby_btn.elf
   2132       0       0    2132     854 standby_btn.elf

riscv64-unknown-elf-size struct_direct_gpio.elf
   text	   data	    bss	    dec	    hex	filename
    560	      0	      0	    560	    230	struct_direct_gpio.elf
   1904       0       0    1904     770 struct_direct_gpio.elf
```
This is the toolchain that I'm using:
```
 riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/issue-1646/installed-tools/libexec/gcc/riscv64-unknown-elf/14.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/issue-1646/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/issue-1646/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g04696df09 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/issue-1646/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.2.0 (g04696df09)
```

DanSkareda, 2025-01-01:
Sorry, but I don't really understand compilation with "make", so I don't understand what you advise me. I just use what is in the examples. I installed the compiler for translation as recommended (I use Raspberry OS, i.e. Debian for compilation):
"apt-get install build-essential libnewlib-dev gcc-riscv64-unknown-elf libusb-1.0-0-dev libudev-dev gdb-multiarch"

Could my problem be related to the fact that I am trying to load the blink.bin program into the circuit itself without an external crystal? I have it connected only with 3V3, GND and SWIO, i.e. with three wires. And I have the LED with the resistor connected to PC0 and GND.

TommyMurphyTM1234, 2025-01-01:
> Sorry, but I don't really understand compilation with "make", so I don't understand what you advise me. 

I'm not advising you of anything.
I'm just confirming that I can reproduce the issue with the example program (and many or most others) being a lot larger than they were in the past so something has changed over the past year or so to cause this.
I don't know if it's an anomaly or simply expected due to changes and enhancements to the ch32v003fun framework.
Maybe @cnlohr or somebody else might be able to comment?

I haven't had a chance to try to run the recompiled (and larger) blink example on my board yet but I expect that I would get the same results as you did.

> Could my problem be related to the fact that I am trying to load the blink.bin program into the circuit itself without an external crystal? 

Presumably not if other examples work fine?

> Interestingly, when I compile the blink_raw example, it works, the led blinks.

Maybe you can interactively debug your recompiled blink program to see why it fails?

DanSkareda, 2025-01-02:
I have made several attempts and I will clarify the problem - according to my findings, the problem concerns the PC0 pin, which in the example "example/blink" does not blink if I compile it with the "make" command (see above how I do it). Today I tried the PD4 pin and it blinks in all cases. So the compiler has a problem with the syntax that writes the blinking of the PC0 pin in the example "example/blink". I ask the author to look into it and fix it. Thank you. (I add that I tested it on the CH32V003A4M6 chip)

cnlohr, 2025-01-02:
That's very odd.  It is normally a permissions issue when `make` can't find the programmer, but glad your issue is resolved.

TommyMurphyTM1234, 2025-01-02:
Hi @cnlohr - just curious regarding the sizes of the complied examples now versus earlier last year - is it expected that many of them are so much larger? Some of them are nearly 4x larger than before. Is there any point/need in me analysing this further or is it the way it is due to changes to the framework over time?

cnlohr, 2025-01-02:
@TommyMurphyTM1234 no that is not expected, can you attach some .map and .lst files for the worst offenders?

cnlohr, 2025-01-03:
AAH I wonder if it's because of the:

```
#if FUNCONF_DEBUG_HARDFAULT && ( FUNCONF_USE_DEBUGPRINTF || FUNCONF_USE_UARTPRINTF )
```

Can you try setting 

```
#define FUNCONF_DEBUG_HARDFAULT 0
```

in your funconf?

Maybe we should tighten that up some.

TommyMurphyTM1234, 2025-01-03:
> AAH I wonder if it's because of the:
> 
> ```
> #if FUNCONF_DEBUG_HARDFAULT && ( FUNCONF_USE_DEBUGPRINTF || FUNCONF_USE_UARTPRINTF )
> ```
> 
> Can you try setting
> 
> ```
> #define FUNCONF_DEBUG_HARDFAULT 0
> ```
> 
> in your funconf?
> 
> Maybe we should tighten that up some.

Thanks, let me check tomorrow. 👍

TommyMurphyTM1234, 2025-01-03:
Hi @cnlohr

Looks like you are correct:
```
# Recompiled straight out of the box/repo
riscv64-unknown-elf-size blink.elf
   text    data     bss     dec     hex filename
   1904       0       0    1904     770 blink.elf

riscv64-unknown-elf-nm --print-size --size-sort --reverse-sort blink.elf
0000046c 000001bc t mini_vpprintf
000002b2 000000e4 t _write.constprop.0
000001e8 000000c0 t mini_itoa.part.0
0000069e 0000009c T main
00000000 0000009c T InterruptVectorDefault
00000000 0000009c T InterruptVector
0000011e 0000008a t mini_pad
000003aa 0000007a T handle_reset
000000a0 00000058 T __divsi3
0000064a 00000054 T SystemInit
000001a8 00000040 t _puts
000000a8 0000002c T __udivsi3
000000a8 0000002c T __hidden___udivsi3
00000432 00000028 T WWDG_IRQHandler
00000432 00000028 T USB_LP_CAN1_RX0_IRQHandler
00000432 00000028 T USB_HP_CAN1_TX_IRQHandler
00000432 00000028 T USBWakeUp_IRQHandler
00000432 00000028 T USBPD_WKUP_IRQHandler
00000432 00000028 T USBPD_IRQHandler
00000432 00000028 T USBHS_IRQHandler
00000432 00000028 T USBHSWakeUp_IRQHandler
00000432 00000028 T USBFS_WakeUp_IRQHandler
00000432 00000028 T USBFS_IRQHandler
00000432 00000028 T USBFSWakeUp_IRQHandler
00000432 00000028 T USART4_IRQHandler
00000432 00000028 T USART3_IRQHandler
00000432 00000028 T USART2_IRQHandler
00000432 00000028 T USART1_IRQHandler
00000432 00000028 T UART8_IRQHandler
00000432 00000028 T UART7_IRQHandler
00000432 00000028 T UART6_IRQHandler
00000432 00000028 T UART5_IRQHandler
00000432 00000028 T UART4_IRQHandler
00000432 00000028 T TIM9_UP_IRQHandler
00000432 00000028 T TIM9_TRG_COM_IRQHandler
00000432 00000028 T TIM9_CC_IRQHandler
00000432 00000028 T TIM9_BRK_IRQHandler
00000432 00000028 T TIM8_UP_IRQHandler
00000432 00000028 T TIM8_TRG_COM_IRQHandler
00000432 00000028 T TIM8_CC_IRQHandler
00000432 00000028 T TIM8_BRK_IRQHandler
00000432 00000028 T TIM7_IRQHandler
00000432 00000028 T TIM6_IRQHandler
00000432 00000028 T TIM5_IRQHandler
00000432 00000028 T TIM4_IRQHandler
00000432 00000028 T TIM3_IRQHandler
00000432 00000028 T TIM2_TRG_IRQHandler
00000432 00000028 T TIM2_IRQHandler
00000432 00000028 T TIM2_CC_IRQHandler
00000432 00000028 T TIM2_BRK_IRQHandler
00000432 00000028 T TIM1_UP_IRQHandler
00000432 00000028 T TIM1_TRG_COM_IRQHandler
00000432 00000028 T TIM1_CC_IRQHandler
00000432 00000028 T TIM1_BRK_IRQHandler
00000432 00000028 T TIM10_UP_IRQHandler
00000432 00000028 T TIM10_TRG_COM_IRQHandler
00000432 00000028 T TIM10_CC_IRQHandler
00000432 00000028 T TIM10_BRK_IRQHandler
00000432 00000028 T TAMPER_IRQHandler
00000432 00000028 T SysTick_Handler
[... 195 lines cut ...]
0000011a 00000002 T DMA1_Channel1_IRQHandler
0000011a 00000002 T CAN2_TX_IRQHandler
0000011a 00000002 T CAN2_SCE_IRQHandler
0000011a 00000002 T CAN2_RX1_IRQHandler
0000011a 00000002 T CAN2_RX0_IRQHandler
0000011a 00000002 T CAN1_SCE_IRQHandler
0000011a 00000002 T CAN1_RX1_IRQHandler
0000011a 00000002 T Break_Point_Handler
0000011a 00000002 T BB_IRQHandler
0000011a 00000002 T AWU_IRQHandler
0000011a 00000002 T ADC1_IRQHandler
0000011a 00000002 T ADC1_2_IRQHandler

# Toolchain details
riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/ch32v003fun/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/14.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/ch32v003fun/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/ch32v003fun/riscv-gnu-toolchain/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/ch32v003fun/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.2.0 ()
```
Do you need me to do any further tests - e.g. provide more details, try other example projects, etc.?

cnlohr, 2025-01-03:
No more tests at this time.  I will add this to my todo for this month.

closed 2025-01-03 by cnlohr