cnlohr/ch32fun #483: `cpp_virtual_methods` example fails to build out of the box

TommyMurphyTM1234, 2025-01-01:
Perhaps related to the fact that this example overrides the usual ch32v003fun compilation flags?

- https://github.com/cnlohr/ch32v003fun/blob/14a146e878cc9380355a9b5d177f3b1a4cf5b372/examples/cpp_virtual_methods/Makefile#L8-L24

Transcript of what happens when attempting to build it from the repo:
```
# RISC-V GCC toolchain details
riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/issue-1646/installed-tools/libexec/gcc/riscv64-unknown-elf/14.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/issue-1646/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/issue-1646/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g04696df09 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/issue-1646/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.2.0 (g04696df09)

# Clone ch32v003fun
git clone https://github.com/cnlohr/ch32v003fun

# Try to build the cpp_virtual_methods example
cd ch32v003fun/examples/cpp_virtual_methods
make
riscv64-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32V003 -DMCU_PACKAGE= -DTARGET_MCU_LD=0 -DTARGET_MCU_MEMORY_SPLIT= ../../ch32v003fun//ch32v003fun.ld > ../../ch32v003fun//generated_ch32v003.ld
riscv64-unknown-elf-gcc -o cpp_virtual_methods.elf ../../ch32v003fun//ch32v003fun.c cpp_virtual_methods.cpp example.cpp  -g -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../../ch32v003fun/ -nostdlib -DCH32V003 -I. -Wall -fno-rtti -DCPLUSPLUS -Wl,--print-memory-usage -Wl,-Map=cpp_virtual_methods.map -L../../ch32v003fun//../misc -lgcc -T ../../ch32v003fun//generated_ch32v003.ld -Wl,--gc-sections
cc1: warning: command-line option '-fno-rtti' is valid for C++/D/ObjC++ but not for C
../../ch32v003fun/ch32v003fun.h: Assembler messages:
../../ch32v003fun/ch32v003fun.h:13518: Error: unrecognized opcode `csrr a5,mepc', extension `zicsr' required
lto-wrapper: fatal error: riscv64-unknown-elf-gcc returned 1 exit status
compilation terminated.
/home/user/issue-1646/installed-tools/lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [../../ch32v003fun/ch32v003fun.mk:233: cpp_virtual_methods.elf] Error 1

# For comparison this is the blink example (ignore the minichlink error)
make
riscv64-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32V003 -DMCU_PACKAGE= -DTARGET_MCU_LD=0 -DTARGET_MCU_MEMORY_SPLIT= ../../ch32v003fun/ch32v003fun.ld > ../../ch32v003fun/generated_ch32v003.ld
riscv64-unknown-elf-gcc -o blink.elf ../../ch32v003fun/ch32v003fun.c blink.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun/../extralibs -I../../ch32v003fun -nostdlib -I. -Wall  -Wl,--print-memory-usage -Wl,-Map=blink.map -L../../ch32v003fun/../misc -lgcc -T ../../ch32v003fun/generated_ch32v003.ld -Wl,--gc-sections
Memory region         Used Size  Region Size  %age Used
           FLASH:        1904 B        16 KB     11.62%
             RAM:           0 B         2 KB      0.00%
riscv64-unknown-elf-objdump -S blink.elf > blink.lst
riscv64-unknown-elf-objcopy -O binary blink.elf blink.bin
riscv64-unknown-elf-objcopy -O ihex blink.elf blink.hex
make -C ../../ch32v003fun/../minichlink all
make[1]: Entering directory '/home/user/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/user/ch32v003fun/minichlink'
../../ch32v003fun/../minichlink/minichlink -w blink.bin flash -b
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
make: *** [../../ch32v003fun/ch32v003fun.mk:242: cv_flash] Error 224
```

cnlohr, 2025-01-02:
Merged #485 - does that fix this issue?

TommyMurphyTM1234, 2025-01-02:
> Merged #485 - does that fix this issue?

Yes, it does. I checked this already. Thanks a lot. 👍

closed 2025-01-02 by TommyMurphyTM1234