cnlohr/ch32fun #484: Problem with unbricking CH32V003J4M6 using WCH-LinkE

xinaovo, 2025-01-01:
Hi there! I compiled and uploaded the template code(which uses the debug pin PD1) to the MCU and I found that I couldn't upload other code to the MCU any more. So I tried to unbrick it using command `minichlink -u` but it stuck at `Connection starting` for a while and failed. The log is shown below.

```
minichlink -u
Found WCH Link
Entering Unbrick Mode
Connection starting
Programmer wasn't initialized? Fixing

DMStatus After Halt: /0/ffffffff
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Unknown chip type.  Report as bug with picture of chip.
Vendored: ffffffff
marchid : ffffffff
HARTINFO: ffffffff
Chip Type: 0
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Unknown chip type.  Report as bug with picture of chip.
Vendored: ffffffff
marchid : ffffffff
HARTINFO: ffffffff
Error: Flash is not unlocked (CTLR = ffffffff)
Error: Flash is not unlocked (CTLR = ffffffff)
Error: Flash is not unlocked (CTLR = ffffffff)
```
However, after reinstalling the WCH vendor driver and using "Clear All Code Flash-By Power Off" in the WCH-Link Utility, the MCU was unbricked successfully.

[image]
[image]

btw, the connection between MCU and WCH-LinkE programmer as well as the version of WCH-LinkE and WCH-Link Utility is shown below:
```
WCH-LinkE > CH32V003J4M6
SWDIO > Pin 8
GND > Pin 2
3V3 > Pin 4

WCH-LinkE version(reported by minichlink): 2.15
WCH-Link Utility version: 2.30
```

roboticboyer, 2025-01-04:
See [here](https://youtu.be/9UHotTF5jvg?si=JZXjj0JedtVYHrXl) the procedure/trick with minichlink -u (required a power off)

jptrzy, 2025-02-08:
> See [here](https://youtu.be/9UHotTF5jvg?si=JZXjj0JedtVYHrXl) the procedure/trick with minichlink -u (required a power off)

I also have the same issue, even thou I used the suggested method.

xinaovo, 2025-02-09:
> See [here](https://youtu.be/9UHotTF5jvg?si=JZXjj0JedtVYHrXl) the procedure/trick with minichlink -u (required a power off)

Hi there! Thank you for your reply. I tried the trick in the video but it didn't work. In theory, the programmer can control its power output and power off the MCU by itself and that's how the official tool works. However, [rvprog](https://github.com/wagiminator/MCU-Flash-Tools#rvprog), a python-based CH32Vxxx utility, works just fine. So I think it should be the problem of minichlink.

cnlohr, 2025-02-10:
What issue did you have with minichlink? (What what physical programmer are you using) I've tried to make it as robust as possible for unbricking.

xinaovo, 2025-02-12:
> What issue did you have with minichlink? (What what physical programmer are you using) I've tried to make it as robust as possible for unbricking.

I used wchlink-e with firmware version v2.15 as my programmer and tried to unbrick a SOP-8 CH32V003J4M6. I tried `minichlink -u` to unbrick the MCU. It didn't power off the chip and power it up again to enter the "unbrick mode". However, other tools like rvprog or the official tool works just fine.

> Hi there! I compiled and uploaded the template code(which uses the debug pin PD1) to the MCU and I found that I couldn't upload other code to the MCU any more. So I tried to unbrick it using command [...]

cnlohr, 2025-02-14:
This is a little surprising.  Is this in Windows?  Or Linux?  And, also, are you recompiling the .exe.  It's been quite some time since a new .exe has been checked into the main repo.

cnlohr, 2025-02-14:
Also, you may want to iteratively work on this with some folks live?  If you're interested ping me on discord, and I can give you an invite.  My name on Disocord is `cnlohr`

xinaovo, 2025-02-15:
> This is a little surprising. Is this in Windows? Or Linux? And, also, are you recompiling the .exe. It's been quite some time since a new .exe has been checked into the main repo.

I tested the pre-built minichlink as well the one compiled by myself before. None of them worked. I also tested the latest pre-built version (ac7f5f9) just now but it didn't work. My Windows version is Windows 11 Home China 26100.3194 and I installed WinUSB
WinUSB (v6.1.7600.16385) for WCHLink-E interface 0 (interface 1 is the serial port.). 
[image]

[image]

When I tried to use the latest minichlink to unbrick the MCU, it stucked at `Connection starting` and it failed after a while, just like before. The full log is shown below.
`./minichlink.exe -u
Found WCH Link
Entering Unbrick Mode
Connection starting
DMStatus After Halt: /0/ffffffff
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Unknown chip type.  Report as bug with picture of chip.
Vendored: ffffffff
marchid : ffffffff
HARTINFO: ffffffff
Chip Type: 0
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Unknown chip type.  Report as bug with picture of chip.
Vendored: ffffffff
marchid : ffffffff
HARTINFO: ffffffff
Error: Flash is not unlocked (CTLR = ffffffff)
Error: Flash is not unlocked (CTLR = ffffffff)
Whole-chip erase
Unknown chip type.  Report as bug with picture of chip.
Vendored: 00000000
marchid : 00000000
HARTINFO: 00000000`

BTW, the test setup is shown below. I uploaded a blink program which uses PD1 and "bricked" the chip.

[image]

GHtzz, 2025-04-12:
this append to me too, any way to brick ?

cnlohr, 2025-04-15:
What if you start unbricking with power disconnected, then apply power?

xinaovo, 2025-05-31:
Hi cnlohr, I tried the latest version of minichlink on Windows 11, it still behaves like before. When I tried unbricking with power disconnected, minichlink gave error messages immediately, before I hooked the power up. However, another python-based programming software, `rvprog`, worked just fine even without disconnecting the power at the beginning.

Error messages are shown below:
`Found WCH Link
Entering Unbrick Mode
Connection starting
DMStatus After Halt: /0/0fffffff
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: 0fffffff
Chip Type: fff
Unknown chip type.  Report as bug with picture of chip.
Vendored: ffffffff
marchid : ffffffff
HARTINFO: 1fffffff
Chip Type: 0
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: 1fffffff
Chip Type: fff
Unknown chip type.  Report as bug with picture of chip.
Vendored: ffffffff
marchid : ffffffff
HARTINFO: 0fffffff
Error: Flash is not unlocked (CTLR = ffffffff)
Error: Flash is not unlocked (CTLR = ffffffff)
Error: Flash is not unlocked (CTLR = ffffffff)`

cnlohr, 2025-06-02:
Can you show the whole command line? I am not sure how that path could be executed with `minichlink -u`

xinaovo, 2025-06-04:
> this append to me too, any way to brick ?

Try `rvprog`. It seems to work just fine in my case.

closed 2025-11-15 by cnlohr