cnlohr/ch32fun #493: Examples don't get compiled

Plaque-fcc, 2025-01-15:
```shell
$ cd examples/blink_raw/ && make
```

```
riscv64-linux-gnu-gcc -E -P -x c -DTARGET_MCU=CH32V003 -DMCU_PACKAGE= -DTARGET_MCU_LD=0 -DTARGET_MCU_MEMORY_SPLIT= ../../ch32v003fun//ch32v003fun.ld > ../../ch32v003fun//generated_ch32v003.ld
riscv64-linux-gnu-gcc -o blink_raw.elf ../../ch32v003fun//ch32v003fun.c blink_raw.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../../ch32v003fun//../extralibs -I../../ch32v003fun/ -nostdlib -I. -Wall  -Wl,--print-memory-usage -Wl,-Map=blink_raw.map -L../../ch32v003fun//../misc -lgcc -T ../../ch32v003fun//generated_ch32v003.ld -Wl,--gc-sections
In file included from /usr/include/newlib/stdio.h:800,
                 from ../../ch32v003fun//ch32v003fun.c:84:
/usr/include/newlib/ssp/stdio.h:61:40: error: expected declaration specifiers or ‘...’ before numeric constant
   61 |     __builtin___snprintf_chk(str, len, 0, __ssp_bos(str), __VA_ARGS__)
      |                                        ^
../../ch32v003fun//ch32v003fun.c:120:10: note: in expansion of macro ‘snprintf’
  120 | WEAK int snprintf( char * buffer, unsigned int buffer_len, const char* format, ... )
      |          ^~~~~~~~
In file included from /usr/include/newlib/ssp/stdio.h:34,
                 from /usr/include/newlib/stdio.h:800,
                 from ../../ch32v003fun//ch32v003fun.c:84:
/usr/include/newlib/ssp/ssp.h:46:24: error: expected declaration specifiers or ‘...’ before ‘__builtin_object_size’
   46 | #define __ssp_bos(ptr) __builtin_object_size(ptr, __SSP_FORTIFY_LEVEL > 1)
      |                        ^~~~~~~~~~~~~~~~~~~~~
/usr/include/newlib/ssp/stdio.h:61:43: note: in expansion of macro ‘__ssp_bos’
   61 |     __builtin___snprintf_chk(str, len, 0, __ssp_bos(str), __VA_ARGS__)
      |                                           ^~~~~~~~~
../../ch32v003fun//ch32v003fun.c:120:10: note: in expansion of macro ‘snprintf’
  120 | WEAK int snprintf( char * buffer, unsigned int buffer_len, const char* format, ... )
      |          ^~~~~~~~
In file included from /usr/include/newlib/stdio.h:800,
                 from ../../ch32v003fun//ch32v003fun.c:84:
/usr/include/newlib/ssp/stdio.h:55:34: error: expected declaration specifiers or ‘...’ before numeric constant
   55 |     __builtin___sprintf_chk(str, 0, __ssp_bos(str), __VA_ARGS__)
      |                                  ^
../../ch32v003fun//ch32v003fun.c:129:10: note: in expansion of macro ‘sprintf’
  129 | WEAK int sprintf( char * buffer, const char * format, ... )
      |          ^~~~~~~
In file included from /usr/include/newlib/ssp/stdio.h:34,
                 from /usr/include/newlib/stdio.h:800,
                 from ../../ch32v003fun//ch32v003fun.c:84:
/usr/include/newlib/ssp/ssp.h:46:24: error: expected declaration specifiers or ‘...’ before ‘__builtin_object_size’
   46 | #define __ssp_bos(ptr) __builtin_object_size(ptr, __SSP_FORTIFY_LEVEL > 1)
      |                        ^~~~~~~~~~~~~~~~~~~~~
/usr/include/newlib/ssp/stdio.h:55:37: note: in expansion of macro ‘__ssp_bos’
   55 |     __builtin___sprintf_chk(str, 0, __ssp_bos(str), __VA_ARGS__)
      |                                     ^~~~~~~~~
../../ch32v003fun//ch32v003fun.c:129:10: note: in expansion of macro ‘sprintf’
  129 | WEAK int sprintf( char * buffer, const char * format, ... )
      |          ^~~~~~~
/usr/include/newlib/ssp/ssp.h:47:52: error: expected declaration specifiers or ‘...’ before numeric constant
   47 | #define __ssp_bos0(ptr) __builtin_object_size(ptr, 0)
      |                                                    ^
/usr/include/newlib/ssp/string.h:52:7: note: in expansion of macro ‘__ssp_bos0’
   52 |     ((__ssp_bos0(dst) != (size_t)-1) ? \
      |       ^~~~~~~~~~
/usr/include/newlib/ssp/string.h:102:31: note: in expansion of macro ‘__ssp_bos_check3’
  102 | #define memset(dst, val, len) __ssp_bos_check3(memset, dst, val, len)
      |                               ^~~~~~~~~~~~~~~~
../../ch32v003fun//ch32v003fun.c:223:12: note: in expansion of macro ‘memset’
  223 | WEAK void *memset(void *dest, int c, size_t n) { unsigned char *s = dest; for (; n; n--, s++) *s = c; return dest; }
      |            ^~~~~~
In file included from /usr/include/newlib/string.h:180,
                 from ../../ch32v003fun//ch32v003fun.c:85:
/usr/include/newlib/ssp/string.h:52:23: error: expected ‘)’ before ‘!=’ token
   52 |     ((__ssp_bos0(dst) != (size_t)-1) ? \
      |                       ^~
/usr/include/newlib/ssp/string.h:102:31: note: in expansion of macro ‘__ssp_bos_check3’
  102 | #define memset(dst, val, len) __ssp_bos_check3(memset, dst, val, len)
      |                               ^~~~~~~~~~~~~~~~
../../ch32v003fun//ch32v003fun.c:223:12: note: in expansion of macro ‘memset’
  223 | WEAK void *memset(void *dest, int c, size_t n) { unsigned char *s = dest; for (; n; n--, s++) *s = c; return dest; }
      |            ^~~~~~
[... 126 lines cut ...]
../../ch32v003fun//ch32v003fun.c:437:12: note: in expansion of macro ‘memmove’
  437 | WEAK void *memmove(void *dest, const void *src, size_t n)
      |            ^~~~~~~
In file included from /usr/include/newlib/string.h:180,
                 from ../../ch32v003fun//ch32v003fun.c:85:
/usr/include/newlib/ssp/string.h:52:23: error: expected ‘)’ before ‘!=’ token
   52 |     ((__ssp_bos0(dst) != (size_t)-1) ? \
      |                       ^~
/usr/include/newlib/ssp/string.h:98:32: note: in expansion of macro ‘__ssp_bos_check3’
   98 | #define memmove(dst, src, len) __ssp_bos_check3(memmove, dst, src, len)
      |                                ^~~~~~~~~~~~~~~~
../../ch32v003fun//ch32v003fun.c:437:12: note: in expansion of macro ‘memmove’
  437 | WEAK void *memmove(void *dest, const void *src, size_t n)
      |            ^~~~~~~
/usr/include/newlib/ssp/string.h:52:38: error: expected ‘)’ before ‘?’ token
   52 |     ((__ssp_bos0(dst) != (size_t)-1) ? \
      |                                      ^
/usr/include/newlib/ssp/string.h:98:32: note: in expansion of macro ‘__ssp_bos_check3’
   98 | #define memmove(dst, src, len) __ssp_bos_check3(memmove, dst, src, len)
      |                                ^~~~~~~~~~~~~~~~
../../ch32v003fun//ch32v003fun.c:437:12: note: in expansion of macro ‘memmove’
  437 | WEAK void *memmove(void *dest, const void *src, size_t n)
      |            ^~~~~~~
make: *** [../../ch32v003fun/ch32v003fun.mk:233: blink_raw.elf] Error 1
```

TommyMurphyTM1234, 2025-01-15:
Why are you trying to compile the ch32v003fun example using the RISC-V Linux toolchain rather than the RISC-V bare-metal toolchain?

Plaque-fcc, 2025-01-15:
> Why are you trying to compile the ch32v003fun example using the RISC-V Linux toolchain rather than the RISC-V bare-metal toolchain?

I am new and had no specific intention to prefer one instead the other. All I did was I followed the instructions provided with the project.

TommyMurphyTM1234, 2025-01-15:
You should use the RISC-V bare-metal toolchain:

- https://github.com/cnlohr/ch32v003fun?tab=readme-ov-file#getting-started
  - https://github.com/cnlohr/ch32v003fun/wiki/Installation

TommyMurphyTM1234, 2025-01-15:
You can probably use the latest xPack binary toolchain to avoid having to build your own...

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/38

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/
  - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/releases/
    - https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/blog/2024/12/06/riscv-none-elf-gcc-v14-2-0-3-released/

Note that the xPack toolchain uses the prefix `riscv-none-elf-` and not the "usual" `riscvXX-unknown-elf-`.

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/user/#miscellaneous

Plaque-fcc, 2025-01-15:
I managed to build it, and it seems to have resolved my issue. Thank you very much, closing this one.

closed 2025-01-15 by Plaque-fcc

TommyMurphyTM1234, 2025-01-15:
Great, glad it helped! 👍

cnlohr, 2025-02-03:
@Plaque-fcc what OS were you on?  You should not need to install any additional compilers that don't already come with the host OS on any debian/mint/ubuntu OSes.

cnlohr, 2025-02-03:
(I would like to be able to reproduce this locally, but between all of my Debian/Mint/Ubuntu systems, I cannot identify any issues or off-nominal behaviors)

Plaque-fcc, 2025-02-04:
Ubuntu latest LTS (ARM64) and Ubuntu LTS before that (x86_64).

You could've seen what exists in the repo, it's outdated.

But after building the toolchain (there's not much room on my AMD x86_64 machine, so I offloaded this task to my ARM64 one) with required params, everything works.

cnlohr, 2025-02-04:
Total side note:  Everything should work in all combinations with the `dev` branch.  But, I am disabling the automatic defaulting to gcc-riscv-linux-gnu.

Plaque-fcc, 2025-02-05:
> Total side note: Everything should work in all combinations with the `dev` branch. But, I am disabling the automatic defaulting to gcc-riscv-linux-gnu.

Answering your question: yes, currently it builds just fine with this GCC I already have on the original PC where it all started:
```
$ riscv64-unknown-elf-gcc -v
gcc version 10.2.0 ()
```

cnlohr, 2025-02-06:
So everyone else is aware, while you *can* use riscv64-linux-gnu, you *should* use riscv64-unknown-elf, the binaries *will* be smaller, and it offers no benefit.