cnlohr/ch32fun #494: Why does `ch32v003fun.mk` allow the RISC-V Linux toolchain to be used

TommyMurphyTM1234, 2025-01-15:
See here: 

- https://github.com/cnlohr/ch32v003fun/blob/95cd42d8952d7bc82c53ce4b8f27e18d40116907/ch32v003fun/ch32v003fun.mk#L4-L6

Why does it allow for the RISC-V Linux rather than bare-metal toolchain to be used? Isn't it the case that all ch32v003fun stuff such as example programs need to be built using the bare-metal toolchain?

Allowing the Linux toolchain to be used runs the risk of confusion such as this: 

- https://github.com/cnlohr/ch32v003fun/issues/493

orbitalair, 2025-01-25:
I accidentally installed the riscv-linux version on Ubuntu.  Either the Ubuntu search shows that version first, or the online RISCV Assembly course I first used recommended that one.

In either case I got the same issues here, and it took me several days to figure out why it wouldnt work.

I would suggest to move the risv-linux usage in the Make.  I would move it to the last position, and print a message that riscv linux is installed, but we need to use riscv-unknown instead, and give a link.

I would also add a note to the install instructions.

TommyMurphyTM1234, 2025-01-26:
> I would suggest to move the risv-linux usage in the Make. I would move it to the last position, and print a message that riscv linux is installed, but we need to use riscv-unknown instead, and give a link.

My inclination would be to eliminate the check for the RISC-V Linux toolchain altogether unless there is a clearly understood/stated reason why and in what context using it with the ch32v003fun sources makes sense.

orbitalair, 2025-01-30:
The RISC-V Linux toolchain would be used only if you wanted to run the program under a linux os.  A program would be looking for glibc for core funcs, as i read it.

in other words, I do not see the use case for even trying the riscv-linux- toolchain either.

cnlohr, 2025-02-03:
I don't understand what the issue is.  What materially goes wrong when using riscv64-linux-gnu?  I almost exclusively use it in all of my testing.  We provide appropriate flags to the compiler to make it compile correctly for bare metal.  It's really annoying to have to get bespoke compilers for things.

cnlohr, 2025-02-03:
It's also really convenient because just about every distribution has a package that just has this compiler, and it's versatile for virtually all RV32/RV64/Bare Metal/Linux tasks.

prosper00, 2025-02-03:
> It's also really convenient because just about every distribution has a package that just has this compiler, and it's versatile for virtually all RV32/RV64/Bare Metal/Linux tasks.

at least on Debian, it tries to pull in glibc, and you get this: https://github.com/cnlohr/ch32v003fun/issues/493

TommyMurphyTM1234, 2025-02-03:
> I don't understand what the issue is. What materially goes wrong when using riscv64-linux-gnu? I almost exclusively use it in all of my testing. We provide appropriate flags to the compiler to make [...]

Using a Linux rather than a specific bare-metal toolchain to compile code for a bare-metal target is simply wrong in my opinion. Maybe with `-ffreestanding` it might be acceptable but that's generally only used for the likes of a Linux/rich OS bootloader or other early boot component or kernel. Isn't the C ABI also different between the Linux toolchain (even with `-ffreestanding`?) and the bare-metal toolchain? If the ch32fun stuff works with the Linux toolchain then I suspect that the code generated may well be less optimal than that generated by the bare-metal toolchain. But I'd need to experiment to check.

orbitalair, 2025-02-03:
On Ubuntu which is Debian based, you get compile errors because the default
makefile does not add the correct flags that you are using.

Seems better to remove Linux chain, and simply use the unknown bare metal
chain and reduce any confusion by 1 step.

cnlohr, 2025-02-03:
> On Ubuntu which is Debian based, you get compile errors because the default
> makefile does not add the correct flags that you are using.
> 
> Seems better to remove Linux chain, and simply use the unknown bare metal
> chain and reduce any confusion by 1 step.
> […](#)

I only use Ubuntu, Debian (And Mint) systems.  What compile errors are you seeing?

I have absolutely no interest in installing anything that doesn't come with the default OS if I can avoid it.

cnlohr, 2025-02-03:
What OS, specifically are you seeing any issues on?  It is possible the user may have been on some ancient compiler that just didn't support modern code.

Regarding optimization, there is no perf/size difference.  Otherwise I would have seen it when working on rv003usb.

There is no reason not to use the -linux-gnu compiler toolchain that I can find.

prosper00, 2025-02-03:
> > On Ubuntu which is Debian based, you get compile errors because the default
> > makefile does not add the correct flags that you are using.
> > Seems better to remove Linux chain, and simply use the unknown bare metal
> > chain and reduce any confusion by 1 step.
> > […](#)
> 
> I only use Ubuntu, Debian (And Mint) systems. What compile errors are you seeing? EDIT: I see the comment.
> 
> I have absolutely no interest in installing anything that doesn't come with the default OS if I can avoid it.
> 
> EDIT: Investigating now.

both toolchains are in the default repos:

```
$ apt search gcc-riscv
gcc-riscv64-linux-gnu/testing 4:14.2.0-1 amd64
  GNU C compiler for the riscv64 architecture

gcc-riscv64-unknown-elf/now 13.2.0-10+12 amd64 [installed,local]
  GCC cross compiler for Risc-V processors
```

cnlohr, 2025-02-03:
Ok, I was able to find a system where only having `gcc-riscv64-linux-gnu` installed AND newlib is installed causes the errors seen in the ticket.

It can be trivially fixed by adding the following two lines to `ch32v003fun.c` before including the headers, so that the newlib SSP stuff doesn't get included.

```c
#define _SSP_STRING_H_
#define _SSP_STDIO_H_
```

That said, this is ugly, and, indeed, I have verified that it does produce suspicious "different" code even though it does work.

This does give me more heebey geebies than I remember, so I will remove it in the dev branch, which will become master.

cnlohr, 2025-02-03:
AH! I see, it's using the bigger libgcc.  It all seems fine though.  We could just fix it by adding those two #define's.

cnlohr, 2025-02-03:
Do you have a strong preference to remove support instead of just fixing compile for those users, @prosper00 ?

orbitalair, 2025-02-03:
My 10 minute research shows me that Tommy Murphy has been on this topic for a long time, many years of messages out there, so I defer to him on  the chain technical items.  

I suggest to remove the riscv-linux option, and speak in the install instructions to load the riscv-unknown chain and newlib.

This both, aligns with most other riscv instructional websites, and removes a point of issues for a few folks (like me), who would have given up, except i am stubburn that way, and seldom give up on a issue like compiling problems.

it wouldnt be gcc if there werent > 2 chains, and each broken in a different way, now would it??

We could always build out own riscv toolchain !!  (yeah, no, lets dont do that).

cnlohr, 2025-02-03:
I believe I will take this out of the -dev branch at this time then, and let the chips continue to fall.

prosper00, 2025-02-03:
> Do you have a strong preference to remove support instead of just fixing compile for those users, [@prosper00](https://github.com/prosper00) ?

@cnlohr - no preference either way. Given the size and scope of Debian and debian-based distros, it would drive the least amount of confusion to default to the gcc-riscv64-unknown toolchain if both are available.

TommyMurphyTM1234, 2025-02-03:
> My 10 minute research shows me that Tommy Murphy has been on this topic for a long time, many years of messages out there, so I defer to him on the chain technical items.

Thanks, but I'm always open to correction. I intend to play around with building things using the Linux and the bare-metal toolchains just to compare the results, clarify matters (if only for myself) and sanity check my earlier comments.

> it wouldnt be gcc if there werent > 2 chains, and each broken in a different way, now would it??

I don't understand this. What exactly do you consider "broken" with the RISC-V Linux and bare-metal toolchains and which incarnations in particular?

> We could always build out own riscv toolchain !! (yeah, no, lets dont do that).

If the bare-metal toolchain provided by a Linux distro's software repositories isn't adequate then a possible alternative is @ilg-ul's xPack binary toolchain distribution:

- https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/38#issue-2665995260

cnlohr, 2025-02-04:
This is addressed now in #506 , I am removing the default fallback to `riscv64-linux-gnu-gcc` because it seems particularly risky.

closed 2025-02-04 by cnlohr