cnlohr/ch32fun #495: Support for uart and usb?

zuidwijk, 2025-01-16:
Im wondering if the small sot8 version can do a simple rx-uart to usb? So only receiving data from serial and send that to usb. 

Im looking for a solution and this small sot8 version of ch32v003 looks perfect.

cnlohr, 2025-01-16:
Windows doesn't allow CDC-serial low-speed devices as far as I'm aware.  If you can find any other examples of a low-speed device enumerating as a serial port in Windows, I'd love to know about it so we can copy whatever they're doing.

prosper00, 2025-01-17:
its been a long time since I've used VUSB, and even longer since I've used Windows, but I do seem to recall that the VUSB bitbang library for avr can do CDC

Some googling turns up a bunch of old projects. The most recent i can find is https://github.com/wagiminator/VUSB-AVR/tree/main/vusb_avr/libraries/VUSB_CDC

cnlohr, 2025-01-17:
I believe none of them work with anything past Windows 8.

bikeNomad, 2025-01-20:
There is this low-speed CDC driver: https://github.com/protaskin/LowCDC-Win10x64/blob/master/lowcdc.inf

cnlohr, 2025-02-03:
> There is this low-speed CDC driver: https://github.com/protaskin/LowCDC-Win10x64/blob/master/lowcdc.inf

As far as I understand, this does not work on Windows-based installs from the last 5 years.

cnlohr, 2025-02-03:
Have you been able to validate the functionality?