cnlohr/ch32fun #502: Not an issue rather a question regarding NRF24L01 software SPI

RhonyD, 2025-01-24:
I have ported my avr code for NRF24L01 to ch32v003 with harware SPI. It's working nice. But I want to ask whether any one has used software SPI (bit banging) for NRF24L01. I tried to to do so but it's not working. Especially the transmit function is not working.

maxgerhardt, 2025-01-24:
I don't see why a software SPI would not be possible on a ch32v003. All it needs for that are functioning GPIO pins. There are lots of example codes like [SoftSPI.cpp](https://github.com/MajenkoLibraries/SoftSPI/blob/master/src/SoftSPI.cpp) that you could port over, all it needs is pinMode, digitalWrite, digitalRead, and maybe a delay. Can you share the exact code with us? Have you tried scoping out the SPI signals this writes and reads?

RhonyD, 2025-01-25:
Soft spi is working perfectly in other devices like memory card reader, shift registers etc. But specially in NRF24L01 it's not working and that too in transmitter mode. I will certainly share the code.