cnlohr/ch32fun #509: Set DATA0/DATA1 during flashing

frysztak, 2025-02-05:
I'm wondering if it's possible to set Option Data (DATA0/DATA1) during flashing? Assuming Option Data is being used to store user preferences, setting those bytes during flashing would allow to set default values.

monte-monte, 2025-02-06:
Can you elaborate what exactly you mean/want to achieve?
If you want, you can write with ``minichlink -w 0x1FFFF800 0x64BYTES_HEX_OF_OPTION_BYTES``

frysztak, 2025-02-06:
My idea was to store devices preferences as DATA0/DATA1. I wanted to define default, non-zero values somewhere in my app and flash everything (app itself + DATA0/DATA1) in one go. As far as I can tell it'd require messing around with linker definitions and I never got it working. But using `minichlink` to set those values seems far easier. Thank you.

dwillmore, 2025-02-06:
Take a look at issue #508 .

monte-monte, 2025-02-07:
You can just add that command to your makefile and have default setting as a variable there.

closed 2025-02-07 by frysztak