cnlohr/ch32fun #518: Why are DMA defines above channel 3 missing the 'R' int he register define.

markusdd, 2025-02-09:
According to spec they are all named the same CFGR1/2/3/4...7.

But in the Defines from channel 4 onwards the R is missing.

Is there a rationale for this?

dwillmore, 2025-02-09:
That does look like a bug.  The V1.7 RM shows:

[image]

markusdd, 2025-02-09:
Precisely.

That is why I am bringing it up, I noticed in my editors auto complete that the defines looked slightly off.

TommyMurphyTM1234, 2025-02-09:
Just for reference purposes...

Correct:

- https://github.com/cnlohr/ch32v003fun/blob/0a9b33a825cc8283c780fe620eaca0342c733b4f/ch32fun/ch32v003hw.h#L1171-L1172

Incorrect? 

- https://github.com/cnlohr/ch32v003fun/blob/0a9b33a825cc8283c780fe620eaca0342c733b4f/ch32fun/ch32v003hw.h#L1243-L1244

markusdd, 2025-02-09:
pretty much. Channel4-7 have the R missing after the CFG.

not a big deal as such but it seems off. (could get annoying if someone starts doing real nasty preprocessor stuff though) xD

dwillmore, 2025-02-09:
If either of you has the time to create a PR, that would be appreciated.  I'm in the middle of adding the 00X chips and I don't want to lose my place in this huge file. :)

markusdd, 2025-02-09:
I can prepare on, but just a heads up: this has propagated into all derived headers as well from the other chips^^

markusdd, 2025-02-09:
see #519 

we got lucky, no examples use these channels :)

dwillmore, 2025-02-09:
Yeah, I checked to make sure no examples used them.  I wasn't trying to make your life harder. :)  Thank.  We'll get this reviewed shortly.

PR #519 (2025-02-13): Fix naming bug in DMA CFGR header for channels 4 to 7
closes #518
closed 2025-02-13 by cnlohr