cnlohr/ch32fun #528: Failed to compile the latest version(ac7f5f9) using vscode in Windows

xinaovo, 2025-02-15:
Hi there! I set up the recommended toolchain in Windows and tried to compile an example project. I've added the toolchain directory into the PATH environment variable. It failed and the log is shown below.
`Executing task: "make.exe" 

process_begin: CreateProcess(NULL, dirname ../../ch32fun/ch32fun.mk, ...) failed.
process_begin: CreateProcess(NULL, dirname ../../ch32fun/ch32fun.mk, ...) failed.
process_begin: CreateProcess(NULL, dirname ../../ch32fun/ch32fun.mk, ...) failed.
process_begin: CreateProcess(NULL, dirname ../../ch32fun/ch32fun.mk, ...) failed.
process_begin: CreateProcess(NULL, dirname ../../ch32fun/ch32fun.mk, ...) failed.
make.exe: *** No rule to make target `/ch32fun.c', needed by `GPIO.elf'.  Stop.`

Besides, vscode failed to include `ch32fun`. There maybe preblem related to the makefile(?)

[image]
[image]
[image]
[image]

xinaovo, 2025-03-05:
#546 has fixed this problem. Now it works.

closed 2025-03-05 by xinaovo

cnlohr, 2025-03-05:
Sorry about the issue.  We're in a bit of a timultuous phase right now, with making support for new chips a high priority.