cnlohr/ch32fun #535: uart_tx_rx_demo

joyhope, 2025-02-25:
I do not have x305 board, I use the sample in ch32v20x with minor changes for SysTick->CNT.  In funconfig.h, I removed CPU define.  I tried, using PB10, PB11 as the demo, but I could not get anything.  

Is there anything I missing？

joyhope, 2025-02-25:
It is my problem, I found there are many differents for two CPUs. 

Try close to the original demo:  UART3 is using the same PB10, PB11,  DMA had to use DMA 2, 3 for TX, RX. 

I make changed, but it has not work. 

`#include "ch32fun.h"
#include <stdio.h>

// Set UART baud rate here
#define UART_BR 115200

// Define your RX buffer and its size
#define RX_BUFFER_SIZE 512
uint8_t rx_buffer[RX_BUFFER_SIZE];

// Variable to store the current RX buffer write position (updated by the IDLE interrupt)
volatile uint16_t rx_buffer_write_pos = 0;

// Variable to store the current RX buffer read position
volatile uint16_t rx_buffer_read_pos = 0;

/**
 * USART3 IDLE interrupt handler.
 */
__attribute__( ( interrupt ) ) __attribute__( ( section( ".srodata" ) ) ) void USART3_IRQHandler( void );
void USART3_IRQHandler( void )
{
	// Check for IDLE interrupt
	if ( USART3->STATR & USART_STATR_IDLE )
	{
		// Clear the IDLE interrupt flag.
		// Source, WCH documentation: reading status and then data register clears the idle bit;
		USART3->STATR;
		USART3->DATAR;

		// Update the position so the rx functions can read the data.
		rx_buffer_write_pos = ( RX_BUFFER_SIZE - DMA1_Channel3->CNTR ) % RX_BUFFER_SIZE;
	}
}

/**
 * Initialize the USART pheripheral
 */
static void uart_setup( void )
{
	// Enable UART and GPIOB
#if 0
	RCC->APB2PCENR |= RCC_APB2Periph_GPIOB | RCC_APB1Periph_USART3;
#else
	funGpioInitB();
#endif

#if 0
	// Setup PB10 as TX
	GPIOB->CFGHR =
		( GPIOB->CFGHR & ~( 0xF << ( 4 * 2 ) ) ) | ( ( GPIO_Speed_10MHz | GPIO_CNF_OUT_PP_AF ) << ( 4 * 2 ) );

	// Setup PB11 as RX
	GPIOB->CFGHR = ( GPIOB->CFGHR & ~( 0xF << ( 4 * 3 ) ) ) | ( ( GPIO_Speed_In | GPIO_CNF_IN_FLOATING ) << ( 4 * 3 ) );
#else
	funPinMode(PB10, (GPIO_Speed_10MHz | GPIO_CNF_OUT_PP_AF));
	funPinMode(PB11, (GPIO_Speed_In | GPIO_CNF_IN_FLOATING));
#endif
	// Setup UART for Tx 8n1
	USART3->CTLR1 = USART_WordLength_8b | USART_Parity_No | USART_Mode_Tx | USART_Mode_Rx;
	USART3->CTLR2 = USART_StopBits_1;

	// Enable Tx and Rx DMA event
	USART3->CTLR3 = USART_DMAReq_Tx | USART_DMAReq_Rx | USART_HardwareFlowControl_None;
[... 139 lines cut ...]
	while ( 1 )
	{
		// Check if data is available in the RX buffer
		int bytes_available = dma_uart_available();
		if ( bytes_available > 0 )
		{
			uint8_t buffer[RX_BUFFER_SIZE + 6] = { 0 };

			buffer[0] = 'E';
			buffer[1] = 'c';
			buffer[2] = 'h';
			buffer[3] = 'o';
			buffer[4] = ':';
			buffer[5] = ' ';

			// Timeout is set to 10ms, but it shouldn't be necessary as data is already available
			int bytes_received = dma_uart_rx( &buffer[6], bytes_available, 10 );

			printf("received %d\r\n", bytes_received);
			// Respond with 'Echo: ' + received data
			dma_uart_tx( buffer, bytes_received + 6 );
		}
	}
}
`

joyhope, 2025-02-25:
Missing rcc setup

	RCC->APB1PCENR |= RCC_APB1Periph_USART3;
	RCC->APB2PCENR |= RCC_APB2Periph_GPIOB ;

The code works..

joyhope, 2025-02-25:
The library is very helpful for the ch32 serial cpu.  Thanks to the author.

joyhope, 2025-02-25:
small change: 
` static uint16_t dma_uart_available()
 {
  return  (rx_buffer_write_pos - rx_buffer_read_pos + RX_BUFFER_SIZE) % RX_BUFFER_SIZE;
 }`

cnlohr, 2025-03-11:
Would you please consider pushing this as an actual demo?  A number of people would appreciate it, and, your issue is poorly formatted, so it would be difficult for people to get the information they need from it.

cnlohr, 2025-03-11:
@joyhope  Would you please consider pushing this as an actual demo?  A number of people would appreciate it, and, your issue is poorly formatted, so it would be difficult for people to get the information they need from it.