cnlohr/ch32fun #550: How to use `funAnalogRead(ANALOG_X)`?

BrianPugh, 2025-03-04:
This might be obvious from actual experimentation, but I'm trying to figure out `ch32fun` while my development boards are in the mail.

The `gpio_and_adc.c` example has the following:

```
	funPinMode( PA1, GPIO_CFGLR_IN_ANALOG );    // Corresponds to analog in A0
	funPinMode( PA2, GPIO_CFGLR_IN_PUPD );      // Corresponds to analog in A1
	funPinMode( PC4, GPIO_CFGLR_OUT_50Mhz_OD ); // Corresponds to analog in A2
	funPinMode( PD2, GPIO_CFGLR_OUT_50Mhz_PP ); // Corresponds to analog in A3
```

which then correspond to:
```
			funAnalogRead( ANALOG_0 ),
			funAnalogRead( ANALOG_1 ),
			funAnalogRead( ANALOG_2 ),
			funAnalogRead( ANALOG_3 ) );
```

Is it just that the order in which `funPinMode` is called determines which pins get associated with which `ANALOG_x` tag?

prosper00, 2025-03-04:
you'd need to check the datasheet to see what analog channel is associated to which GPIO/Pin. You could also write some macro definitions if that would help it make more sense to you. #define PA1_Analog ANALOG_0 sort-of-thing

BrianPugh, 2025-03-04:
with a focus on the ch32v003, [from the datasheet](https://www.wch-ic.com/downloads/CH32V003DS0_PDF.html), is this the relevant chart (Page 13)?

[image]

This almost matches `gpio_and_adc.c` example, but it has A0 & A1 swapped. Is that an error in the example?

prosper00, 2025-03-04:
Sure looks that way. I haven't tested that example myself, but the datasheet is the 'golden source of truth'

BrianPugh, 2025-03-04:
thanks! I'll close this issue once I get my hands on some hardware and can try it out myself; I'll open a PR if the current example is incorrect.

cnlohr, 2025-03-04:
Thanks! Please check - if it is wrong I'd love to correct. 

I vaguely recall A1/A0/A2 being flipped from when I was using capsense.

BrianPugh, 2025-03-12:
alright, so I finally got some hardware, and first off; thank you so much for this project; I've literally never had an easier time getting code onto a microcontroller compared to ch32fun.

I was messing with the ADC code, and reduced the example to this:
```C
#include "ch32fun.h"
#include <stdio.h>

int main()
{
	SystemInit();

	funGpioInitAll();
	funAnalogInit();

	funPinMode( PA1, GPIO_CFGLR_IN_ANALOG );    // Corresponds to analog in A0
	funPinMode( PA2, GPIO_CFGLR_IN_ANALOG );      // Corresponds to analog in A1
	funPinMode( PC4, GPIO_CFGLR_IN_ANALOG ); // Corresponds to analog in A2
	funPinMode( PD2, GPIO_CFGLR_IN_ANALOG ); // Corresponds to analog in A3

	while(1)
	{

		printf( "%d %d %d %d\n",
			funAnalogRead( ANALOG_0 ),
			funAnalogRead( ANALOG_1 ),
			funAnalogRead( ANALOG_2 ),
			funAnalogRead( ANALOG_3 ) );

		Delay_Ms(1000);
	}
}
```

I am probing the different pins with a 10k ohm resistor connected to 3v3. I am able to get PC4 and PD2 to read correctly, but I cannot get PA1 or PA2 to work. Any thoughts?

cnlohr, 2025-03-12:
Can you post a copy of your funconfig.h as well?  If you configure the external oscillator, it will break analog inputs from there on out.

BrianPugh, 2025-03-12:
alright, this is because I'm using the 

the manual says this:

> PA1 and PA2 as crystal pins, so R4, R5 resistors are not soldered by default, P1 row of pins lead to PA1
and PA2 pins function cannot be used, if you need to use PA1 and PA2 pins as ordinary I/O, you need to
weld their own, while the need to remove Y1, C7, C8.

In slightly better english:
1. From the factory, these pins are soldered up to a crystal.
2. To use them as GPIO, you need to solder bridge R4, R5 and remove Y1, C7, C8.

Because I'm using it in it's factory default configuration, it makes sense why PA1 and PA2 are not working. I'll remove these soon and give this another test.

BrianPugh, 2025-03-12:
with the crystal and it's caps removed, plus with R4 and R5 bridged, PA1 and PA2 actually function now.

As suspected, by the datasheet:
* `PA1` corresponds with `ANALOG_1`
* `PA2` corresponds with `ANALOG_0`

I'll make a PR updating the example.

cnlohr, 2025-03-12:
Merged!

PR #559 (2025-03-12): Fix swapped `A0`/`A1` in `gpio_and_adc.c` example. 
Add additional notes regarding the external oscillator for the ch32v003 evaluation board.

Resolves #550.
closed 2025-03-12 by cnlohr