cnlohr/ch32fun #556: GPIO_Speed_50MHz or should it be 30MHz?

BrianPugh, 2025-03-11:
In the ch32v003 reference manual, the GPIO can be configured to 10mhz, 2mhz, or 30mhz; however in ch32fun the corresponding enum name is `GPIO_Speed_50MHz`. After viewing WCH's official examples, it seems like this was a [typo that was fixed in their 2.0 release (link to random github that mirrors it)](https://github.com/cjacker/ch32v003evt_gcc_makefile/commit/cbd65ebc87145c2a43f8ab7a737fb7f449cad466#diff-17138fc50836498466f33d2f61a293ea83fae6defd8516e59190b263889003e4L40). It might be good to deprecate/alias the 50MHZ enum and rename it to 30MHz.

[image]

cnlohr, 2025-03-12:
The original reference manual called it 50MHz everywhere, and the slew rates when measured are closer to 50 the 30 if compared to 10 as a reference.  So we decided to keep it 50.

cnlohr, 2025-03-12:
I wonder if there is a material difference between the early silicon and later silicon 🤔

cnlohr, 2025-03-12:
I am tempted to click submit on this https://github.com/cnlohr/ch32fun/pull/558

BrianPugh, 2025-03-12:
I'm trying to find a changelog somewhere that could possibly offer up an explanation for the change, but I don't speak/read Chinese 😅