cnlohr/ch32fun #557: Unknown Chip Error

PhysicsUofRAUI, 2025-03-12:
I was just trying to set up a toolchain for the ch32v203F8P6 and got the following error. 

```
riscv64-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32V203 -DMCU_PACKAGE=2 -DTARGET_MCU_LD=2 -DTARGET_MCU_MEMORY_SPLIT= ../../ch32fun//ch32fun.ld > ../../ch32fun//generated_CH32V203F6P6_.ld
riscv64-unknown-elf-gcc -o mcotest.elf ../../ch32fun//ch32fun.c mcotest.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32imac -mabi=ilp32 -DCH32V20x=1 -static-libgcc -I/usr/include/newlib -I../../ch32fun//../extralibs -I../../ch32fun/ -nostdlib -I. -Wall  -Wl,--print-memory-usage -Wl,-Map=mcotest.map -lgcc -T ../../ch32fun//generated_CH32V203F6P6_.ld -Wl,--gc-sections
Memory region         Used Size  Region Size  %age Used
           FLASH:         700 B        32 KB      2.14%
             RAM:          0 GB        10 KB      0.00%
riscv64-unknown-elf-objdump -S mcotest.elf > mcotest.lst
riscv64-unknown-elf-objcopy  -O binary mcotest.elf mcotest.bin
riscv64-unknown-elf-objcopy -O ihex mcotest.elf mcotest.hex
make -C ../../ch32fun//../minichlink all
make[1]: Entering directory '/home/kodyrogers/Documents/ch32fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/kodyrogers/Documents/ch32fun/minichlink'
../../ch32fun//../minichlink/minichlink -w mcotest.bin flash -b
Found WCH Link
WCH Programmer is CH32V307 version 2.7
Already Connected
Full Chip Type Reply: [4] 81-55-01-01-00-00-00-00-00
Unknown chip fallback
No detected chip.
Setup success
Interface Setup
Writing image
Chip Type: 000
Unknown chip type.  Report as bug with picture of chip.
Vendored: 00000000
marchid : 00000000
HARTINFO: 00000000

Image written.
```

I assumed the place to post would be here, but please let me know if I am wrong.

[image]
[image]

cnlohr, 2025-03-12:
I don't know anyone who's used the CH32V203F8P6. I am worried because the part overloads the SWD interface

[image]

I think it should be OK. But I am not very confident. 

What pins on the programmer are you connecting SWD and SWC into?

PhysicsUofRAUI, 2025-03-12:
Hey. It is working now. 

There were multiple wiring errors on my part. 
1. I was connecting the wrong side of the double row connector from the programmer. 
 - I ended up connected the SWDIO and SWCLK to LEDs to see what happens
 - When I had them connected wrong one was one all the time and the other was off. Looked at the pin out and realized the one that was on was connected to 3V3
 - wired to the right pins and the LEDs blinked. Then I wired the controller.
2. I was trying out your 'examples_v20x/blink' and had the LED on pin 20. For some reason I thought I saw PA14 that on the datasheet, but I am looking at it now and can't find it. Switched to PA10 and then it worked.

As for using the odd chip I bought a bunch of them a year or two ago when almost all the chips were sold out from LCSC and this was the only version left. It is like double or triple the price that got all the buzz but I figured I should get some to play with. Then when I have finally found the time the cheap ones are available again.

Thank you for making this library!

PhysicsUofRAUI, 2025-03-12:
I would like to offer to add this chip, but honestly not entirely sure where to start right now. Also it definitely appear it is not a popular chip. There is only 47 in stock in LCSC and it is 0.71/each!

cnlohr, 2025-03-12:
Just to clarify, are you up and running?  If so I take it you were able to use SWDIO and SWCLK?  If so, please let us know and close this issue.

PhysicsUofRAUI, 2025-03-13:
Yes I am!

closed 2025-03-13 by PhysicsUofRAUI